pageauc/speed-camera #2: Analysing larger images?

afletch, 2017-02-05:
When using a pi camera v2 with a higher resolution, would it aid accuracy to be processing higher resolution images, and if so how is this achieved?

Changing CAMERA_WIDTH and CAMERA_HEIGHT seems to only zoom in, creating a larger output image but blurred/pixelated version with heavy cropping of all sides.

pageauc, 2017-02-05:
Motion tracking uses captured frames from a camera video stream using the
class PiVideoStream in speed-cam.py. The video stream resolution is based
on the CAMERA_WIDTH and CAMERA_HEIGHT variables.  This would be a portion
of the Camera's full resolution frame .per
http://picamera.readthedocs.io/en/release-1.12/fov.html   Actual opencv
motion processed is done with a cropped portion of the video stream
based on

image_crop = image2[y_upper:y_lower,x_left:x_right]

at approx line 355 in function  speed_camera() and this determines the
detection accuracy so if you increase the CAMERA video stream resolution
you might also need to increase the crop area variables if you wanted
better tracking accuracy.

To get faster response a separate camera image is NOT taken.  Since there
is already a video stream frame, It is used for the final saved image and
is just scaled up in size using WINDOW_BIGGER variable as a multiplier.
Setting this to 0 or 1 saves camera video frame size image that is not
resized.  Using the original video frame avoids stopping/restarting video
stream to setup and take a final image that takes time and may not capture
the moving object due to the time delay in switching from stream to photo
mode.

The camera video frame size can be made larger up to the maximum size video
allowed by the particular camera module.  This will have performance and
memory consequences and possibly stability issues.  V2 has better specs.
This would slow frame processing speed but since OpenCV processing is based
on a cropped area of the full size video frame, performance should be based
on the image_crop variable above.  The settings are a compromise since not
all users will be running RPI3 with V2 camera module.

I have not tested very large video stream size settings too much.  Note
please be aware that the camera still internally crops full pixel
capability per   http://picamera.readthedocs.io/en/release-1.12/fov.html

I would suggest you experiment with various settings on your hardware to
see how far you can push settings to maximize final image output and
detection performance but please realize you will still not be able to get
full camera pixel resolution per link above.

Your feedback is appreciated
Regards
Claude ....

afletch, 2017-02-05:
Thanks for the detailed explanation.
Despite having played with this extensively, it's still rather confusing what the values should be.

Would it be possible to provide some example settings for a higher-spec combination of Pi3 and cam2, using higher FPS and higher resolution tracking/images? I'm sure this would be useful for others too, and serve as an example from which to further test and tweak.

Thanks again.

pageauc, 2017-02-05:
I just did a test on my RPI3 using CAMERA_WIDTH 1920 CAMERA_HEIGHT=1088 and
WINDOW_BIGGER = 1

I did not change the webserver settings but I looked at the final 1080
output image and it is pretty good quality.  I also did Not change the
default opencv detection area settings.  They are per below.

y_upper = 100          # Exclude event if y less that this value default=100
y_lower = 200          # Exclude event if y greater than this value
default=200
x_left  = 25              # Exclude event if x less than this px position
Default=25
x_right = 295

NOTE the webserver size iframe would need to be increased as well but not
be  very usable on smaller monitor sizes.

FPS response was good, but remember opencv performance is base on the
detection area restriction per x and y detection exclusion settings.  If
you make the detection area the full size of the image I would imagine
performance would be slower but have not tested.

I suggest you do some testing.

So if you are trying to maximize final output image capture quality then
you can just increase the camera width and height variables and set window
bigger = 1 .  This is only on a V2 camera module and suitable RPI with a
quad core for best results.

Settings changes need to be optimized for hardware, performance, stability
and results you are looking for.

Let me know how your testing goes and what results you are trying to achieve

Regards
Claude .....

afletch, 2017-02-05:
Much appreciated, I shall take this and test further and feed back.

pageauc, 2017-02-05:
fletch

I appreciate your feedback.  I have updated speed-cam.  makehtml.py now
creates image links on image and data field to allow viewing full size
images.  I have also added config.py.240 and config.py.720 for lower power
and higher power RPI's.  The default is 720.  You may need to customize the
motion detection area but try not to make it too big since it will affect
opencv processing speed.  Also fixed a few potential issues in speed-cam.py
if WINDOW_BIGGER and image_bigger variables are set to zero or less.  This
would cause a python crash.  Note you will need to regenerate the html
pages and will need to clear browser cache.  I think you will like the
changes.

To get the changes rerun the github curl command as the install script has
changed

curl -L https://raw.github.com/pageauc/rpi-speed-camera/master/speed-install.sh
| bash

Note when working with a larger image the various exclusion settings should
be reviews such as min area Etc.  since the pixel range id different.
I have done some basic testing so if you notice something let me know via
github issue.

Please let me know your testing feedback

Again thanks for your input.
Claude ....

pageauc, 2017-02-06:
See latest release for updates

closed 2017-02-06 by pageauc

pageauc, 2017-02-06:
I have updated speed-cam.py to ver 3.6 that now auto creates html path
folder(s) and associated images symlink pointing to the image folder path.
This is required for the webserver to work properly within its root html
folder.

Please upgrade.  I believe I have now covered most of the issues with the
new release ...

Regards
Claude ...