pageauc/speed-camera #3: findContours() error since recent change

afletch, 2017-02-05:
Since the recent changes (mentioning newer opencv which raspbian doesn't have?), `speed-cam.py` fails with:

```
Traceback (most recent call last):
  File "speed-cam.py", line 521, in <module>
    speed_camera()
  File "speed-cam.py", line 395, in speed_camera
    thresholdimage, contours, hierarchy = cv2.findContours( thresholdimage,cv2.RETR_EXTERNAL,cv2.CHAIN_APPROX_SIMPLE )
ValueError: need more than 2 values to unpack
```

Rolling back the changes in this area, made in https://github.com/pageauc/rpi-speed-camera/commit/cfa677e9a56592c82f8da5a03d55671b7897976f , works around the problem.

pageauc, 2017-02-05:
I have tested under latest opencv2 and 3 using python 2 and 3 without issue.  I previously had code to detect python2 and 3 since opencv2 and 3 had slightly different syntax.  The new versions now have the same syntax so I took the detection out.  This means you will need the latest Jessie and opencv updates.  
It looks like you are running under python2 but not sure what version of Raspbian you are using.  My suggesting would be to run latest Jessie and opencv versions.  If you rerun the install this should update Rasbian but if you are not running Jessie you will have problems.  

This is type of thing is due to changes being made to the python libraries.
Let me know if this is still a problem ....

Regards Claude ...

pageauc, 2017-02-05:
I have updated speed-cam.py and added a try except for the findcontours.  It should run the alternate syntax and recover with the new syntax.  If that fails it will exit with a message.  I do not have older build so if you can test this and it runs OK let me know.

Thanks Claude ...

pageauc, 2017-02-05:
Did some testing and opencv error was not a value error.  Just replaced with a try except and this seems to work OK.  Download ver 3.4 and this should work.  Let me know if there are still problems.  I could not just check for opencv3 vs 2 since both now have same syntax except for older versions of python 2.  This should be a good solution.

closed 2017-02-06 by pageauc