pageauc/speed-camera #5: Some help with IMAGE_VIEW_FT

nibelungen, 2017-02-15:
Hi,

maybe i can get some help with the IMAGE VIEW FT value - imho there must be something wrong.
I think the speed measured is nearly double as fast as they really drive.
Heres a litte image of the road with the camera in place. Maybe the angle is the problem?

[image]

Im using the following calculation (320*45)/120 = 120

Config looks like:
`# Motion Event Settings
# =====================
IMAGE_VIEW_FT = 120     # Set the width in feet for the road width that the camera width sees 120px 45ft = (320*45)/120
SPEED_MPH = False       # Set the speed conversion  kph=False  mph=True
track_len_trig = 100   # Length of track to trigger speed photo Default=100
track_timeout = 1      # Number of seconds to wait after track End (prevents dual tracking)
event_timeout = 2      # Number of seconds to wait for next motion event before starting new track

# Camera Settings
# ===============
CAMERA_WIDTH = 320    # Image stream width for opencv motion scanning default=320
CAMERA_HEIGHT = 240    # Image stream height for opencv motion scanning  default=240
CAMERA_FRAMERATE = 75  # framerate for video stream default=35 90 max for V1 cam. V2 can be higher
CAMERA_ROTATION = 0    # Rotate camera image valid values are 0, 90, 180, 270
CAMERA_VFLIP = True   # Flip the camera image vertically if required
CAMERA_HFLIP = True   # Flip the camera image horizontally if required
`
Thanks in advance!!

pageauc, 2017-02-15:
Just looking at your calculation I would not think a typical vehicle would
be 45 ft.  Most vehicles would be 15 - 25 ft  Basically the calculation is
using something of a known length and the number of pixels it appears on
the image (using the calibration image with hatch lines).  This calibrates
object speed for any objects at the reference distance away.  Eg  an object
20 ft long appears small (fewer pixels long) the farther away it is away
from the camera.  For more details see wiki
https://github.com/pageauc/rpi-speed-camera/wiki/Calibrate-Camera-for-Distance

The system works best when speed is calculated near the middle of the
camera view and not at a large angle but if you have a quad core RPI then
it is fast enough to allow shortening the track length

You can also use trial and error.  Set camera in calibration mode and drive
by with your vehicle at a known speed and adjust IMAGE_VIEW_FT to give the
known speed.  Then anything of any length moving at that distance will have
a pretty accurate speed.   You can also reduce the width size of the motion
crop area.  This will restrict motion tracking to a smaller area (length).
You would also have to adjust

The latest version of speed camera is release 3.8 and default setting is
for quad core RPI taking 720p although there is an alternate 480 config.py.

I believe your initial formula indicates a length of vehicle that is not
correct since 45 ft would appear to be too long for a normal car which is
the majority of traffic and easier to get a real measurement for that model
of vehicle.

Let me know if this helps.  Once the camera is calibrated correctly it
seems to give reasonable results although a four lane road would be a bit
of a problem since traffic going in different directions would be different
distances away.  I was thinking of adding a setting for adjusting speed for
traffic moving from left to right and right to left.  The camera speed can
measure anything moving and it does not have to be vehicles.

Let me know how you are doing
Claude ...

nibelungen, 2017-02-15:
Hi Pete,

thanks for your really long answer. 
The car length could always be different or has it to be a specific length?
45 ft is in my case the distance from the camera to the middle of the road.
120 px i have counted in the already reduced area.

Maybe this calib pictures  makes it a bit clearer:

[image]

I will change it to 720p and make a test drive - if the driven speed is 45 km/h and the calculated speed would be for example 90 km/h then half of the IMAGE_VIEW_FT would be the right value?

Thanks,
andreas

pageauc, 2017-02-16:
Looks like vehicle pixel length is approx 95 pixels.  Not sure of vehicle
length but it would appear to be a compact car maybe 16 ft long  Vehicle
looks similar to a toyota camry or honda civic.  For better accuracy you
would need to actually measure vehicle or know it's overall length.
Therefore using the formula

( 320 x 16 ) / 95 = 52

IMAGE_VIEW_FT = 52

This calibration once confirmed would give a good speed reading for
vehicles of any length in the camera view.

Hope this helps.
Claude ...

nibelungen, 2017-02-16:
Okay. So the distance from camera to object is not relevant? 
And if for example a Bus will pass it will work too? 
Sorry for the questions but i currently dont understand the math behind that...

pageauc, 2017-02-16:
It is pretty simple.  You are essentially calibrating a real distance for
each pixel at a specific distance away from the camera.  The farther
something is away from the camera the longer distance each pixel will
represent.  So if you calibrate a length that each pixel represents. You
can then calculate pixels moved in a unit of time or speed. Using the
calibration information you can convert an actual distance per hour eg
pixels per hour, gives kph or mph or any distance length per hour or unit
of time thus speed.

Hope this explains the idea and how the calibration results in being able
to calculate actual speed.

Claude ...

nibelungen, 2017-02-16:
Ah - got it, i think... 
So i dont need the distance from camera to vehicle, just have to calculate how long f.e. one meter is at the car position in pixel... So i could go out there with a big ruler and put it in place where the car will drive and measure the 1 meter ruler in pixel?!
Will do this tomorrow - it´s dark outside, no chance for good picture quality.

Thanks again and i think this 'issue' could be closed!

best regards,
Andreas

pageauc, 2017-02-16:
Yes that would work.

nibelungen, 2017-02-17:
Hi Claude,

i updated the values according to real date, but i think there is still something wrong, the speed
is much too low - will make a test drive later but so far i can say the speed of most cars is over 30 kph.
```
# Motion Event Settings
# =====================
IMAGE_VIEW_FT = 57     # Set the width in feet for the road width that the camera width sees
SPEED_MPH = False       # Set the speed conversion  kph=False  mph=True
track_len_trig = 100   # Length of track to trigger speed photo Default=100
track_timeout = 1      # Number of seconds to wait after track End (prevents dual tracking)
event_timeout = 2      # Number of seconds to wait for next motion event before starting new track

# Camera Settings
# ===============
CAMERA_WIDTH = 1280    # Image stream width for opencv motion scanning default=320
CAMERA_HEIGHT = 720    # Image stream height for opencv motion scanning  default=240
CAMERA_FRAMERATE = 25  # framerate for video stream default=35 90 max for V1 cam. V2 can be higher
CAMERA_ROTATION = 0    # Rotate camera image valid values are 0, 90, 180, 270
CAMERA_VFLIP = True   # Flip the camera image vertically if required
CAMERA_HFLIP = True   # Flip the camera image horizontally if required
```

Maybe there is something else i have to take care of?

[image]

nibelungen, 2017-02-17:
Here is a calib picture of an [opel meriva](http://media.opel.com/media/intl/en/opel/vehicles/newmeriva/2014.html) length = 4288 mm

[image]

I count 330 px - 14.06824 ft for the vehicle 1280px width
(1280*14.06)/330 = 54,53

pageauc, 2017-02-17:
increase track_len_trig so speed is triggered when vehicle is fully into
the motion detection box.  or try the 240 config.py profile to see
difference

pageauc, 2017-02-18:
Taking a look at my code for calculating speed it is using the camera width
view eg 320, 720 etc as part of the formula.  I did this to calculate the
full distance of the camera view in ft.  This can be a little problematic
and after looking at the code it would appear that it should just use a
conversion of calibrated distance that each pixel represents without
needing the distance of the camera view.  Eg if a vehicle is 100 pixels
long in the calibration image and the vehicle is 20 ft long then there
would be 5 pixels per foot.  It the vehicle traveled 100 pixels in 1 second
that would be 100/5 = 20 ft per second.  Then just convert ft/sec to mph or
kph.

I will look at simplifying the calibration procedure and fix the code

speed = pixels-in-a-ft * actual-pixels-traveled/actual-travel-time *
convert-units

This takes camera view out of the formula.  I am leaving on a trip this
morning and will be away for ten days.  Will look at this when I get back
and simplify the code and calibration.  Most cars lengths
http://www.automobiledimension.com/ (not trucks)  fall within a certain
range.  I have been thinking of trying to autocalibrate based on the most
common smallest vehicle so I might add this as an option if the speed
camera is being used to monitor vehicle speed.  Please note that a lot of
times the camera can be used to measure any moving object not necessarily
vehicles.

It will be a few weeks before I can look at the demo code but if you have
time you can also do it since it base on the information above.

Claude ..

nibelungen, 2017-03-06:
Hello Claude,

i hope you enjoy your trip. In the mean time i tried to pass the camera multiple times with the exact same speed and tried to calculate the IMAGE_VIEW_FT. Not with the best luck...
I also tried to set up a larger value for track_len_trig, with no luck - everything over 150 will result in no complete calculation - new track starts but never completes. Everything below is fine, maybe the image size is too big?? I will sitch to 640x480 just for a test.

-Andreas

afletch, 2017-03-12:
I would echo the comments being made here, there does seem to be something amiss with the calculations.

When calibrating with (1280 * 15.5) / 400 = 49.6 I am seeing reported speeds around half what they should be (I'm driving the vehicle at a steady speed).

pageauc, 2017-03-12:
I am redoing the speed calculation method.  The first thing is to make sure
the object contour is completely inside the crop area plus a small buffer
zone.  This will look at the contour corners and make sure the contour is
well within the measurement area with a buffer zone around.  The buffer
zone is required to ensure the contour edges are completely inside the crop
measurement zone.  This will ensure you are not getting a partial contour
as objects start to enter the crop area.  Doing this will avoid problems
with contour center point not moving very much as object enters or leaves
the measurement zone.

Second I am simplifying the calculations.  You will have two variables one
for the length of a calibrations object in pixels and the second will be
the actual object measured length in millimeters.  There will still be
option to convert to mph but it will be based on metric units.  This avoids
the need to manually calculate distance conversion using the image size and
simplifies method since you only need a calibration object pixel size and
actual object size in mm

I have done some of the changes but want to standardize the way the
crop/buffer area is positioned.  I would like to have a default for
positioning crop area at center of image based on a specified crop length
and width.  eg 320 x 240 or similar.  You would need to position the camera
view to position the center crop area at the target measurement area.
Another option might be to override the center position to a user specified
location on the video stream..

I am finding that using a video stream of 720p does heat up the RPI so for
long term operation maybe a smaller stream size might be better.  Testing
is on a RPI3

Some of the above has been coded but have not tested yet.

The original speed camera code processed the whole image for motion and
resized the small image bigger as an option for saving to disk.  The
current code streams larger image and crops area for motion tracking.  This
maintains performance and allows a better quality final saved image with
potentially better resolution but did cause problems due to not taking
partial contours into effect.  Eg as a vehicle (object) is entering
measurement area it will only be partially in the measurement area until it
finally is completely in the measurement crop area.  The above should
resolve this problem

Please remember this was developed as a demo.

I will try to complete these changes in the next week or so.  If you have
any questions or suggestions please contact me.

Regards
Claude ...

pageauc, 2017-03-14:
I have updated the speed camera to simplify calibration using cal_obj_px and cal_obj_mm

I have done some preliminary testing and it appears to be working OK although testing was not extensive due to my location.

Please give this a try and let me know if it runs better.  Note the default config.py is based on 240p image size and has the best performance.  The 720p config is slower

Let me know results of your testing

Thanks Claude Pageau

nibelungen, 2017-03-15:
Hi Claude,

thanks again for your efforts - for me i have never expected to get a solution neither expected to get it in a specific time i just wanted to test your script an my own and stuck at wrong speeds...

I just updated to the new version and give some feedback tomorrow after some cars has passed by.

Again - thanks for your work and support!!

-Andreas

pageauc, 2017-03-15:
Looking forward to your test Results.   Have you run a calibration yet.  My
240p default values is approx 38 ft from camera to road.

Also you could try different values for track_len_trig variable. I have set
it to 50 but try shorter or a bit longer to see results

Claude

pageauc, 2017-03-16:
Released version 4.10  that now correctly adjusts cx due to crop area and displays opencv window positions on full image correctly.  The buffering area now seems to work OK.  Still doing more testing.  Also increased crop dimensions to allow for larger vehicles if the road is close to the camera.

pageauc, 2017-03-16:
Will take a look at two calibrations.  One for L2R and one for R2L.  This will improve accuracy for vehicles moving in both lanes.

nibelungen, 2017-03-17:
First results look good - but there is a little difference of 4-5 kmh, but maybe my calib was not good enough, have to make some more tests and have to update to your new version.

best regards
-Andreas

pageauc, 2017-03-17:
Hi

Please note I have release version 4.10 that fixes some issues with
filtering contours correctly in the crop area.  What I forgot was that the
cx and cy values are based on the crop image and not the full size image.
This meant the filtering to verify a contour was completely inside the crop
area plus buffer space was not working as expected.  This now seems to be
working better.  There can be issues with certain lighting, shade etc that
will result in a contour changing shape.  You can see this sometimes in the
tracking when sometimes it appears to go backward with a slow object but if
the track is long enough this should average out somewhat.   I was thinking
of adding a second calibration for objects going in the opposite
direction.  This would correct for vehicles going L2R or R2L.  Let me know
what you thinks

Thanks for your help to get me motivated.
Claude ....

afletch, 2017-03-17:
I believe I'm getting more accurate figures with the new version, but I need to do some further testing to be certain.

Thanks for all your efforts!

nibelungen, 2017-04-08:
Hi Claude,

after a long time of try and error im nearly done with calibratiing.
Currently my settings are
cal_obj_px = 110        
cal_obj_mm = 4366.0 
and the driven speed was 30 km/h and measured speed was 24.1 km/h. Is there a way to adjust the values to get the correct speed or will i need to adjust and test again? Would that mean that each logged speed was 5.9 km/h too slow?

Best regards, and thanks for your work!

pageauc, 2017-04-08:
As long as the distance of the object away from the camera is the same.  ie
 same side of the road, then speed should be consistent with slight
variations.  All the calibration is saying is that a pixel at a specific
distance away from the camera will represent so many millimeters.
Obviously in practice this can have issues.  Objects to left and right will
be farther away than at the middle of the camera view.  Cars in the near
lane will be closer that those in the far lane (This could be adjusted in
the program but I have not done this.  So there are areas where errors can
occur.

You can adjust the calibration values to adjust for real world conditions
by increasing one or both of the values.  I suggest you just change one.

The whole calibration just uses a sample length at a know distance to
figure out what distance each pixel represents at that distance away from
the camera.  If the objects is farther away then cal_obj_mm will remain the
same but the number of pixels will be fewer.  Likewise if the object is
closer to the camera it will take up more pixels (appear bigger)  Once
calibration is done for a single known object, then speed of any object can
be calculated as long as it is the same distance away from the camera. .

Calibration should be very simple.  The actual length of an object should
remain the same.  ie if you are using the same car or object to calibrate
it will always be the same length in mm.  The only variable will then be
how far the vehicle is away from the camera.  The farther away the lower
the cal_obj_px will be and the closer it is to the camera the more pixels
it will take up in the camera view and cal_obj_px will be a larger value..

Hope this helps
Claude ...

pageauc, 2017-04-08:
Also there are limits to this speed camera.  If an object is very far away
from the camera, then the number of pixels if takes up will be smaller
until if it is far enough away it might only be one pixel and open cv would
not be able to distinguish it as a contour.  Also I calculate the center of
the contours because opencv x, y, h and w can vary as the object moves
along.  I pick the center because it tends to be more stable reference
point.

closed 2017-06-29 by pageauc

afletch, 2017-06-30:
Apologies for not feeding back before, but this has slipped down my list.

I've had this running now for a few months and my general feeling is that there are inconsistencies in the speed reported between different vehicles. Some vehicles are very accurate and others are not. I think it depends on a few factors:

- the background against which the vehicle is moving:
if the background us busy/complex then accuracy of speed is reduced (eg. good against a solid surface like a road, bad against a housing/garden backdrop)

- the colour of the vehicle
this seems to make a big difference, and combines with the first item above. Where there's a greater contrast between the moving vehicle and the background, the better the detection and more accurate the speed.

- the direction of travel
This is an odd one, which I can't explain. Even the same vehicle travelling in different directions seems to be better L-R than R-L

I expect much of the above is down to opencv and way beyond your control, but thought I'd give my feedback in any case, such that it may be useful in some way.

Thanks for your hard work

pageauc, 2017-06-30:
Thanks for your feedback.  This is just a project I wrote for fun to tinker with. Some people re purpose the code for other things beside vehicles.  One user rewrote it in java and some have written it for PC. 

Recently added support for webcam's and did have fun figuring out how to change resolution and other features on a USB webcam on the RPI.  Also did some work on makehtml.py code web page html formatting.  
The project is more of a learning tool and not intended to be super accurate but with more work could be improved by optimizing for direction.  I have a version (not on github) that I use for tracking motion and speed in any direction.  As this is just a hobby I mainly do it for myself as a learning exercise.  I have been retired for eleven years and non of my friends use computers for any type of content creation like music, video, imaging, science, programming, design, problem solving Etc.  To me this seems like a waste of brain power but each their own.  
Anyway this is just my way of providing something back to society where some young person might find it and learn something that might help them.
Appreciate your Comments 
Claude ....