pageauc/speed-camera #10: Release 6.5 Now Compatible with Unix, Windows and RPI

pageauc, 2017-08-11:
I have modified the speed camera code to be compatible with windows and unix platforms using a USB Web Cam as well as a RPI using either a RPI camera module or Web Cam

On windows you should run the latest version of python that includes numpy and opencv. https://www.python.org/downloads/

I have tested on my windows 7 computer and resolved compatibility issues.  The new code includes 
optional free disk space management feature 
and
optional Creation of sub folders that triggers a new folder base on specified max files and/or age of sub folder in hours

Please raise a GitHub issue if you find and problems or have suggestions.
Claude ....

ChrisPetr0, 2017-08-14:
This is so awesome.  I just found this project and was going to buy a raspberry pie but now I'm going to test it out on Windows.  I suppose the documentation to get it running on Windows doesn't exist or needs to be written?  Thanks again, awesome project, can't wait to get it going.

pageauc, 2017-08-15:
Download the latest release from today.  Fixed a bug that crashed speed
camera.  Also fixed makehtml.py so it will recognize subdirectories.

I found performance on my Windows PC with a logitech web cam was slower
than a single core RPI as far as I can tell.  Did not do actually test..

Code does work on windows except for bash scripts so menubox.sh does not
work.  I have not put up instructions except that you need latest python
that includes numpy.

Best way to install is accessing github repo at
https://github.com/pageauc/rpi-speed-camera
Select download zip or you could clone as well

I ran from IDLE session and this seemed to work OK.  Running from a cmd
prompt cut off logging messages.  Use an editor to change config.py.
Initial setup defaults to Calibration mode so as soon as you update
calibration data change calibrate = False per GitHub Repo instructions and
Wiki.
the webserver worked great on windows so I could review speed camera
images.  Also I would recommend looking at configuring for subdirectories
by max file.  I have mine set to 1500 in a directory since one day amounts
to approx 700-750 images.  This setting will create a date/time named
subdirectory in the image folder.
It took a few changes to make the program cross platform.  I am still
thinking that creating a web management interface would be more universal.
Original program was RPI only so I developed the menubox.sh whiptail menu
for speed cam management.  I have done some web development but I would not
call myself an expert by any means.

Let me know how you make out.  I would be interested in any suggestions.

Claude ...

ChrisPetr0, 2017-08-16:
Claude - Thanks - I haven't been able to give this attention yet, but certainly will and thankyou for your detailed response.  I live on a residential "cut - through " street that my township continues to do nothing about.  The speed limit is 25mph and I'm sure I've seen folks doing 50mph + basically every day during the rush hours.  I can't wait to get some tangible evidence and see if something will ever happen so this tool is near and dear to me.  Thanks again and I'll provide an update when I get it running.

closed 2017-09-06 by pageauc