pageauc/speed-camera #11: config.py.720 causes error

turbo-donkey, 2017-08-18:
speed-cam.py version 6.8 using python2 and OpenCV2    written by Claude Pageau
Logging to Console per Variable verbose=True
Traceback (most recent call last):
  File "./speed-cam.py", line 450, in <module>
    def filesToDelete(mediaDirPath, extension=image_format):
NameError: name 'image_format' is not defined

pageauc, 2017-08-18:
Just update repo for all config.py files.  Fixed missing var in 720 and made more consistent with layout of default 240 config.py.  I do most of my testing on a single core older RPI since it does not have the performance of a quad core RPI.  I am sorry I did not take the time to test the 720 bug.  The new configuration turns subdirectory creation on setting maxfiles to 1000 after which a new dated subdir will be created.  This makes it easier to browse images in web browser.  Also I have turned on disk space management by default with free space set to 500 MB.  That should prevent running out of disk space.  The default is to check space every 24 hours. 

IMPORTANT:  Now that speed camera runs under windows and non RPI Unix distros (and maybe even Mac) I have renamed the repo to speed-camera instead of rpi-speed-camera.  The new install script will install to that folder.  I would suggest you save your data and reinstall the latest version.  You may already have that but thought I would mention it.

Sorry for the problem.  
change to the speed camera folder     

    wget -O config.py.720 https://raw.github.com/pageauc/speed-camera/master/config.py.720
    cp config.py config.py.bak
    cp config.py.720 config.py 

Transfer any customization if required.
Start speed camera and test
 
Let me know if this resolves issue
Thanks for your notification
Claude ...

turbo-donkey, 2017-08-19:
Just copied config.py.720 to config.py and stuck some basic config in, and its loading correctly now!

Time to get calibrating, many thanks!

closed 2017-08-19 by turbo-donkey