pageauc/speed-camera #14: Images not being saved in the image folder

Zeal0usD, 2017-09-17:
Followed the instructions and started the program and the web server and even within the lines i am not getting any saves but i can see the red dot move across a motion target, i am testing in night time using a pi cam. Not sure how much light is required for a save.

Is there anything i could be missing to prevent the saving of images.

closed 2017-09-17 by Zeal0usD

pageauc, 2017-09-18:
Default settings should work OK.  I would try during day and review log entries.  Run speed-cam.py directly and view output.  Looks like your camera is working since it is tracking OK.  Let me know if day works better.  BTW please let me know if version of speed-cam.py is latest 6.8 version or is it earlier.

pageauc, 2017-09-18:
I see you closed the issue.  Did you resolve problem?
Regards Claude ...

luisfranciscocesar, 2017-10-30:
6.9v, I also do not save the images, do I execute it and it runs everything well, but does not save the images it captures?

luisfranciscocesar, 2017-11-03:
I also do not save the images, only sometimes save, but not always

pageauc, 2017-11-03:
Are you running 240 or 720 config  (headless or with a monitor).  720 config will have slower response especially on single core RPI's.  Do you think problem is with fast or slow moving objects?
You could try adjusting the track_len_trig variable in config.py.  Also there is a default 2 second event_timeout  setting so slow moving objects that do not move past the track_len_trig setting will not trigger an image.  Also Raspbian is not a real time operating system so there may be other activities that the cpu is doing.

luisfranciscocesar, 2017-11-06:
ok, I'm going to try it, is there any way to get images with more quality of the cars ?, as does the raspitill

pageauc, 2017-11-06:
The image is taken from the video stream either 240 or 720p resolution or any custom resolution you set.  This is done to be able to catch the moving object.  If you try to take a camera still image it takes time to close the stream and allow time for the camera to set AWB so by the time the image is taken fast moving objects would most likely be out of the frame.  Video resolution is inferior to still images.  You could try using the 720p config video stream.  See Wiki for more details.
Claude ...

luisfranciscocesar, 2017-11-07:
I'm looking for the option for image quality, but there is no "quality"
because I notice that the quality is not good, in raspitill there is an option called quality, with that it is improved, but here there is not. ?

luisfranciscocesar, 2017-11-08:
Can you integrate "sharpness", or is this the option?

pageauc, 2017-11-08:
There are several things you can do to improve image resolution and clarity.  Use 720p configuration for higher resolution.  This should be with a quad core RPI if possible.

    cd ~/pi-timolo
    cp config.py config.py.bak
    cp config.py.720 config.py

Increase fps variable to try to reduce blur.  You may have to try different settings.  Default is 25 fps.  This also depends if you are using a RPI3 or quad core cpu and if RPI camera module is original or newer v2 camera.  for more detail see https://www.raspberrypi.org/forums/viewtopic.php?t=145815

    CAMERA_FRAMERATE = 60    

This setting can be as high as 90 fps but might be limited by opencv processing time.  This may also cause some overheating so monitor that as well per command. 

    /opt/vc/bin/vcgencmd measure_temp

If you are using a web cam then it will depend on the usb limitations and web cam specs.  You will have to experiment.

Let me know your results.  At the moment I am working on other things so would appreciate any feedback from your testing.

Thanks Claude ...

pageauc, 2017-11-08:
OK.  I have done some testing and it would appear that lowering the fps will improve image quality but moving objects may still blur a little.  I have added and modified the configuration files.  I now have a 640x480 stream size in config.py.480 file.  You can overwrite the config.py with this version to implement per previously mentioned commands.  Note I have not tested on V2 camera module.
Claude ...

luisfranciscocesar, 2017-11-09:
thanks for the changes, but it would not increase the fps, if only the calculations would be made to the crob only and not to the whole image, that speed-cam only makes the calculations on a cropped area, and not the whole image, it's just an idea I do not know if you could do that. So I think you could increase the resolution beyond 720p, and improve the performance.

pageauc, 2017-11-09:
Yes you can increase the resolution to eg 1080p or max video resolution and set a specific crop area since that is what opencv processes.  I have not tried this but you can do that from the config.py file.  Just set a specific upper, lower, left, and right motion crop settings and try to keep to a minimum to reduce opencv overhead.  I would suggest you give it a try.

pageauc, 2017-11-09:
I have updated configurations to enable image motion rectangle.  Also added a 1080p experimental configuration.  My prelim testing seems to be very slow performance on RPI3.  If you test that config let me know your results.

Claude ...

pageauc, 2017-11-09:
Just recently update config files and install.  1080p is rather sluggish compared with 240 and 480.  This is probably due to handling larger image event though opencv only needs to process the crop area.  Found performance seemed to improve when I reduced fps to 15.  Let me know your results.

luisfranciscocesar, 2017-11-10:
If I also had the same results with 1080p, I noticed that it miscalculates the speed, some cars are 0.5kph, but they are faster than that.

pageauc, 2017-11-10:
Did you calibrate?  Also the track_len_trig variable may need to be increased to improve accuracy.  Note this is an experimental config.  If you have issues I would suggest moving to default 480 config.  Also this program started as a personal challenge for me.  I have been retired for 11 years and programming is just one of my hobbies.  I do not claim program is perfect.  Also accuracy can vary due to objects being closer or farther than the calibration distance ie opposite sides of road.  Also camera angle issues. ie objects can be closer in the center of the camera view than at the edges of the camera view.  Also Raspbian is not a real time operating system so it may do operations that can slow processing and skew results.
Perfection would need a lot more work to achieve for a hobby program.
Claude ...

luisfranciscocesar, 2017-11-10:
if I am playing with the variables, with which version of pi camera you use, I notice that the v2.1 and the 1.3v, have more resolution than the v1.0, but the 1.3v seems to me to have better image quality, . it seems to me. does not have the resolution that 2.1v, but more quality I think if

luisfranciscocesar, 2017-11-10:
another thing that would be interesting, would be to turn 90 degrees, to measure vertically, not the camera, but the one that measures, I saw that look in horizinal, and the rule that appears red is horizontal, is it possible to put it vertically ?, but without turning the camera

pageauc, 2017-11-10:
Yes it would be possible to measure movement horizontally and vertically
with the camera remaining in normal orientation. This would be similar to
one of my track-inout project
here https://github.com/pageauc/track-inout
YouTube Video Here. https://youtu.be/9et7QpbT5cc