pageauc/speed-camera #17: tracking vehicules

AmalEzzouhri, 2018-02-27:
Hello
how can i use deep reinforcement learning in tracking many cars in video plzz

pageauc, 2018-02-27:
Here is an explanation and some sample code for tracking multiple objects.
This might help.  There are also deep learning methods.

https://www.learnopencv.com/object-tracking-using-opencv-cpp-python/

also

https://github.com/srianant/kalman_filter_multi_object_tracking

I suggest you do more googling for various methods of tracking multiple
objects.  It can be harder than you think.

Claude ...

pageauc, 2018-02-27:
Adrian has quite a few advanced blogs on deep learning tracking methods
that can be used to track multiple objects

See sample post here.

https://www.pyimagesearch.com/2017/09/18/real-time-object-detection-with-deep-learning-and-opencv/

closed 2018-03-19 by pageauc