pageauc/speed-camera #19: How to use ip cam?

radhitee, 2018-03-25:
Hi, your code is awesome, I've tried this code,but I want to use my IP cam, how to use ip cam with this code?
Thanks

pageauc, 2018-03-25:
I do not have an IP camera to do testing but you could try reading directly from IP camera using  config.py  WEBCAM_SRC variable per sample below.  Substitute your IP camera URL per sample below.  Give that a try and let me know if it works.

WEBCAM = True
WEBCAM_SRC = "http://192.168.0.153:5000/video_feed?dummy=mjpg"

If not you would need to write another threaded class to read the IP camera video stream from the IP camera.  See sample code below

    import cv2
    import urllib 
    import numpy as np

    stream=urllib.urlopen('http://192.168.0.153:5000/video_feed')
    bytes=''
    while True:
        bytes+=stream.read(1024)
        a = bytes.find('\xff\xd8')
        b = bytes.find('\xff\xd9')
        if a!=-1 and b!=-1:
            jpg = bytes[a:b+2]
            bytes= bytes[b+2:]
            i = cv2.imdecode(np.fromstring(jpg, dtype=np.uint8),cv2.CV_LOAD_IMAGE_COLOR)
            cv2.imshow('i',i)
            if cv2.waitKey(1) ==27:
                exit(0)

radhitee, 2018-03-25:
Thanks, I will try

pageauc, 2018-03-26:
Below is a class link for reading IP camera stream and converting to jpg.  It could be imported directly.  Note it does not handle login to camera.  If I get a chance I will setup one of my RPI's as an IP camera ant do some testing to read stream into speed camera similar to class code below.  As we will shortly be travelling back home this may take a while.  Claude ...

    https://github.com/BrandonJoffe/home_surveillance/blob/master/system/Camera.py