pageauc/speed-camera #20: After Application Crash Unable to connect to USB Camera on Windows

BigDuke6nc, 2018-03-26:
After the application crashes with the following error:

Traceback (most recent call last):
  File "C:\Users\****\Downloads\speed-camera-master\speed-cam.py", line 1084, in <module>
    speed_camera() # run main speed camera processing loop
  File "C:\Users\****\Downloads\speed-camera-master\speed-cam.py", line 763, in speed_camera
    image_crop = image2[y_upper:y_lower, x_left:x_right]
TypeError: 'NoneType' object has no attribute '__getitem__'

The application does not detect the USB camera and I receive the following error. Only a machine reboot will resolve the issue.

2018-03-26 08:36:14 INFO     <module>   Initializing USB Web Camera Try .. 1
2018-03-26 08:36:18 WARNING  speed_camera Problem Connecting To Camera Stream.
2018-03-26 08:36:18 WARNING  speed_camera Restarting Camera.  One Moment Please ...
2018-03-26 08:36:22 INFO     <module>   Initializing USB Web Camera Try .. 2
2018-03-26 08:36:26 WARNING  speed_camera Problem Connecting To Camera Stream.
2018-03-26 08:36:26 WARNING  speed_camera Restarting Camera.  One Moment Please ...
2018-03-26 08:36:30 INFO     <module>   Initializing USB Web Camera Try .. 3
2018-03-26 08:36:34 WARNING  speed_camera Problem Connecting To Camera Stream.
2018-03-26 08:36:34 WARNING  speed_camera Restarting Camera.  One Moment Please ...
2018-03-26 08:36:38 INFO     <module>   Initializing USB Web Camera Try .. 4
2018-03-26 08:36:38 ERROR    <module>   USB Web Cam Not Connecting to WEBCAM_SRC 0
2018-03-26 08:36:38 ERROR    <module>   Check Camera is Plugged In and Working
2018-03-26 08:36:38 ERROR    <module>   on Specified SRC
2018-03-26 08:36:38 ERROR    <module>   and Not Used(busy) by Another Process.
2018-03-26 08:36:38 ERROR    <module>   speed-cam.py 8.6 Exiting Due to Error

pageauc, 2018-03-26:
This appears to be a problem with USB cam being still used.  I programmed the 4 retries to allow time to acquire the USB camera.  I have not experienced this issue.  Might also be the thread is still running when the crash occurs and does not release USB cam.  What are the details of the initial crash.  What seems to be causing crash.  Are there any error messages.?
Release 8.61 on GitHub - added a vs.stop after retries failed but don't think this is the problem.  I think the original failure is the problem.  Sometimes if you try to read an image file before it is fully written or you are tying to save to a remote drive you can crash opencv.  If you provide details of original crash I will try to see if I can trap error and close thread.
Waiting for Details
Claude ....

BigDuke6nc, 2018-03-26:
Hi Claude. Thanks for the quick reply. I did post the original error. Here:
Traceback (most recent call last):
File "C:\Users****\Downloads\speed-camera-master\speed-cam.py", line 1084, in 
speed_camera() # run main speed camera processing loop
File "C:\Users****\Downloads\speed-camera-master\speed-cam.py", line 763, in speed_camera
image_crop = image2[y_upper:y_lower, x_left:x_right]
TypeError: 'NoneType' object has no attribute 'getitem'

I figured that after the crash the process still has a hold of the camera. I stopped all processes of Python after the crash as it was still running post crash and it didn't help.

Also, Thanks for creating this wonderful program. It has been fun to play with.

pageauc, 2018-03-26:
I know windows 10 has issues with webcams and drivers can sometimes cause problems.  I tested using windows 7 and seemed to be OK for me.  I ran it for many hours.  The error seems to occur because an incomplete stream image was received and could not be cropped.  I have release version 8.7 that does a try except around the image cropping.  This will issue an error message and continue to try to read another image.  Give this version a test.  Download the ver 8.7 version of speed-cam.py from github.  Please test with your existing config.py and try to reproduce error.  It is possible the webcam hardware has a glitch due to overheating or some other issue and sends an incomplete image that caused the crash.  If you see the error message a few times then this change should prevent crash.  If you get a lot of error messages it could be problem with webcam not being able to keep up.
Let me know results of your testing.
Claude ...

BigDuke6nc, 2018-03-26:
Thanks for the update. Drivers are up to date on the web cam. I do see a bunch of those errors. Log info below. I started receiving these errors out of the blue and nothing is being "tracked". After the error message the web cam is no longer detected on the computer prior to this version. With the new version you get the error message you are trapping but at least the webcam does not get removed from the computer and no restart is required.

2018-03-26 14:15:16 INFO     speed_camera New  - cxy(179,124) Start New Track
2018-03-26 14:15:16 INFO     speed_camera  Add - cxy(192,124) 33.1 mph px=13/75 C=3 103x67=3259 sqpx R2L
2018-03-26 14:15:16 INFO     speed_camera  Add - cxy(205,124) 27.0 mph px=26/75 C=5 103x67=3266 sqpx L2R
2018-03-26 14:15:16 INFO     speed_camera  Add - cxy(218,124) 25.4 mph px=39/75 C=3 103x68=3185 sqpx L2R
2018-03-26 14:15:16 INFO     speed_camera  Add - cxy(232,124) 25.2 mph px=53/75 C=2 103x68=3223 sqpx L2R
2018-03-26 14:15:16 INFO     speed_camera  Add - cxy(245,124) 24.7 mph px=66/75 C=3 103x68=3255 sqpx L2R
2018-03-26 14:15:16 INFO     speed_camera  Out - cxy(162,137) Dist=83 is >=15 px C=4 63x35=1410 sqpx L2R
2018-03-26 14:15:16 INFO     speed_camera  Out - cxy(176,138) Dist=69 is >=15 px C=4 64x34=1447 sqpx L2R
2018-03-26 14:15:16 INFO     speed_camera  Out - cxy(189,137) Dist=56 is >=15 px C=4 65x35=1452 sqpx L2R
2018-03-26 14:15:16 INFO     speed_camera  Out - cxy(203,123) Dist=42 is >=15 px C=4 63x64=2035 sqpx L2R
2018-03-26 14:15:16 INFO     speed_camera  Out - cxy(218,136) Dist=27 is >=15 px C=6 62x38=1424 sqpx L2R
2018-03-26 14:15:17 INFO     speed_camera  Add - cxy(232,122) 8.3 mph px=53/75 C=3 62x65=1927 sqpx L2R
2018-03-26 14:15:17 INFO     speed_camera  Add - cxy(245,122) 9.5 mph px=66/75 C=5 63x65=2269 sqpx L2R
2018-03-26 14:15:17 INFO     speed_camera  Add - cxy(258,122) 10.47 mph px=79/75 C=4 65x65=2198 sqpx L2R
2018-03-26 14:15:17 INFO     speed_camera  Ave Speed is 10.5 mph L2R 
2018-03-26 14:15:17 INFO     speed_camera  Saved media/images\speed-20180326-1402/speed-20180326-141517.jpg
2018-03-26 14:15:17 INFO     speed_camera End  - Tracked 79 px in 0.423 sec
2018-03-26 14:23:18 ERROR    speed_camera image2 Stream Image is Not Complete. Cannot Crop.
Previous line repeats  100 or so times.

pageauc, 2018-03-26:
OK Thanks.  Not sure if problem is due to overheating of webcam (possible),  USB cable problem, driver of some other issue.  It appears once camera fails it does not send images again so problem is not intermittent.  How long does it take for camera to fail?

If you have a consistent time for cam to fail I would suspect overheating.  You could try putting camera in a fridge or freezer for a period of time to cool it off, then try again and see if it runs longer before failing.  This would indicate problem is due to overheating. 

Try another web cam and see if problem occurs.  If it is OK then would indicate problem with original camera.  If consistent failure with original camera, then might be computer related. 

Have you tried another usb port to see if problem is possible USB port issue.

Try these and let me know results
Thanks Claude ...

pageauc, 2018-03-26:
Thanks for the update.  If you could get another USB web cam that might help trouble shooting.  I would think cable issue would be intermittent but anything is possible.  Try webcam with no hub and short cable then work if things work OK, systematically add connections back in.  You could try a single USB extension cable rather than use a hub.  
Let me know if you solve the problem. 
Regards Claude ...

pageauc, 2018-03-26:
> Thanks. I tried using another web camera software and left it running and
> after a period of time that application crashed and the camera disconnected
> as well. The cable was not long enough to reach the window so I had it
> chained through a USB hub. That means there could be several points of
> failure. I will continue to troubleshoot. Thanks for adding the trap to
> help with troubleshooting.

Check if the webcam has a reset.  Sometimes there is a small hole that you
can put a paper clip into and hold for 10 seconds or so.  You can check web
for your webcam manual and see if there is a reset.
Claude ...