pageauc/speed-camera #24: Not tracking events

scottcim, 2018-05-05:
Hi,
I'm using a rpi3 and going through the inital setup.  Install went great.  I'm going through the calibration process and I'm not able to trigger a motion tracking event.  
- I start the speed-cam.py and webserver.py (calibration =true)
- I log into my rpi thorough my desptop via the IP:8080.  The SPEED-CAMERA Media page loads, but nothing is captured when I try to trigger an event.  If I check the links on the right they are empty (except for the txt under webserver.txt).
- Manually checked the speed-cam media folders and they are empty.

What I've tested: 
- I've tested the pi-camera and it is working correctly.  
- I've enabled gui_window to display the OpenCV video window and that displays video with the motion area and moving tracking box as expected.
- Log_data_to_CVS is set to True.
- If I kill the processes and run speed-cam.py initiates the tracking and displays it to the screen.

If you need any additional information, let me know,

Thanks!!

scottcim, 2018-05-05:
An update on the issue.  It's only capturing very slow moving objects. It took calibration photos when it captured someone walking down the street.  I tried to calibrate using my car going 10 mph, going past three times.  It did not take calibration photos for any of these.  I went by going under 5mph and it did capture the data and calibration photo.

pageauc, 2018-05-05:
> Closing issue. Disabling the plugin (720picam) seems to have corrected the
> tracking. All vehicles are now being captured.

Yes I have reduced the default configuration to 320x240 since I found it
works more consistently.  The 720 was to try to get better image resolution
but
you need compiled opencv3 for optimum processing.  Motion tracking is more
demanding.

Thanks for the feed back
Claude ...

pageauc, 2018-05-05:
If you need better image resolution you could try 640x480 since I have found it works OK with RPI-3.