pageauc/speed-camera #26: application crashing

jschaftenaar, 2018-06-03:
My application often crashes. After running a trace I see this error being raised.

```
File "/usr/lib/python2.7/trace.py", line 819, in <module>
    main()
  File "/usr/lib/python2.7/trace.py", line 807, in main
    t.runctx(code, globs, globs)
  File "/usr/lib/python2.7/trace.py", line 513, in runctx
    exec cmd in globals, locals
  File "speed-cam.py", line 1125, in <module>
    speed_camera() # run main speed camera processing loop
  File "speed-cam.py", line 968, in speed_camera
    image_prefix)
  File "speed-cam.py", line 619, in saveRecent
    shutil.copy(filename, recentDir)
  File "/usr/lib/python2.7/shutil.py", line 119, in copy
    copyfile(src, dst)
  File "/usr/lib/python2.7/shutil.py", line 69, in copyfile
    raise Error("`%s` and `%s` are the same file" % (src, dst))
shutil.Error: `media/images/speed-20180602-1437/speed-20180603-140835.jpg` and `media/recent/speed-20180603-140835.jpg` are the same file

```

The error is pretty self explanatory without diving further into the code is there a config file I could change to fix this issue?  maybe a threshold or delay somewhere?

pageauc, 2018-06-03:
What OS are you running on.  saveRecent executes an shutil copy if creation of a symlink fails.  I don't see a message about Unix symlink failing.  It might be that there is an existing file in the recent rather than a symlink.  You may want to try clearing all files from the recent folder since there may be a combination of symlinks and files.  Do ls -l to check folder.  Al directory entries should be symlinks on a unix system and only files on a non unix system.  Then restart.  You did not mention what version of speed-cam.py you are running.  The latest should be version 8.86 

BTW I have been running this code continuously on a RPI B single core for over three weeks.

Please keep me updated on your findings.

Claude ...

pageauc, 2018-06-03:
BTW.  Also check on permissions.  When the max entries for recent is reached the oldest symlink/file gets deleted ready for the creation of the new one.  Anyway see if all permissions are OK and files have writeable  permissions.
Just a thought.  Claude ...

jschaftenaar, 2018-06-03:
Thank you for the quick reply. 

`Distributor ID: Raspbian
Description:    Raspbian GNU/Linux 9.4 (stretch)
Release:        9.4
Codename:       stretch`

I am running the master checkout from yesterday. 

I will remove all the files in recent. Will report back if this seems to solve the issue.

Next to this I am using a modified version of webcam480 plugin, this is the config for this
```
# ---------------- User Configuration Settings for speed-cam.py ---------------------------------
#         Ver 8.4 speed-cam.py webcam480 Stream Variable Configuration Settings

#######################################
#    speed-cam.py plugin settings
#######################################

# Calibration Settings
# --------------------
cal_obj_px = 370       # Length of a calibration object in pixels
cal_obj_mm = 4500.0    # Length of the calibration object in millimetres

# Crop Area for motion detection Tracking
# ---------------------------------------
x_left  = 150          # Default= 150 Exclude event if x less than this px position
x_right = 810          # Default= 490 Exclude event if x greater than this px positioni
y_upper = 300          # Default= 140 Exclude event if y less that this value
y_lower = 600          # Default= 340 Exclude event if y greater than this value

# Motion Event Settings
# ---------------------
SPEED_MPH = True      # Set the speed conversion  kph=False  mph=True
MIN_AREA = 200         # Default= 200 Exclude all contours less than or equal to this sq-px Area
track_len_trig = 75    # Default= 75 Length of track to trigger speed photo
x_diff_max = 35        # Default= 35 Exclude if max px away >= last motion event x pos
x_diff_min = 1         # Default= 1  Exclude if min px away <= last event x pos
track_timeout = 0.0    # Default= 0.0 Optional seconds to wait after track End (Avoid dual tracking)
event_timeout = 0.7    # Default= 0.7 seconds to wait for next motion event before starting new track
log_data_to_CSV = True # Default= True True= Save log data as CSV comma separated values

# Camera Settings
# ---------------
WEBCAM = True          # Default= False False=PiCamera True=USB WebCamera

# Web Camera Settings
WEBCAM_SRC = 0         # Default= 0   USB opencv connection number
WEBCAM_WIDTH = 960     # Default= 640 USB Webcam Image width
WEBCAM_HEIGHT = 720    # Default= 480 USB Webcam Image height

# Camera Image Settings
# ---------------------
image_font_size = 15   # Default = 15 Font text height in px for text on images
image_bigger = 1.5     # Default = 1.5 Resize saved speed image by value

# ---------------------------------------------- End of User Variables -----------------------------------------------------```

pageauc, 2018-06-03:
Plugin looks OK.  Are you saving data to a system SD card,  External HD,
Memory Card or Network Share?

I have been running speed-cam.py on stretch RPI3b+ and Jessie RPI2B quad
although not continuously.  My window single core RPI B runs continuously
with a 2.5 HD mounted
All these run OK.

Failure Sequence is

Recent Symlink creation fails.  This can be OS being non unix, disk error
issue or symlink already exists. The message says file already exists, but
files are named by date-HHMMSS format so I cannot imagine a duplicate
filename unless there is a clock problem and there is a file within the
same second.  I suppose I could add hundred of second to file naming but
did not think that would be needed.
Then shutil tries to copy file to recent folder when symlink fails.
Subsequently shutil fails with duplicate exists.  Sounds like maybe a very
fast capture with within same second or disk issue and confusing error
message, or clock problem causing duplicate file name..

If all else fails you could try a fresh install.  Let me know progress.

Claude ...

jschaftenaar, 2018-06-08:
After some fiddling with the config I found that increasing the track_timeout setting from 0.0 to 0.5 seems to make it stable. I can possibly shorten this timeout a bit and will test with it some more but for now this issue is solved.

Deuceman927, 2018-06-12:
I'm having a very similar issue.  I have not had the chance to run logging yet, but updating the track_timeout seems to have lengthened the time between service crashes.

pageauc, 2018-06-12:
I have run the code on various models of RPI for months at a time without
issue on NTFS, Unix Ext4 hard drives as well as on RPI SD card.  I suggest
you run speed-cam.py from console with verbose on.  or send logging to file
(can be run from background task on menubox.sh) You can send logging to
file per config.py config.py variables

verbose = True
loggingToFile = True

I suspect error is due to write error to disk.  Also an external USB
storage devices must be NTFS or Unix formats since there are issues with
symlinks on FAT filesystems
https://www.linuxquestions.org/questions/slackware-14/creation-of-symbolic-link-not-permitted-in-fat-mounted-drive-128751/

Send me summary of results of logging messages that occur around time of
crash.  I will take a look to determine problem.

Claude ...

pageauc, 2018-06-12:
I had a previous issue where the problem was a bad USB camera.  Testing showed it crashed other apps after a period of time when being used.  Resolution was to replace faulty hardware.
Claude ...

closed 2018-06-17 by pageauc