pageauc/speed-camera #27: Hashmarkds off Center

BigDuke6nc, 2018-06-11:
The hash marks are off-center from the calibration image. Do you know how I can adjust those so it would be easier to capture my image? They appear fine when using the pi camera but when using a USB webcam they are misaligned. I had a rather difficult time fine tuning the calibration of the pi camera and difficulty mounting the pi in the window so I went with USB camera instead. See image link for hash mark example. [https://imgur.com/a/rRLUbFY](https://imgur.com/a/rRLUbFY)

pageauc, 2018-06-11:
OK Thanks.  I have never calibrated with webcam since speed camera originally was written for the picamera only.  I will fix the problem and maybe add a setting in the calibration function to set the color of the hash lines to allow better contrast if required.  I am currently working on an option to put the speed data into a mysql database. 
Thanks for your feedback.
Claude ..

pageauc, 2018-06-11:
It was a pretty minor change since I already had an image_width variable and made some minor changes to the calibration function.  I tested with one of my web cams.  It is running on an old wheezy single core RPI with 500 MB Ram memory.  I ran it on 320x240 and it worked OK.  If you are having issues with response not being good enough then start at 320x240 res and work your way up.  The larger the opencv camera image the more processing has to be done.

You can upgrade using the menubox.sh UPGRADE menu pick.

Let me know results of your testing.
Regards Claude .

BigDuke6nc, 2018-06-11:
Hash mark issue has been fixed with this version. Thanks for the quick response.

closed 2018-06-12 by pageauc