pageauc/speed-camera #29: Distance from road

DataTables, 2018-06-20:
Really looking forward to setting this up - one question though if I may: in the wiki it says:

> Please note that if road is too close and/or vehicles are moving too quickly then the camera may not capture motion and/or record vehicle in speed photo.

Do you have any further details on how far away from the road the camera needs to be to get a reliable detection? I'll struggle to get it much further away than 10 meters unfortunately. Another option I thought of might be a fish eye lens, although I guess that might confuse the image processing.

pageauc, 2018-06-20:
With regards to your initial question re distance from camera.   You should be OK except for large buses/trucks/trailers, etc.  Too close and contours will not track accurately track since opencv contour may lock onto different locations.  Too far and contours become too small and get overpowered by things moving closer to the camera.  10 meters might be OK.  You could try mounting camera higher to gain more distance.

Claude ...

DataTables, 2018-06-20:
Awesome - thanks! The 10m was with the height... Interesting to know that you are pulling in the stream directly that should indeed help. I'll give it a whirl and let you know how I go. I'm fairly certain trucks and buses will fill the frame, so that might cause a problem.

I could place the camera at an angle to view "up the road" giving more road coverage, but would that just confuse matters?

closed 2018-07-25 by pageauc