pageauc/speed-camera #32: Duplicate File Error and SQL Error

BigDuke6nc, 2018-07-06:
Application is crashing when there is a duplicate track at the same time. Unable to write the record in the database and to the file system. Log link below.

[https://pastebin.com/33hWPpQd](https://pastebin.com/33hWPpQd)

BigDuke6nc, 2018-07-06:
No worries. I thought I had changed those values as well but when I looked they were default with the deployment. I can probably tweak those values.

pageauc, 2018-07-06:
I have added tenth of second to file naming.  I think this should be good enough but it could be easily changed to hundredths if needed.  This should not happen with normal reset times but this should solve your problem and not make naming too long.

I have tested bu Let me know result
Claude ...

BigDuke6nc, 2018-07-06:
Seems to have fixed the issue.

closed 2018-07-06 by BigDuke6nc