pageauc/speed-camera #34: Feature request: Support for RTSP IP cameras

rafik24, 2019-05-13:
Hi Claude,

Thank you for this great project.

I was wondering if you are planning to enhance the inputs to support RTSP IP cameras ?

Merci,
Rafik

pageauc, 2019-05-20:
I don not use ip cameras to not able to test.  Here is an article regarding how to use opencv to capture ip camera images.  Not sure if if works but if you try this and it works then let me know.  

https://stackoverflow.com/questions/49978705/access-ip-camera-in-python-opencv

rafik24, 2019-05-20:
Thanks Claude,
I will give it a try and come back to you.
Rafik

 
 

https://stackoverflow.com/questions/49978705/access-ip-camera-in-python-opencv

pageauc, 2019-05-20:
If you need help I can do some of the coding but you will have to test.
Regards Claude ...

pageauc, 2019-05-20:
You might try setting up speed camera for webcam  in config.py  

    WEBCAM = False

    WEBCAM_SRC = '"tsp://cam_ip:cam_port/1"
or
    WEBCAM_SRC = "rtsp://user:passwd@cam_ip:cam_port/1"

You will need to substitute user, passwdm cam_ip and cam_port to match your IP camera
Give this a try to see if you get errors.

Then try to run to see if it works.

closed 2019-09-26 by pageauc