pageauc/speed-camera #37: New bug :(

richiejarvis, 2019-07-02:
Hi Claude,

I've not investigated yet, but I note you are in the process of changing the time format.  Well, it breaks it :(

I've roll my Production instances to the previous version, but wanted to let you know ASAP, so sorry for the 'raw' bug report.

Cheers,

Richie

```pi@pi3:~/speed-camera$ ./speed-cam.py
Loading ...
----------------------------------------------------------------------
speed-cam.py 9.21   written by Claude Pageau
2019-07-02 16:54:32 INFO     <module>   Pi Camera Module is Enabled and Connected supported=1 detected=1
----------------------------------------------------------------------
Note: To Send Full Output to File Use command
python -u ./speed-cam.py | tee -a log.txt
Set log_data_to_file=True to Send speed_Data to CSV File speed-cam.log
----------------------------------------------------------------------

Debug Messages .. verbose=True  display_fps=False calibrate=False
                  show_out_range=True
Plugins ......... pluginEnable=False  pluginName=picam240
Calibration ..... cal_obj_px_L2R=60 px  cal_obj_mm_L2R=5182 mm  speed_conv_L2R=0.19320
                  cal_obj_px_R2L=75 px  cal_obj_mm_R2L=5182 mm  speed_conv_R2L=0.15456
                  (Change Settings in /home/pi/speed-camera/config.py)
Logging ......... Log_data_to_CSV=True  log_filename=speed-cam.csv (CSV format)
                  loggingToFile=False  logFilePath=speed-cam.log
                  SQLITE3 DB_PATH=/home/pi/speed-camera/data/speed_cam.db  DB_TABLE=speed
Speed Trigger ... Log only if max_speed_over > 0 mph
                  and track_counter >= 5 consecutive motion events
Exclude Events .. If  x_diff_min < 1 or x_diff_max > 20 px
                  If  y_upper < 75 or y_lower > 185 px
                  or  x_left < 5 or x_right > 315 px
                  If  max_speed_over < 0 mph
                  If  event_timeout > 0.30 seconds Start New Track
                  track_timeout=0.00 sec wait after Track Ends (avoid retrack of same object)
Speed Photo ..... Size=960x720 px  image_bigger=3.0  rotation=0  VFlip=False  HFlip=False
                  image_path=media/images  image_Prefix=speed-
                  image_font_size=12 px high  image_text_bottom=True
Motion Settings . Size=320x240 px  px_to_kph_L2R=0.310920  px_to_kph_R2L=0.248736 speed_units=mph
OpenCV Settings . MIN_AREA=100 sq-px  BLUR_SIZE=10  THRESHOLD_SENSITIVITY=20  CIRCLE_SIZE=5 px
                  WINDOW_BIGGER=1 gui_window_on=False (Display OpenCV Status Windows on GUI Desktop)
                  CAMERA_FRAMERATE=20 fps video stream speed
Sub-Directories . imageSubDirMaxHours=1 (0=off)  imageSubDirMaxFiles=1000 (0=off)
                  imageRecentDir=media/recent imageRecentMax=0 (0=off)
Disk Space  ..... Disabled - spaceTimerHrs=0  Manage Target Free Disk Space. Delete Oldest jpg Files
                  spaceTimerHrs=0 (0=Off) Target spaceFreeMB=500 (min=100 MB)

----------------------------------------------------------------------
2019-07-02 16:54:32 INFO     <module>   Initializing Pi Camera ....
2019-07-02 16:54:35 INFO     isSQLite3  Success: File is sqlite3 Format /home/pi/speed-camera/data/speed_cam.db
2019-07-02 16:54:35 INFO     db_check   Success: sqlite3 Connected to DB /home/pi/speed-camera/data/speed_cam.db
2019-07-02 16:54:35 INFO     speed_camera sqlite3 DB is Open /home/pi/speed-camera/data/speed_cam.db
2019-07-02 16:54:35 INFO     speed_notify Plugin Disabled per pluginEnable=False
2019-07-02 16:54:35 INFO     speed_notify Logging to Console per Variable verbose=True
2019-07-02 16:54:35 INFO     speed_notify Press ctrl-c in this terminal session to Quit
2019-07-02 16:54:35 INFO     speed_notify Begin Motion Tracking .....
2019-07-02 16:54:36 INFO     speed_camera Reset- event_timer 0.94>0.30 sec Exceeded
2019-07-02 16:54:36 INFO     speed_camera New  - 0/5 xy(40,29) Start New Track
2019-07-02 16:54:36 INFO     speed_camera  Add - 1/5 xy(50,30) 50.91 mph D=10/20 C=1 65x26=1081 sqpx L2R
2019-07-02 16:54:36 INFO     speed_camera  Add - 2/5 xy(62,33) 30.91 mph D=12/20 C=1 63x23=999 sqpx L2R
2019-07-02 16:54:36 INFO     speed_camera  Add - 3/5 xy(73,32) 62.54 mph D=11/20 C=1 62x25=1001 sqpx L2R
2019-07-02 16:54:36 INFO     speed_camera  Add - 4/5 xy(84,33) 31.28 mph D=11/20 C=1 61x24=966 sqpx L2R
2019-07-02 16:54:36 INFO     speed_camera  Add - 5/5 xy(94,33) 53.28 mph D=10/20 C=1 61x24=959 sqpx L2R
Traceback (most recent call last):
  File "./speed-cam.py", line 1467, in <module>
    speed_camera() # run main speed camera processing loop
  File "./speed-cam.py", line 1160, in speed_camera
    image_path, image_prefix)
  File "./speed-cam.py", line 700, in subDirChecks
    if subDirCheckMaxHrs(subDirPath, maxHours, prefix):
  File "./speed-cam.py", line 669, in subDirCheckMaxHrs
    dirDate = datetime.datetime.strptime(dirStr, "%Y-%m-%d-%H:%M")
  File "/usr/lib/python2.7/_strptime.py", line 332, in _strptime
    (data_string, format))
ValueError: time data '20190702-1643' does not match format '%Y-%m-%d-%H:%M'
```

pageauc, 2019-07-02:
This is used to name sub folders and did not previously have a problem  I checked my speed camera and it changed directories fine.  Will do some more testing to see if I can duplicate error.

richiejarvis, 2019-07-02:
Ahhhh - is it complaining about my old speedcam.db data then??

pageauc, 2019-07-02:
If you check under SubDirCheckMaxHars line 662 there is a Note to self on
line 665 about error checking.  That is because you can get into trouble if
you change from maxfiles to maxhours.  I notice you have
imagesubdirmaxhours = 1.  That means you want a new subdirectory created
every hour.  Not sure why.  Default is to create a new subdirectory every
1000 images.  I would recommend you set imageSubDirMaxHours = 0  and
imageSubDirMaxFiles= 1000 or so.
If you want images created by elapsed time period then start with a new
blank media/images folder and set imageSubDirMaxHours to a higher number
like every 24 hours.

Let me know what you were intending for subfolder creation.  The reason I
added subfolder creation by max files in folder or by elapsed time was to
allow long term operation and not have huge numbers of files in one
folder.  It makes for better response for web interface and easier
management.

Let me know how you do.  I will also do some checking.

Claude ...

richiejarvis, 2019-07-02:
Hi Claude,

Thank you.  I just came to report the same to you!   So we both found it.  Cool :)

I was planning to rotate dirs each day, that was all.  I'll carry on playing!

Btw - my third camera just went live in a different location in the village, and I can give you read-only dashboard access if you are curious to watch your creation in action :)

Cheers,

Richie

pageauc, 2019-07-02:
I would be interested in your dashboard.
Thanks Claude

richiejarvis, 2019-07-03:
Hi Claude,

Sure - here you go: https://45dabd8dc60e4165953b1028ccb443ae.europe-west1.gcp.cloud.es.io:9243/s/admin/app/kibana#/dashboard/f3088330-9c32-11e9-8c02-57fb7d58fcea?_g=(refreshInterval%3A(pause%3A!f%2Cvalue%3A300000)%2Ctime%3A(from%3A'2019-06-28T23%3A00%3A00.000Z'%2Cto%3Anow))

username: view
password: ChaileySpeedwatch

Cheers,

Richie

richiejarvis, 2019-07-03:
btw - i am in the process of transferring the data from my lab up, so please excuse the lack of data!

closed 2019-07-03 by richiejarvis

pageauc, 2019-07-03:
Thanks for the link to your project.  Keep up the good work.  I will check
occasionally to see how your data looks.
My camera has run for several years.  That is why I added sqlite3.  My Data
is not always vehicles, since it can pick up birds flying by, me cutting
the grass, pedestrians walking on our side of the street, etc.  I have
thought of adding post processing to have opencv recognize catalog
objects.  It would be nice to have a breakdown of vehicle, types like cars,
trucks, buses, motorcycle, bicycle, backhoes, garbage trucks Etc).
BTW are you processing night data?

Regards Claude ...

richiejarvis, 2019-07-03:
Hi Claude,

Thank you for the kind words.  Yes, we are processing night data at the
moment.  sqlite3 is great, and works really well.  There are 3 cameras
active in the village the moment, and I am building out 3 more at the
moment, so we should have reasonable coverage soon.

On the identification part, yes, I would love to do that.  I noted that the
webpage tries to guess at the moment, so I am thinking about storing that
info as an initial cut once I figure out where to get it from.  Also, It
turns out that a few folks round here are dev's, so we might get more bells
and whistles added in future.  Which makes me wonder, I don't really want
to fork your project and work on it without you, so what do you think about
adding more contributors?  I fully understand if you say no, and of course,
if that is the case you are welcome to work on the fork of course!

Look forward to hearing from you.

Cheers,

Richie

pageauc, 2019-07-03:
I accept github project push/merge requests from contributors after
reviewing changes to make sure they are OK.  Anyone can clone the project
and send updates to the project.
Some dev's have translated speed-cam to javascript.  One user rewrote speed
cam project in a more package cross platform format.  These were separate
forks.
Please note I have been retired for approx 13 years and just do this as a
hobby.  Speed cam started as a demo and gradually got more complex to
handle python2 and 3.  Use webcam, better logging and error checking Etc
Claude ...

Claude ...

pageauc, 2019-07-09:
Hi Richie

I have updated speed-cam.py to version 9.3.  This fixes some bugs and adds
some better error messaging.

Fixed Fatal error when imagesSubDirMaxHours is greater than 0.
Note Normally you would set subdir creation for max files or elapsed hours
but you could set both.  If you set by max hours and duration is short you
may may get empty folders.  I still recommend using max files to create
subfolders rather than elapsed time.

Also fixed DeleteOldFiles function to give better error message.  Sometimes
you might delete files in the media/images folder and break the symlink in
recent folder.  This creates an error when globbing the recent folder to
delete oldest symlink file since some symlinks do not point to a file.  The
fix is to delete all files in the media/recent folder.  This should fix the
problem and new recent symlinks will be created that correctly point to
existing image files.

Give this version a try and let me know if you have any issues.

You can upgrade by running the menubox.sh UPGRADE menu pick

Thanks for your feedback
Regards Claude ....

richiejarvis, 2019-07-09:
Thanks Claude - I'll give it a try on my new rpi's ;)

Loving the work - and you are teaching me python too - thanks!

Cheers,

Richie