pageauc/speed-camera #38: Please help

jasonsuthers, 2019-08-28:
Hi Claude

Thanks for this project, although ive been pulling my hair out all day

Fron GUI when i run 
pi@raspberrypi:~/speed-camera $ ./speed-cam.py

I get this following Error

Loading ...
----------------------------------------------------------------------
speed-cam.py 9.4   written by Claude Pageau
----------------------------------------------------------------------
Traceback (most recent call last):
  File "./speed-cam.py", line 216, in <module>
    if not os.path.exists(DB_DIR):  # Check if database directory exists
  File "/usr/lib/python2.7/genericpath.py", line 26, in exists
    os.stat(path)
TypeError: coercing to Unicode: need string or buffer, tuple found
pi@raspberrypi:~/speed-camera $ 

I have the following Dir
/home/pi/speed-camera/data/speed_cam.db
i also added 
/home/pi/speed-camera/database
not sure if it should be there but i seen ref to it i think will ive been looking through code

Any help would be greatly appreciated

Thank You
Jason

selbstmacher, 2019-08-29:
I've got the same issue. If running this in a vmware machine with debian 10 64Bit.
I already installed everything with user pi to ensure the paths are correct but it didn't help.
I think it has something to do with python, on debian 10 both 3.7 and 2.7 are installed, but I'm not used to python.

jasonsuthers, 2019-08-29:
I know why its causing the fault now after some reading , but i cant see were its actually broken
if i can crack it i will let you know

pageauc, 2019-08-29:
OK.  I found some extra comma's in the config.py variable file in the  #
Sqlite3 Settings approx line 132.  Not sure this is the problem
but it would appear this may be a problem.  You can download a new copy per
below or edit your existing config.py. Try this
and let me know.  Claude ...

    wget -O config.py
https://raw.github.com/pageauc/speed-camera/master/config.py

jasonsuthers, 2019-08-30:
Hi Claude, that was the problem its all working fine now i just edited the file, you wont belive how many times ive looked at the section today and couldnt see the problem, although i know a lot more about python now :)

Thank You and best regards
Jason

pageauc, 2019-08-30:
Please Note I have released ver 9.5 that does not hard code the db
directory but rather does a

DB_DIR_PATH = os.path.join(baseDir, DB_DIR)

You can upgrade to this ver 9.5 and edit config.py for and edit config.py
variable below
to be safe.

    DB_DIR = "data"

The wayward comma's were my problem and did not show up in testing since I
already had an active sqlite3 database path
Regards Claude ...

Let me know how you are doing.  I would be interested.
Regards Claude ...

jasonsuthers, 2019-09-01:
Hi Claude

Thanks im just working through bringing a IP Camera feed into the program, Ive managed with some success then managed to break it after making some changes im still learning python, 
When i have it running again i can share with you hopefully you can point me in the right dirrection

Regards
Jason

selbstmacher, 2019-09-02:
The fix also worked for me.
IP camera is working just fine.

pageauc, 2019-09-02:
Can you send me IP camera code changes or configuration you are using and
model of IP camera .
Great Appreciated
Claude ...

selbstmacher, 2019-09-02:
It worked pretty straight forward.
I just entered the rstp url that was displayed from the tool "ONVIF Device Manager".
It was important to enter the correct width and height.
Thats it, here is my configuration:
`
# Camera Settings
WEBCAM = True         # Default = False False=PiCamera True=USB WebCamera

# Web Camera Settings
WEBCAM_SRC = "rtsp://192.168.189.64/RtspTranslator.12/camera"         # Default= 0   USB opencv connection number
WEBCAM_WIDTH = 768     # Default= 320 USB Webcam Image width
WEBCAM_HEIGHT = 272    # Default= 240 USB Webcam Image height
`

pageauc, 2019-09-02:
Wow.  That is exactly what I have seen mentioned and what opencv also
mentions for receiving an IP cam stream.  Setting the camera
width and height are also important.  I will include this in the wiki and
as a comment in the config.py.  Maybe create a plugin.
I still don't have an IP cam since I have more than enough RPI's.

Thanks
Claude ...

jasonsuthers, 2019-09-03:
Wow i was over complicating matters on my end , works perfectly :)

Regards
Jason

selbstmacher, 2019-09-03:
I don't know if this should be like this:
[image]
but I would expect that the rectangle is there where the motion is.
Or is this like it should be?

pageauc, 2019-09-03:
Send me a copy of your config.py file and plugin if you are using one.
Thanks

selbstmacher, 2019-09-07:
Flipping had to be turned off.
Now everything works perfect.

closed 2019-09-26 by pageauc