pageauc/speed-camera #39: Show camera

ZIKO94ZIKO, 2019-09-22:
Hi sir , i hope you are doing well , 
firstly i want  thank you so much for your great project, it's very helpful.
i run your program its work well but the display of the camera does not show , i change the variables `gui_windows=True` ans the some problem i cant see the display of my camera. please i need your help

ZIKO94ZIKO, 2019-09-22:
`'gui_window_on':True,
    'show_thresh_on':True,
    'show_crop_on':True,
    'verbose':True,`

`if gui_window_on:
            cv2.imshow('Difference Image',differenceimage)
            image2 = speed_image_add_lines(image2, cvRed)
            image_view = cv2.resize(image2, (image_width, image_height))
            cv2.imshow('Movement (q Quits)', image_view)
            if show_thresh_on:
                cv2.imshow('Threshold', differenceimage)
            if show_crop_on:
                cv2.imshow('Crop Area', image_crop)
            # Close Window if q pressed
            if cv2.waitKey(1) & 0xFF == ord('q'):
                cv2.destroyAllWindows()
                logging.info("End Motion Tracking ......")
                vs.stop()
                still_scanning = False
        if display_fps:   # Optionally show motion image processing loop fps
            fps_time, frame_count = get_fps(fps_time, frame_count)
`

pageauc, 2019-09-23:
You need to set the variables in the config.py file.  Seems like you are setting them in the default_settings dictionary variable.  The dictionary variables are just used as a backup in case the config.py is missing or some variables are not in the config.py file.  I added this just as a backup not intending that users would modify the source code.  So change the config.py file and you should be OK as long as you run from a RPI GUI desktop and a monitor attached.
Let me know how you make out.
Claude ...

ZIKO94ZIKO, 2019-09-23:
Hello my dear sir 
Thank you so much for your reply,  it's work very well hh , You are a true genius .. i have a question Sir :  how can be integrate source code about Real-time license plate recognition with 'openalpr' in your project , i find this project in GitHub i try to run openalpr_camera.py  it's work ,  [https://gist.github.com/jkjung-avt/790a1410b91c170187f8dbdb8cc698c8](url) .
So  i want to detect the speed and LPR of vehicle  in real time using the  caméras IP

ZIKO94ZIKO, 2019-09-24:
@pageauc

closed 2019-09-26 by pageauc