pageauc/speed-camera #41: How can be calculate the speed

zongoalbert, 2019-09-30:
Hello Mr. Pageeac
 Thank you for sharing your Great project .My end of study project topic, is almost the same project like you did,  it's about the vehicle speed detection at real time using Camera IP in a highway , I took a look in your code , I can't understand how you calculate the speed  ,and how can you know the real distance from ( fps and distance between the camera and object ) ?
Also i see you  calculated the speed from  L2R or R2L .  It's possible to calculate the speed from front to behind F2B and from behind to front B2F?

pageauc, 2019-09-30:
I you have any code improvements send me a github push and I will review
Thanks Claude ....

zongoalbert, 2019-09-30:
Hii sir thank you for your feedback ,  :  i  relied on this PDF for the calculation of the speed "" https://pdfs.semanticscholar.org/efb3/7b45f4c9241c7d4ddf1295450dba6cf5cd06.pdf to calculate the speed""   

you can find the code  as below  : 
 `import cv2
import dlib
import time
import threading
import math

#carCascade = cv2.CascadeClassifier('myhaar.xml')
video = cv2.VideoCapture('0')

WIDTH = 1280
HEIGHT = 720

# speed = K * d_pixels / time      // ( time = 1 / fps )    between 2 frames 

def estimateSpeed(x1, x2): 
	d_pixels = math.sqrt(math.pow(x2[0] - x1[0], 2) + math.pow(x2[1] - x1[1], 2))

	K = 15.13 # K is coeficient in relation with FOV and angle of view , u can find how to calculate the speed in this link 	
			#https://pdfs.semanticscholar.org/efb3/7b45f4c9241c7d4ddf1295450dba6cf5cd06.pdf

	d_meters = d_pixels * K 
	#print("d_pixels=" + str(d_pixels), "d_meters=" + str(d_meters))
	fps = 18
	speed = d_meters * fps * 3.6
	return speed
	

def trackMultipleObjects():
	rectangleColor = (0, 255, 0)
	frameCounter = 0
	currentCarID = 0
	fps = 0
	
	carTracker = {}
	carNumbers = {}
	carx1 = {}
	carx2 = {}
	speed = [None] * 1000
	
	# Write output to video file
	out = cv2.VideoWriter('outpy.avi',cv2.VideoWriter_fourcc('M','J','P','G'), 10, (WIDTH,HEIGHT))

	while True:
		start_time = time.time()
		rc, image = video.read()
		if type(image) == type(None):
			break
		
		image = cv2.resize(image, (WIDTH, HEIGHT))
		resultImage = image.copy()
		
		frameCounter = frameCounter + 1
		
		carIDtoDelete = []

		for carID in carTracker.keys():
			trackingQuality = carTracker[carID].update(image)
			
			if trackingQuality < 7:
				carIDtoDelete.append(carID)
				
		for carID in carIDtoDelete:
			print ('Removing carID ' + str(carID) + ' from list of trackers.')
			print ('Removing carID ' + str(carID) + ' previous x.')
			print ('Removing carID ' + str(carID) + ' current x.')
			carTracker.pop(carID, None)
			carx1.pop(carID, None)
			carx2.pop(carID, None)
		
		if not (frameCounter % 10):
			gray = cv2.cvtColor(image, cv2.COLOR_BGR2GRAY)
			cars = carCascade.detectMultiScale(gray, 1.1, 13, 18, (24, 24))
			
			for (_x, _y, _w, _h) in cars:
				x = int(_x)
				y = int(_y)
				w = int(_w)
				h = int(_h)
			
				x_bar = x + 0.5 * w
				y_bar = y + 0.5 * h
				
				matchCarID = None
			
				for carID in carTracker.keys():
					trackedPosition = carTracker[carID].get_position()
					
					t_x = int(trackedPosition.left())
					t_y = int(trackedPosition.top())
					t_w = int(trackedPosition.width())
					t_h = int(trackedPosition.height())
					
					t_x_bar = t_x + 0.5 * t_w
					t_y_bar = t_y + 0.5 * t_h
				
					if ((t_x <= x_bar <= (t_x + t_w)) and (t_y <= y_bar <= (t_y + t_h)) and (x <= t_x_bar <= (x + w)) and (y <= t_y_bar <= (y + h))):
						matchCarID = carID
				
				if matchCarID is None:
					print ('Creating new tracker ' + str(currentCarID))
					
					tracker = dlib.correlation_tracker()
					tracker.start_track(image, dlib.rectangle(x, y, x + w, y + h))
					
					carTracker[currentCarID] = tracker
					carx1[currentCarID] = [x, y, w, h]

					currentCarID = currentCarID + 1
		
		#cv2.line(resultImage,(0,480),(1280,480),(255,0,0),5)

		for carID in carTracker.keys():
			trackedPosition = carTracker[carID].get_position()
					
			t_x = int(trackedPosition.left())
			t_y = int(trackedPosition.top())
			t_w = int(trackedPosition.width())
			t_h = int(trackedPosition.height())
			
			cv2.rectangle(resultImage, (t_x, t_y), (t_x + t_w, t_y + t_h), rectangleColor, 4)
			
			# speed estimation
			carx2[carID] = [t_x, t_y, t_w, t_h]
		
		end_time = time.time()
		
		if not (end_time == start_time):
			fps = 1.0/(end_time - start_time)
		
		#cv2.putText(resultImage, 'FPS: ' + str(int(fps)), (620, 30),cv2.FONT_HERSHEY_SIMPLEX, 0.75, (0, 0, 255), 2)

		for i in carx1.keys():	
			if frameCounter % 1 == 0:
				[x1, y1, w1, h1] = carx1[i]
				[x2, y2, w2, h2] = carx2[i]
		
				# print 'previous x: ' + str(carx1[i]) + ', current x: ' + str(carx2[i])
				carx1[i] = [x2, y2, w2, h2]
		
				# print 'new previous x: ' + str(carx1[i])
				if [x1, y1, w1, h1] != [x2, y2, w2, h2]:
					if (speed[i] == None or speed[i] == 0) and y1 >= 275 and y1 <= 285:
						speed[i] = estimateSpeed([x1, y1, w1, h1], [x2, y2, w2, h2])

					#if y1 > 275 and y1 < 285:
					if speed[i] != None and y1 >= 180:
						cv2.putText(resultImage, str(int(speed[i])) + " km/hr", (int(x1 + w1/2), int(y1-5)),cv2.FONT_HERSHEY_SIMPLEX, 0.75, (255, 255, 255), 2)
					
					#print ('CarID ' + str(i) + ': speed is ' + str("%.2f" % round(speed[i], 0)) + ' km/h.\n')

					#else:
					#	cv2.putText(resultImage, "Far Object", (int(x1 + w1/2), int(y1)),cv2.FONT_HERSHEY_SIMPLEX, 0.5, (255, 255, 255), 2)

						#print ('CarID ' + str(i) + ' x1: ' + str(carx1[i]) + ' x2: ' + str(carx2[i]) + ' speed is ' + str("%.2f" % round(speed[i], 0)) + ' km/h.\n')
		cv2.imshow('result', resultImage)
		# Write the frame into the file 'output.avi'
		#out.write(resultImage)

		if cv2.waitKey(33) == 27:
			break
	
	cv2.destroyAllWindows()

if __name__ == '__main__':
	trackMultipleObjects()`

pageauc, 2019-09-30:
Yes I have seen this code before.  Performance on a RPI is an issue.  Also
installing dlib on a raspberry pi can be a challenge.

zongoalbert, 2019-09-30:
Yes Sir, im using my laptop not RPI  ,i tested it but the value of speed not true , as you see in this picture 
[image]

like you see the value of speed indicated is 28 Kps but the real value is 54 Kps

zongoalbert, 2019-09-30:
Yes Sir, im using my laptop not RPI ,i tested it but the value of speed not true , as you see in this picture: 
[image]
like you see the value of speed indicated is 28 Kps but the real value is 54 Kps , Could explain how can i play with value of parameters to get the estimation speed. Thanks

pageauc, 2019-09-30:
Speed camera is designed to work with camera pointing at 90 degrees to
road.  It will not work accurately with the angle you are showing.  Motion
tracking will trigger and image can be used for ALPR but speed will not be
useful
Claude ...

closed 2019-10-02 by pageauc