pageauc/speed-camera #42: speed

ZIKO94ZIKO, 2019-10-02:
Hi sir i saw this  question https://github.com/pageauc/speed-camera/issues/41#issue-500179767 im interested to know its possible to  change angle of view and fps  to detect the real speed of the object .

ZIKO94ZIKO, 2019-10-02:
i mean its possible to detect the speed +ALPR of object without pointing the camera at 90 degrees to
road. for example fixing  the camera at 45 degrees and detect ALPR and speed booth of theme.

ZIKO94ZIKO, 2019-10-03:
HI sir , i hope you are very well , please tell me how can be integrate change degree of the camera , im very interested to know.

pageauc, 2019-10-03:
You could do some trial and error to see if it can be done accurately.  The
other option is to use some sort of distance range sensor suitable for the
distance then read and determining speed by time and distance changes from
sensor.  Not sure if sonar sensor has enough range.  laser ranging might be
an option but more expensive.  https://www.dfrobot.com/category-55.html

ZIKO94ZIKO, 2019-10-04:
Hello Sir thank you, and please excuse me for the inconveniences. The idea that can be worked on is detecting the speed + alpr by camera drone in highway , this will allow to exceed the capabilities of the normal radar and allow to show the maximum possible number of cars, and can be limited the speed for example : activate max_speed_over=120 kps ، you can find an article as below explain how can be calculate the speed estimation in real time using camera with parametre specific: field of view, angle of view, fps, resolution and altitude between the camera and read : https://www.google.com/url?sa=t&source=web&rct=j&url=https://pdfs.semanticscholar.org/efb3/7b45f4c9241c7d4ddf1295450dba6cf5cd06.pdf&ved=2ahUKEwj7hKDcsIHlAhU7RBUIHaONCHsQFjAEegQIBhAC&usg=AOvVaw2_mZKX6dFfu8_KuCmnUe64&cshid=1570150182587

pageauc, 2019-10-04:
Are you planning on doing this project?

ZIKO94ZIKO, 2019-10-04:
No sir hhhhh ,I got the idea when i see the question 41.

ZIKO94ZIKO, 2019-10-04:
I got the idea when i see the question 41. but i  Think of doing this starting from your project ,  and change firstly some lines about speed That empowers us detecting the speed of the car which is in front of camera or located of 45 degree of the camera .

pageauc, 2019-10-04:
Good luck with your project.  I would be interested in your progress
Regards
Claude ...

ZIKO94ZIKO, 2019-10-04:
Thank you so much Sir , i'm so glad to have meet you ^^

ZIKO94ZIKO, 2019-10-07:
HI Sir, i wish a good start of the week i edited your code from my needs , and i tested by  video input `WEBCAM_SRC = '3.mp4' ` its work  but i cant find the images in data base , i think should be add some lines for linking data base , could you please a quick look in the code?  Thank's
[image]

ZIKO94ZIKO, 2019-10-07:
Fyi the value of speed of course not true hhh its just a test.

pageauc, 2019-10-07:
where is your code?

ZIKO94ZIKO, 2019-10-07:
Hi sir , Please find attached a file containing code :  https://drive.google.com/drive/folders/1s9bbCHeqMNaPQ7XH7GLWsv6oTZQVKw3J

ZIKO94ZIKO, 2019-10-07:
please sir , if you check it give me an feedback.

ZIKO94ZIKO, 2019-10-08:
Hi Sir please , give just a explanation : why the photos does not  saving in folder images. Thanks

pageauc, 2019-10-16:
Did not test this but see how this works

ZIKO94ZIKO, 2019-10-16:
Hello Sir , i have  already seen your videos ,thank you for your answer

ZIKO94ZIKO, 2019-10-17:
the problem is just how to integrate in your code Field of view( FOV ) , and angle of the camera and height (between the camera and earth)  to detect the the speed in deference position , then we will on a right to choose the position of camera ,which gives us detection ALPR and the speed booth of them.

ZIKO94ZIKO, 2019-10-19:
Hi Mr.Claude , Thank you for the last version of config.py ,it's work very well , please give me some advises about this stuff , i tried since one mouth to add this variables but doesn't succeed.

pageauc, 2019-10-19:
The previous variables were auto calculated and the previous config.py
settings were ignored.  This latest change
checks to see if the variables exist (imported from config.py)  If they
exist then the will override the automatic
calculation for motion area.  See code for details.

ZIKO94ZIKO, 2019-10-19:
yees Sir  ,i already understand-ed the variables put it in config.py and the last modification config.py.new , i mean by **this stuff** (the variable in this question `https://github.com/pageauc/speed-camera/issues/42#issuecomment-543153619` )

ZIKO94ZIKO, 2019-10-19:
> the problem is just how to integrate in your code Field of view( FOV ) , and angle of the camera and height (between the camera and earth) to detect the the speed in deference position , then we [...]
yees Sir ,i already understand-ed the variables put it in config.py and the last modification config.py.new , i mean by **this stuff** (the variable in this question)

ZIKO94ZIKO, 2019-10-19:
**this question** : the problem is just how to integrate in your code Field of view( FOV ) , and angle of the camera and height (between the camera and earth) to detect the the speed in deference position , then we will on a right to choose the position of camera ,which gives us detection ALPR and the speed booth of them.

pageauc, 2019-10-19:
I have your earlier code using DLIB but installing dlib on an earlier RPI
is a pain and not sure if it can handle image processing fast enough.  Non
RPI may be OK
At any rate I am thinking a second DLIB version will be required.  I am
busy right now and this is not a high priority for me so you may want to
work on the dlib version yourself and create a github repo for the new code
or give me a push for current repo.
Claude ...

closed 2019-10-19 by pageauc