pageauc/speed-camera #43: Cannot change SPEED_MPH to MPH

jeffreypkelly, 2019-10-18:
Hi,
Can you help me? I would like speeds in MPH
I have set SPEED_MPH = True, but I still get speeds in kph.
Is there something else I have to set?

TY!
Jeff

pageauc, 2019-10-18:
Did you stop and restart speed camera.  This can be done using menubox.sh.
When settings are changed you must restart the app.
Claude ...

jeffreypkelly, 2019-10-19:
One more question, please.
I cannot find the y_upper, y_lower, x_left, x_right settings in the config.py file.
I would like to reduce the height of the box.
Are they located in another file?

pageauc, 2019-10-19:
Due to some users wanting to use IP camera's and the hassle of putting
values in plugins,  I now auto detect image resolution (mainly for IP
camera streams) and auto calculate the motion tracking cropped image box.
This code can be found at approx line 1568 of latest release of
speed-cam.py.   I can add a feature to override this auto calculation
but it would be up to the user to supply the correct values.  The current
version just takes the stream image width and height and sets
a fraction of the w or h to set the motion box borders.  Separate values
for images GT width 1000 px.  Otherwise you could play around with the code
x_scale and y_scale
values as appropriate or just override values by putting new values below
auto calculation.  This however would have to be redone after an upgrade
(big hassle)

Suggested code is below That would override x or y settings if the
variables are found in the config.py file otherwise auto calculation would
set the values
This would allow setting x or y separately and have the other auto
calculate or set both and have no auto calculate

try:
    x_left
   x_right
except NameError:
    x_left = int(img_width / x_scale)
    x_right = int(img_width - x_left)

try:
   y_upper
   y_lower
except NameError:
    y_upper = int(img_height / y_scale)
    y_lower = int(img_height - y_upper)

  If you really need this override feature let me know.
Claude ...

pageauc, 2019-10-19:
I have release speed-cam.py ver 10.0 including new config.py with comments
for overriding motion tracking crop detection area
on images.  It will be the users responsibility to make sure custom
values are within image size.

You can upgrade using menubox.sh  UPGRADE menu option.  The updated
config.py will be named config.py.new
See instructions at end of upgrade regarding backup and copy of new
config.py

Please test and let me know if everything is OK

Claude ...

jeffreypkelly, 2019-10-19:
Worked perfect!
TY!
Jeff

closed 2019-10-19 by pageauc