pageauc/speed-camera #44: vehicles not detected - timeout exceeded

pagaille, 2019-10-19:
Hi,

I don't get it. Everything looks working correctly but the vehicles looks not being detected.

The logs says most of the time something like this :

`2019-10-19 14:54:34 INFO     speed_notify Begin Motion Tracking .....
2019-10-19 15:02:30 INFO     speed_camera Reset- event_timer 476.63>0.60 sec Exceeded
2019-10-19 15:02:30 INFO     speed_camera New  - 0/10 xy(70,45) Start New Track
2019-10-19 15:04:19 INFO     speed_camera Reset- event_timer 109.05>0.60 sec Exceeded
2019-10-19 15:04:19 INFO     speed_camera New  - 0/10 xy(74,37) Start New Track
2019-10-19 15:07:27 INFO     speed_camera Reset- event_timer 187.96>0.60 sec Exceeded
2019-10-19 15:07:27 INFO     speed_camera New  - 0/10 xy(62,43) Start New Track
2019-10-19 15:07:28 INFO     speed_camera Reset- event_timer 0.70>0.60 sec Exceeded
2019-10-19 15:07:28 INFO     speed_camera New  - 0/10 xy(63,43) Start New Track
^C
`

The only time I saw the system actually measuring something was when a pedestrian passed in the field 😂 

This is the picture I'm capturing. NB : Cars displayed are parked.

[image]

What should I do ? Am I too close ?

Txs

Matthieu

pageauc, 2019-10-19:
You should not be too close but I think your camera resolution is too big
and therefore is slowing opencv motion detection and causing a timeout.
The faster the objects moves the faster opencv has to track and process
image stream.  In your case the camera is very close and the moving
objects will travel very quickly through frame and the processing won't be
fast enough..

I suggest you change resolution to something like 320x240  (default) or
640x480 and see results.  Note because these are small resolutions. There
is *image_bigger =* a setting
that will increase size of saved image.  There is another one for
opencv *WINDOW_BIGGER
= *display window resizing .

Also I recommend you NOT try to view opencv image in real time. Set
*gui_window_on =
False* This will speed up image processing  You can view saved images in
the web browser interface.
I normal turn off the GUI desktop and run speed-cam.py from SSH session

Also I have added a feature in release 10.0 that allows over riding the
automatic calculation of motion crop area   If a larger image resolution is
used then a smaller motion crop area can be set that can speed image
processing.

Let me know how you manage.
Claude ...

pagaille, 2019-10-19:
First thank you very much for your extensive answer !

I’m running at 320x200 (default settings) but I enabled the GUI to better grasp what was happening. I did that by forwarding X11 through ssh, which is probably harder for the cpu. The screen capture is done on a Retina display that’s why it looks high res.

I’ll try again without the display and by fine tuning the system using the tips you gave me here.

May I suggest to include this tips in the readme ? 

Also : I tough, by reading the calibration procedure, that y_upper and y_lower were configurable from the setup.py file. It looks like it is not the case.

—> I’d like to be able to move that detection window : in my case moving the camera is not easy and the detection window doesn’t cover the whole road.
—> I guess the doc should be corrected in the current situation.

Thanks, I’ll report back.

Matthieu G.  (en mode mobile)

pageauc, 2019-10-19:
Thanks for your reply.  Camera resolution should not be a problem.  Note,
You can change the motion tracking area on Release 10.0 of speed-cam.py.
Upgrade on a RPI using menubox.sh UPGRADE menu pick.
The latest config.py file will be called config.py.new.  See instructions
at end of UPGRADE to copy config.py.new to config.py.  The new config.py
will have a section with rem'd out settings that can over ride the x_left,
x_right, y_upper and y_lower variables so they don't get auto calculated.
This will allow you to customize the motion tracking window.

Please answer some questions
Model of Raspberry PI or computer that you are running speed-cam.py on
Version of operating system Eg Raspbian Buster.
Type of camera

Send me a copy of the config.py that you use for testing below.
Send me a copy of log file (not too long) that you are testing with
verbose is True by default.  You can enable sending log to file with
config.py variable *loggingToFile = True*
optionally you can change the file log name.  Keep the log fairly short but
log enough to show enough object activity.

Thanks Claude ..

pageauc, 2019-10-20:
Mathieu
Was thinking image appears to be tool big and wrong aspect ratio to be
320x240.
Check config.py and set

*pluginEnable = False*

Not sure if you are using webcam or picam but make sure it is set to
320x240.  You should bet a squarer image 4/3 aspect ratio image.
*image_bigger
= 3.0* will resize the saved image by a factor of 3 but this can be changed
if you want smaller or larger saved image.

See the new 10.0 config.py and you can override the default motion tracking
area per comments for setting* x_left, x_right, y_upper, y_lower*

  Restart speed-cam.py

Test and let me know results per previous email
Thanks
Claude ...

pagaille, 2019-10-23:
Hi Claude, tes again for your help.

>Model of Raspberry PI or computer that you are running speed-cam.py on
Pi3B+
>Version of operating system Eg Raspbian Buster.
Yep, Buster, updated.
>Type of camera
Good question. That's a subminiature pi camera 'spy camera' that I had on the shelf. It is based on the OV5647. It's field of view is "54 x 41 degrees"
I bought it here : https://shop.mchobby.be/fr/pi-camera/1117-camera-raspberry-pi-espion-spy-5-megapixel-3232100011175.html
And it's probably that same model : https://www.arducam.com/product/1-4-5-noir-mpspy-camera-module-sensor-flex-cable-raspberry-pi/

Regarding the resolution and aspect ratio, I'm not sure. It could be linked to the X11 ssh forward and my mac display. 

I tested again with version 10 and disabled the UI. I basically get the same results : slow objects are correctly detected and tracked, I get the 5 calibration pictures of a pedestrian passing by.
But I only get one picture or two of cars passing in front of my house. It looks like the fps isn't good enough. What could I do ?

pagaille, 2019-10-23:
This is typically the result I get : 

[image]

- Contour is wrong
- Only one picture recorded in calibration mode
- speed is wrong (and somewhat random)

pagaille, 2019-10-23:
I tried one more try with my car. Neighbours start wondering what I'm doing :-)

I cruised at 24kph.

I got three pictures : 

[image]
[image]
[image]
[image]

And here is the corresponding log : 

```
----------------------------------------------------------------------
2019-10-23 10:38:22 INFO     speed_camera New  - 0/5 xy(164,61) Start New Track
2019-10-23 10:38:22 INFO     speed_camera  Add - 1/5 xy(179,61) 35.97 kph D=15/20 C=6 39x85=1717 sqpx L2R
2019-10-23 10:38:22 INFO     speed_camera  Add - 2/5 xy(186,60) 22.00 kph D=7/20 C=6 40x85=1802 sqpx L2R
2019-10-23 10:38:22 INFO     speed_camera  Add - 3/5 xy(193,59) 10.70 kph D=7/20 C=6 41x85=1834 sqpx L2R
2019-10-23 10:38:22 INFO     speed_camera  Add - 4/5 xy(200,58) 40.50 kph D=7/20 C=5 42x86=1915 sqpx L2R
2019-10-23 10:38:54 INFO     speed_camera Reset- event_timer 31.95>0.30 sec Exceeded
2019-10-23 10:38:54 INFO     speed_camera New  - 0/5 xy(138,13) Start New Track
2019-10-23 10:38:54 INFO     speed_camera  Add - 1/5 xy(130,13) 23.66 kph D=8/20 C=2 116x72=3544 sqpx R2L
2019-10-23 10:38:54 INFO     speed_camera  Add - 2/5 xy(120,12) 29.97 kph D=10/20 C=3 101x73=3367 sqpx R2L
2019-10-23 10:38:54 INFO     speed_camera  Add - 3/5 xy(111,12) 13.38 kph D=9/20 C=5 95x72=3059 sqpx R2L
2019-10-23 10:38:54 INFO     speed_camera  Add - 4/5 xy(102,12) 25.60 kph D=9/20 C=3 100x72=3333 sqpx R2L
2019-10-23 10:38:54 INFO     speed_camera  Add - 5/5 xy(93,11) 26.68 kph D=9/20 C=3 96x73=3312 sqpx R2L
----------------------------- Create Calibration Image -----------------------------

  Instructions for using media/images/calib-20191023-1038547.jpg image for camera calibration

  Note: If there is only one lane then L2R and R2L settings will be the same
  1 - Use L2R and R2L with Same Size Reference Object, Eg. same vehicle for both directions.
  2 - For objects moving L2R Record cal_obj_px_L2R Value Using Red y_upper Hash Marks at every 10 px  Current Setting is 150 px
  3 - Record cal_obj_mm_L2R of object. This is Actual length in mm of object above Current Setting is 4408 mm
      If Recorded Speed 23.9 kph is Too Low, Increasing cal_obj_mm_L2R to Adjust or Visa-Versa
Repeat Calibration with same object moving R2L and update config.py R2L variables
cal_obj_mm_R2L and cal_obj_px_R2L accordingly
  4 - Edit /home/pi/speed-camera/config.py File and Change Values for the Above Variables.
  5 - Do a Speed Test to Confirm/Tune Settings.  You May Need to Repeat.
  6 - When Calibration is Finished, Set config.py Variable  calibrate = False
      Then Restart speed-cam.py and monitor activity.

  WARNING: It is Advised to Use 320x240 Stream for Best Performance.
           Higher Resolutions Need More OpenCV Processing
           and May Reduce Data Accuracy and Reliability.

  Calibration Image Saved To /home/pi/speed-camera/media/images/calib-20191023-1038547.jpg  
  View Calibration Image in Web Browser (Ensure webserver.py is started)

---------------------- Press cntl-c to Quit Calibration Mode -----------------------
2019-10-23 10:38:54 INFO     speed_camera  Saved media/images/calib-20191023-1038547.jpg
2019-10-23 10:38:54 INFO     isSQLite3  Success: File is sqlite3 Format /home/pi/speed-camera/data/speed_cam.db
2019-10-23 10:38:54 INFO     db_check   Success: sqlite3 Connected to DB /home/pi/speed-camera/data/speed_cam.db
2019-10-23 10:38:54 INFO     speed_camera  SQL - Update sqlite3 Data in /home/pi/speed-camera/data/speed_cam.db
2019-10-23 10:38:54 INFO     speed_camera End  - R2L Ave Speed 23.9 kph Tracked 45 px in 0.212 sec Calib 150px 4408mm
----------------------------------------------------------------------
2019-10-23 10:38:54 INFO     speed_camera New  - 0/5 xy(84,11) Start New Track
2019-10-23 10:38:54 INFO     speed_camera  Add - 1/5 xy(75,10) 49.80 kph D=9/20 C=3 98x73=3334 sqpx R2L
2019-10-23 10:38:54 INFO     speed_camera  Add - 2/5 xy(66,11) 27.48 kph D=9/20 C=3 101x74=3365 sqpx R2L
2019-10-23 10:38:54 INFO     speed_camera  Add - 3/5 xy(56,10) 15.41 kph D=10/20 C=3 105x75=3203 sqpx R2L
2019-10-23 10:38:54 INFO     speed_camera  Add - 4/5 xy(47,11) 26.80 kph D=9/20 C=3 105x72=2919 sqpx R2L
2019-10-23 10:38:55 INFO     speed_camera  Add - 5/5 xy(37,10) 29.74 kph D=10/20 C=3 111x75=2915 sqpx R2L
----------------------------- Create Calibration Image -----------------------------

  Instructions for using media/images/calib-20191023-1038550.jpg image for camera calibration

  Note: If there is only one lane then L2R and R2L settings will be the same
  1 - Use L2R and R2L with Same Size Reference Object, Eg. same vehicle for both directions.
  2 - For objects moving L2R Record cal_obj_px_L2R Value Using Red y_upper Hash Marks at every 10 px  Current Setting is 150 px
[... 46 lines cut ...]
           Higher Resolutions Need More OpenCV Processing
           and May Reduce Data Accuracy and Reliability.

  Calibration Image Saved To /home/pi/speed-camera/media/images/calib-20191023-1038552.jpg  
  View Calibration Image in Web Browser (Ensure webserver.py is started)

---------------------- Press cntl-c to Quit Calibration Mode -----------------------
2019-10-23 10:38:55 INFO     speed_camera  Saved media/images/calib-20191023-1038552.jpg
2019-10-23 10:38:55 INFO     isSQLite3  Success: File is sqlite3 Format /home/pi/speed-camera/data/speed_cam.db
2019-10-23 10:38:55 INFO     db_check   Success: sqlite3 Connected to DB /home/pi/speed-camera/data/speed_cam.db
[... cut ...]
2019-10-23 10:38:55 INFO     speed_camera  Add - 3/5 xy(46,13) 15.27 kph D=10/20 C=3 49x70=1521 sqpx R2L
2019-10-23 10:38:55 INFO     speed_camera  Add - 4/5 xy(55,17) 25.90 kph D=9/20 C=6 30x46=787 sqpx L2R
2019-10-23 10:38:55 INFO     speed_camera  Add - 5/5 xy(46,16) 26.50 kph D=9/20 C=4 33x47=907 sqpx R2L
----------------------------- Create Calibration Image -----------------------------

  Instructions for using media/images/calib-20191023-1038555.jpg image for camera calibration

  No
```

pageauc, 2019-10-23:
Output looks OK.  You can try increasing camera FPS for camera to maybe 30
or event slightly more.  Opencv needs processing time depending on the
model of RPI so there is a maximum where there is no advantage to
increasing any more.   Due to the camera being so close you are getting
variations in speed due to contours being picked up at different locations
due to lighting, reflections etc.  If object is farther away the object
appears smaller and these variations are smaller.   You also get variations
since camera pixels at edges of image are farther away than ones in middle
of view.  This also can cause variations in calculations.

Increase camera framerate .to 30 or 35  (config.py)

You could also try increasing track_counter from 5 to slightly higher
(test)  This is also in config.py under motion event settings

Also still not sure of camera resolution.  What does it say on settings
page when you run from console with verbose on.  I am thinking camera is
not same as RPI camera and not compatible with RPI firmware and not setting
resolution per pi-camera python library.

Claude ...

pageauc, 2019-10-23:
Ok camera is OK and is at 320x240 per hash marks on cal image. Claude ..

AndreasW79, 2020-04-29:
Hello Claude,
Here, very different speeds are calculated, do you have an idea why this could be?

[image]
[image]
[image]
[image]

Thanks in advance

closed 2021-06-26 by pageauc