pageauc/speed-camera #46: blend  project

zongoalbert, 2019-11-07:
HI Mr Pageauc,  
it's possible to blend your project with this project : [https://www.youtube.com/watch?v=rw2O3VIx7Wg](url)   , its developed by C++ , the program is : classification of vehicles and detect speed , and draw area of detect manualy with mouse+ automatic calibration .
[image]

pageauc, 2019-11-07:
I am not a developer. I have been retired for 13 years and just code as a hobby.  My Code is open source so anyone is welcome to fork and develop further.
Regards Claude ..

closed 2019-11-07 by pageauc

zongoalbert, 2019-11-13:
Hi Mr,Claude 
Actually I work on a car detection project at real time, when I launch the camera I want to make a screenshot (.png) for any car and save it in a path, which means  i need a function in opencv that does the screenshot of just a car detected (not all windows).

pageauc, 2019-11-14:
Paths of all images are stored in a sql database and can be accessed with a
query.  See the alpr-speed-speed.py for code that accesses the images and
processes them for alpr detection.  A similar method can be used for your
requirements.  Also the contour information is saved along with the image
path and other data that is in the csv optional file.
Claude ...

ZIKO94ZIKO, 2019-11-19:
Yes Sir , but the problem is  want do just screen shoot for the object detectet not all window , because if the camera detect    for example 3  cars in one photo.png ,it's difficult to extract ALPR for all cars ,

pageauc, 2019-11-19:
When you say windows do you mean opencv contours?.  openalpr can detect
multiple license plates that are in a photo.
What problem are you trying to solve?

ZIKO94ZIKO, 2019-11-19:
i mean it's possible to  do a screenshoott of one by one vehicle  , and extract the  alpr for each vehicle , then save it in data base , like the question in this Link : https://stackoverflow.com/questions/58834717/extract-object-and-save-it-as-an-image-png-using-c

pageauc, 2019-11-19:
The explanation only references pseudo code and not actual code.  Also
speed camera just detects largest moving contour that may not and often
does not encompass complete object or vehicle.. Often only a part of object
is covered by the contour. All the suggestion is saying is crop contour
from existing image but does not go into details of detecting complete
contour of object.  I would suggest you investigate coding this on your
own.  Then you will appreciate what you are requesting.
Regards Claude ...