pageauc/speed-camera #47: font colour / white on black bar

i5okie, 2019-12-04:
Hi,

Is it possible to change the font colour, or add a black bar under the text?

i5okie, 2019-12-04:
i just ended up changing cvWhite to cvBlue on the printout.
But otherwise, this tracking model is a bit convoluted. I'm much closer to the road than you had in the demo video on youtube. so its not picking up the whole vehicle.
[image]

It never detects the whole vehicle, it's always a wheel, a shadow, or some small part of the vehicle, so specifying size of the object when calibrating is difficult. unless I mount a ball to the roof of my car or something. I don't know. Im not sure how to calibrate this in my case.

i5okie, 2019-12-04:
Those vertical lines there in calibration pics... can I specify what their spacing is in mm?

pageauc, 2019-12-05:
text is added using cv2.puttext per line 1292 of speed-cam.py.  Yest the text color can be changed.  Currently it is hard coded to cvWhite.  This could be changed to another variable name that can be assigned as a config.py setting to control the font colour.
I you are interested you might want to implement this change and send me a Github Push for the change.  You could also manually change the setting but unfortunately an update would cancel the change.
Claude ...

pageauc, 2019-12-05:
I have update config.py and speed-cam.py to add settings per following

```
image_font_scale = 0.5        # Default= 0.5 Font scale factor that is multiplied by the font-specific base size.
image_font_thickness = 2      # Default= 2  Font text thickness in px for text on images
image_font_color = (255, 255, 255)  # Default= (255, 255, 255) White
```

You can do an menubox.sh UPGRADE then copy the config.py.new per install script instructions. or just add the variables above to your existing config.py.  This will preserve your other settings.

Let me know if this works for you
Claude ...

pageauc, 2019-12-05:
Calibration is just a matter of measuring the pixels per millimeter.  I thought the easiest way was to use the vehicle as the ruler and measure pixels with hash.  This avoids safety issue of someone standing in the street with a tape measure or ruler.  You could also use any object of known length at the distance to be calibrated for.   You could also use a part of the vehicle like distance between door handles , size of door between seams, glass, using calibration hash marks.

i5okie, 2019-12-05:
where do i specify what the real-life measurement is between the hashmarks?

pageauc, 2019-12-05:
This is a link to the wiki. To Calibrate *cal_obj_px* and *cal_obj_mm*
variables perform the following

https://github.com/pageauc/speed-camera/wiki/Calibrate-Camera-for-Distance

i5okie, 2019-12-06:
Ah.
I was just confused with the going and looking at how big cars are etc etc.
This makes a bit more sense.
Thanks!

pageauc, 2019-12-06:
Please note, You can calibrate for left to right and right to left since on
a roadway the distances will be slightly different. otherwise you
can just make both the same if the distance is a long way off, you are on a
one way street, Etc
Claude ...

i5okie, 2019-12-06:
How did you implement that?
So are you assuming that left to right is the distance between hash marks closest to left edge of the screen, and right to left the hash marks closest to right edge and you average log() their distances towards the center?

pageauc, 2019-12-06:
One is for Object (vehicle) moving in the near road lane (closest to you)
the other is the far lane where traffic would be in the opposite direction.
Hope this explains things.
Claude ...

i5okie, 2019-12-06:
yep that makes perfect sense now. thank you!

closed 2021-06-26 by pageauc