pageauc/speed-camera #51: Updated Install guidance

bmcgowan32, 2020-02-05:
I'm having some troubles with motion tracking performance using a Pi 3B+ and latest version of Raspian Buster.  Even at meager performance settings of only 240 pixels (30FPS, with 5 required tracks) I have vehicles scooting through my motion tracking widow without being photoed.  I even tried reducing the motion tracking window to approx 30% of the full camera area.

I'm trying to determine if its a software issue, or if its time for an upgrade to a Pi4.  Most of the install steps for the speed-camera are for Pi3 or older with Jesse.  I'm wondering if there are adjusted install steps for Buster and/or Pi4?  Is the manual build of OpenCV still required?   SO far, I've been running a current Buster image with "apt update" and "apt full-upgrade", but no other package customization. 

Any suggestions for setup from scratch on newer hardware/software would be appreciated.

bmcgowan32, 2020-02-05:
I should note that I have a picam v2 for image capture

bmcgowan32, 2020-02-06:
Thanks Claude!  This is really helpful.  I am going to re-install my Pi3B+ with the lite install of Buster.  I had been using the full install, but then doing all the work via SSH without GUI.  It would certainly trim the install footprint to jettison the desktop packages altogether as they are still running and consuming resources.  I also think that the full Buster install came out of the box with Python3 and OpenCV 4.x.  Starting with the lite image and adding the components as you've suggested will give me better control over the variables when I am tweaking for performance.

I am also interested in tinkering with the Pi4 for additional horsepower.  Do you see any value in using OpenCV 4 vs 3 if using a Pi 4?  Or would you still recommend a barebones install and using OpenCV 3?

pageauc, 2020-02-06:
opencv4 can be faster for certain functions but speed camera uses only a
few of opencv features.  Having opencv with neon enabled can improve
performance.  RPI4 will also help due to cpu speed increase but adding
extra RAM memory may not improve performance unless you are running other
apps as well, like ALPR.

I think lite is a good choice for a console only dedicated app.  Also try
to use a faster SD card if you can.  I just bought three Samsung evo 64 GB
SD cards on Amazon for $12 each.  Free shipping
https://www.amazon.com/gp/product/B06XX29S9Q/ref=ppx_yo_dt_b_asin_title_o01_s01?ie=UTF8&psc=1
Note there are other choices.  I do not have any association with any
vendors.

Let me know how you make out.
Claude ...

bmcgowan32, 2020-02-20:
Thanks for all the advice so far, its been really helpful as I try to narrow down and tweak my install.  I have more thoughts and any suggestions you might have would be super helpful.  Since my last post:

I picked up a Pi4 B and did a fresh install with the lite version of Buster.  No GUI components are installed, I am doing all the setup and control via SSH.  Did only an "apt update" and "apt full-upgrade" after image install, then ran the install script from your github as a normal user.  I am using Python 2.7.16 and OpenCV 3.2.0:

```
pi@raspberrypi:~/speed-camera $ which python
/usr/bin/python
pi@raspberrypi:~/speed-camera $ python --version
Python 2.7.16
pi@raspberrypi:~/speed-camera $ python
Python 2.7.16 (default, Oct 10 2019, 22:02:15) 
[GCC 8.3.0] on linux2
Type "help", "copyright", "credits" or "license" for more information.
>>> import cv2
>>> cv2.__version__
'3.2.0'
>>> 
```
(Would you recommend upgrading to python 3 or openCV 4?  Not sure if there would be a significant need to move beyond what the install script sets up?)

Im sad to say that a new device and fresh install didn't seem to affect my situation significantly.  I've tried moving the pi/camera up and further away from the street, as well as testing at resolutions of 320 (no plugin), 480, and 720.  Lower resolutions do not seem to yield better or more consistent results... Ive tried reducing the crop area to just the most relevant parts of the image to lighten the motion capture processing load, but that doesn't seem to do much either.  Ive tried upping the frames per second (anything above 30 seems to be of no use, even with picam v2).  I have also tried playing with the delay between tracks to give the pi more time to "rest" :-).  But I am still frequently observing vehicles passing directly through the observation zone with 0 recorded tracks when watching the console logs.  Is there anything I can enable to debug why no motion capture happens when an object moves through the frame?  Is it normal that some objects simply dont trigger a track at all?

I have been working under the assumption that I needed to increase image processing performance (or reduce image processing load) to get consistent motion capture of all objects moving through the observation area.  However, I still have vehicles which never trigger a track, or which don't trigger the min of 5 tracks to create an image, or which track successfully but are not fully capturing the contours of the entire object, leading to inconsistent measurement.  Im Including a few sample calibration shots as well to see if you have any suggestions for camera placement, resolution, etc.

Thanks again so much.  I appreciate that you are generous enough to take time to answer questions so fully and quickly.  Hopefully this thread will also provide others with helpful steps for tuning and troubleshooting!
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

jabella, 2021-02-17:
This describes my experience exactly.  I'm running pi4 + picam v2 and have same results.  I definitely seem to have better results at lower resolutions though.

pageauc, 2021-02-17:
Check to see if the gui desktop is running.  This may be the reason for
slower performance.  Run raspi-config and change setting to run terminal
Client rather than desktop.

closed 2021-06-26 by pageauc