pageauc/speed-camera #52: Noob, getting opencv error in IDLE

matchucaligani, 2020-02-21:
[image]
I'm absolutely brand new to coding in general. Trying out your script and getting this error in IDLE.

matchucaligani, 2020-02-21:
[image]
I've also tried it in Command Prompt

matchucaligani, 2020-02-21:
Any ideas?

pageauc, 2020-02-22:
You will need to download and install opencv for python 3 on windows.  Also
you will need to have your path setup to point to python folder

You can find details for installing opencv here
https://www.learnopencv.com/install-opencv3-on-windows/
Hope this helps

closed 2021-04-24 by pageauc