pageauc/speed-camera #54: RTSP stream issues

toto99303, 2020-05-03:
Hello,

First of all thank you for writing and sharing this software with all of us.
I installed it today on my pi4 and I'm using my Hikvision camera RTSP stream.
However sometimes the process stops with this error:

2020-05-03 21:15:57 INFO     speed_camera New  - 0/10 xy(75,24) Start New Track
2020-05-03 21:15:57 INFO     speed_camera  Add - 1/10 xy(63,0) 48.94 kph D=12/20 C=5 110x75=1492 sqpx R2L
2020-05-03 21:15:57 INFO     speed_camera  Out - 1/10 xy(62,0) Min D=0<=1px C=3 73x75=1797 sqpx R2L
2020-05-03 21:15:57 INFO     speed_camera  Out - 1/10 xy(62,0) Min D=0<=1px C=1 72x75=1705 sqpx R2L
2020-05-03 21:15:57 INFO     speed_camera  Add - 2/10 xy(59,35) 1.37 kph D=3/20 C=2 25x40=242 sqpx R2L
2020-05-03 21:15:57 INFO     speed_camera  Add - 3/10 xy(61,39) 10.12 kph D=2/20 C=2 21x36=177 sqpx L2R
2020-05-03 21:16:04 INFO     speed_camera Reset- event_timer 7.00>0.30 sec Exceeded
2020-05-03 21:16:04 INFO     speed_camera New  - 0/10 xy(61,0) Start New Track
2020-05-03 21:16:04 INFO     speed_camera Reset- event_timer 0.36>0.30 sec Exceeded
2020-05-03 21:16:04 INFO     speed_camera New  - 0/10 xy(61,0) Start New Track
[h264 @ 0x28b5900] error while decoding MB 37 20, bytestream -5
[h264 @ 0x2a146f0] error while decoding MB 27 20, bytestream -13
2020-05-03 21:16:24 INFO     speed_camera Reset- event_timer 19.23>0.30 sec Exceeded
2020-05-03 21:16:24 INFO     speed_camera New  - 0/10 xy(137,38) Start New Track
2020-05-03 21:16:24 INFO     speed_camera  Out - 0/10 xy(195,0) Max D=58>=20px C=2 62x75=1816 sqpx L2R
2020-05-03 21:16:24 INFO     speed_camera New  - 0/10 xy(196,0) Start New Track
2020-05-03 21:16:24 INFO     speed_camera  Out - 0/10 xy(53,0) Max D=143>=20px C=1 63x75=1008 sqpx R2L
2020-05-03 21:16:24 INFO     speed_camera New  - 0/10 xy(54,0) Start New Track
Traceback (most recent call last):
  File "./speed-cam.py", line 1607, in <module>
    speed_camera() # run main speed camera processing loop
  File "./speed-cam.py", line 1145, in speed_camera
    grayimage1, contours = speed_get_contours(image2, grayimage1)
  File "./speed-cam.py", line 1006, in speed_get_contours
    image_crop = image[y_upper:y_lower, x_left:x_right]
TypeError: 'NoneType' object has no attribute '__getitem__'

I'm not sure what the problem is, can you help me?

pageauc, 2020-05-03:
Problem is that camera or ip connection from camera may have hiccuped.  The
program did not get any data. Program does trap image problems in the
def speed_get_contours(image, grayimage1): function per

    while not image_ok:
>         image = vs.read() # Read image data from video steam thread
> instance
>         # crop image to motion tracking area only
>         try:
>             image_crop = image[y_upper:y_lower, x_left:x_right]
>             image_ok = True
>         except ValueError:
>             logging.error("image Stream Image is Not Complete. Cannot
> Crop. Retry.")
>             image_ok = False
>

Looks like image was non existent so the ValueError did not trap error
since the exception was  NoneType.  Value error would trap a damaged or
partial image.

If you want to try your hand at programming you could try adding

except NoneType:
     add exception coding here

or add check for none before try

if image is None:

You can add a timer to the loop for a specified timeout period like maybe
60 seconds and then error out.  That way if the IP camera looses connection
or something else and does
not come back the program will not run forever.  This is common in
communications issues.   I do not have an IP camera so it might be best if
you could work
on problem

 This coding should not be too difficult.  I can help if you like but I see
it as a good learning opportunity.

Let me know how you make out.
Claude ...

pageauc, 2020-05-03:
Try adding the TypeError per code below

        try:
            image_crop = image[y_upper:y_lower, x_left:x_right]
            image_ok = True
        except (ValueError, TypeError):
            logging.error("image Stream Image is Not Complete. Cannot Crop.
Retry.")
            image_ok = False

Just need to modify line 1008
Would also be a good idea to add a timeout so does not loop forever.

Claude ...

pageauc, 2020-05-03:
FYI.   I have added code to github repo.  speed-cam.py ver 10.04  In case
you need programming hints or
you are not up for a challenge.

Please test and let me know results.
Thanks Claude ...

toto99303, 2020-05-03:
Wow, that was fast! Will give it a try and report back.

toto99303, 2020-05-04:
So I tested it and it still shuts down the process for some reason.
I know my camera is running over Wifi so I did a 3-4 hour ping test logging and it seems like it's loosing connection randomly for 10-20 seconds. This is probably the issue here, but I've tried extending the script timeout to 30 minutes and even if the camera is back the process does not continue. I'd probably add a script to restart the service every 5 min or so.... Let me know what you think.

pageauc, 2020-05-04:
Did you get an error from python.  Run speed-cam.py from SSH console and
make sure config.py verbose = True is set
You will see the logging data on the SSH session.  Not sure if we
trapped required errors with the exception.  It is also possible that once
communication is lost that it can no longer hand shake so maybe
communication would need to be reset with opencv somehow.  Anyway would be
interested in seeing if there is a python error generated
I would think 60 to 300 second timeout would be more than enough time delay.

Thanks for testing.  My setups use RPI camera.
Claude ...

toto99303, 2020-05-05:
Hello, no error from python... I think it can't re-handshake after it's lost connection.
But I managed to fix the connectivity issues by turning off some features on my camera, so it's now offloaded and responds better. Thanks again for your support!

pageauc, 2020-05-05:
I will leave the code as is since  logging message is clearer.  Also if you
setup watch-app.sh  the if PID not found then speed camera will auto re
start and hopefully
>
> reconnect if camera is back online.  Note there are variables in the
> watch-app.sh script that need to be set. * watch_app_on=true (no spaces)*

Claude ...

toto99303, 2020-05-05:
Awesome, this is exactly what I needed. I did put all three in /etc/rc.local for auto start:

su pi -c "/home/pi/speed-camera/speed-cam.sh start > /dev/null"
sleep 2
su pi -c "/home/pi/speed-camera/webserver.sh start > /dev/null"
sleep 2
su pi -c "/home/pi/speed-camera/watch-app.sh > /dev/null"

Let's see how it goes in 24 hours.

closed 2020-11-03 by pageauc