pageauc/speed-camera #55: Multiple detection - different speed

AndreasW79, 2020-05-04:
Hello Claude,
I have a strange problem, some vehicles are measured twice, at different measuring points as it seems. Unfortunately, the speeds differ again and again and I do not know exactly what I can still check.
Maybe a setting in the motion or open cv settings?

```
SPEED_MPH = False  
track_counter = 10
MIN_AREA = 150
show_out_range = True
x_diff_max = 30   
x_diff_min = 1  
x_buf_adjust = 10
track_timeout = 0.0  
event_timeout = 0.3
max_speed_over = 0 

BLUR_SIZE = 10
THRESHOLD_SENSITIVITY = 20    
```

[image]
[image]

Thank you in advance
Andreas

pageauc, 2020-05-04:
Seems like contours jump from front to back tire sometimes thus generating
large value.  If you are running RPI4 try increasing

track_counter = 15 or possibly 20

This will increase the number of motion detection cycles before calculating
final speed. It may still
be possible that contour can jump.

  I could also add a check to exclude unusually high speed values.

 With close large objects the largest contour can change from one end to
another.

Claude ...

AndreasW79, 2020-05-05:
Hi Claude, thanks for the reply.
I changed track_counter to 15 - i have to order another rpi4 for upgrade this device.

Heres another 'anomly' but i think the bike is to slow for good results.

[image]
[image]
[image]

andrewm659, 2020-06-09:
So I am experiencing the same thing.  How would I change it to start from the center instead of the corners to determine speed?  Did you try that at all?  Did it come up with skewed results?  My issue is I have a tree branch that is being detected.  I wish there was a better way to say only objects of a certain type.

Thanks!

andrewm659, 2020-08-25:
So I am still waiting to see how this goes.  We aren't getting a lot of vehicle traffic.  But it is constantly picking up the tall grass blowing in the wind.  Is there a way to make it only look for larger objects?  Or would that take some more development?

pageauc, 2020-08-26:
Should not be picking up grass moving since it uses motion tracking but is
possible depending on the motion sequence.   In config.py there is a Motion
Setting  MIN_AREA  = 100  This is 100 square pixels so quite a small area.
You can increase this but not too big since it may prevent tracking of some
objects.  Do some trial and error to get a setting that works best.

andrewm659, 2020-08-26:
So this is what it is doing.
[image]

pageauc, 2020-08-26:
red area rectangle is motion tracking detection area.  You should change
the location of the detection area to road area.
This can be done in config.py using  variables below.  Note you will need
to uncomment these in config.py

# Allow user to customize the motion crop area (width) x values
# If variables not found then values will be set automatically base on
image size.
 x_left = 200       # uncomment and change values to override auto calculate
 x_right = 650     # uncomment and change values to override auto calculate

# Allow user to customize the motion crop area (height) y values
# If variables not found then values will be set automatically base on
image size.
 y_upper = 160      # uncomment and change values to override auto calculate
 y_lower = 310      # uncomment and change values to override auto calculate

Depending on your resolution you can adjust values. to position red
detection rectangle on the screen and you can
also reposition the camera to point it correctly.

Normally this is calculated automatically so not sure why your detection
area is located where it is.
If you have problems send me copy of your config.py

Regards Claude ...

andrewm659, 2020-08-26:
Trying your recommendations.

closed 2021-07-02 by pageauc