pageauc/speed-camera #56: Running Speed Camera in background

sishort, 2020-06-15:
Sorry if this is a really stupid user issue to raise, but I have been trying to get Speed Camera running on a raspberry pi that I connect to through SSH.
It works great and is logging all of the data as it should whilst I'm connected through SSH, however as soon as I disconnect, both the speed-cam.py and webserber.py stop working. 
Is this a really obvious thing that I'm missing?
I'd love this to be running independently on a long term basis, but can't really do that if it only functions when logged in through SSH.

sishort, 2020-06-15:
Not sure if this is a recommended solution, but I have this continuing to run after the ssh session ends by using the Screen utility.
Fingers crossed it works and I've just learnt something new.

pageauc, 2020-06-15:
Run the speed camera and webserver from menubox.sh menu system.  Make sure both are not running in ssh session.  Run menubox.sh from ssh session per

`./menubox.sh`

From the menu you can start, stop or check current status of speed camera or webserver.py.   They will be started as background tasks so there is no need to stay logged in.

You can also run the programs in the background using bash scripts per below.  This similar to what menubox.sh runs from the menu picks.  

`
./speed-cam.sh start
./webserver.sh start`

See shell script output for details.  You can also use stop parameter

See the speed camera github wiki for more details like auto starting on boot etc.
https://github.com/pageauc/speed-camera/wiki

sishort, 2020-06-15:
Thanks Claude. Great project. Very useful.

Appreciate the help.

closed 2020-06-15 by pageauc