pageauc/speed-camera #57: macOS Install

flashheartthe4th, 2020-07-06:
Hello,

Is it possible to install this on macOS? I've followed the manual instal instructions, yet get multiple failures when trying to run menubox.sh. I'm running 10.15.5, and before posting all of the debug output, I thought I'd check to see whether it was actually supported.

Thanks

Paul

pageauc, 2020-07-06:
menubox.sh is a bash script that uses whiptail.  depending on the error
message I suspect if mac supports bash scripts then
then most likely you do not have whiptail installed. You can google for
solution.
also see
https://medium.com/@tamirsuliman/getting-started-building-linux-bash-graphical-ui-scripts-447cee97c167
This was one of the first google picks
Let me know how you make out

flashheartthe4th, 2020-07-10:
Its all working now, but the calibration is really vexing me. I've got joggers doing 200+kph.

The measurement window is quite small (3 meters / 160px) are there any other calibration settings you can point me in the direction of? I've reviewed https://github.com/pageauc/speed-camera/wiki/Calibrate-Camera-for-Distance but can't seem to get the speed values to come into line. Image of the crop area attached.

[image]

closed 2021-07-02 by pageauc