pageauc/speed-camera #58: speed-cam.py reboots RaspberryPi 3 B+

will-code-for-pizza, 2020-08-01:
Hi,

on a fresh install of speed-cam with OpenCV 4.4.0 on a RaspberryPi 3 B+, launching "speed-cam.py" results in a reboot of RaspberryPi.

speed-cam.log is emtpy, but I was able to determine, that it happened after the OpenCV windows were drawn.

Do you know this issue ?

Regards
Will

pageauc, 2020-08-01:
I have used opencv 4 without issue but not 4.4.0.  Does problem occur
without drawing opencv windows?
Also are you running watch-app.sh since there is a reboot option for it if
it finds speed-camera not running.  Should not be the case.
Try running speed camera in terminal session without config.py
gui_window_on= False  and verbose = True.

Watch log entries and let me know if there is a error or last command run
Claude ...

will-code-for-pizza, 2020-08-02:
Hi Claude,
I was able to isolate the problem.

It's NOT related to OpenCV 3/4, but to the options
show_thresh_on
show_crop_on

If I set one or both options to 'True', my Raspi crashes and reboots.

Leaving both options to 'False' it seems to be stable now and I was able to monitor the gui_window via ssh X11 forwarding.

Unfortunately there are no log entries after a crash. It seems, that Raspi did write too slow while a crash is initiated.
If I get more informations, I will let you know.

Thanks so far.
Regards
Will

will-code-for-pizza, 2020-08-02:
I was wrong...

The two mentioned options only speed up the crash.
Setting these options to 'False', my Raspi takes about ~ 30 seconds more to crash.

Here is the log from console:

$ ./speed-cam.py 
----------------------------------------------------------------------
speed-cam.py 10.04  written by Claude Pageau
Motion Track Largest Moving Object and Calculate Speed per Calibration.
----------------------------------------------------------------------
Loading  Wait ...
2020-08-02 11:53:10 INFO     <module>   Pi Camera Module is Enabled and Connected supported=1 detected=1
2020-08-02 11:53:13 INFO     <module>   Initializing Pi Camera ....
----------------------------------------------------------------------
Note: To Send Full Output to File Use command
python -u ./speed-cam.py | tee -a log.txt
Set log_data_to_file=True to Send speed_Data to CSV File speed-cam.log
----------------------------------------------------------------------

Debug Messages .. verbose=True  display_fps=True calibrate=True
                  show_out_range=True
Plugins ......... pluginEnable=False  pluginName=picam240
Calibration ..... cal_obj_px_L2R=80 px  cal_obj_mm_L2R=4700 mm  speed_conv_L2R=0.21150
                  cal_obj_px_R2L=85 px  cal_obj_mm_R2L=4700 mm  speed_conv_R2L=0.19906
                  (Change Settings in /home/pi/speed-camera/config.py)
Logging ......... Log_data_to_CSV=False  log_filename=speed-cam.csv (CSV format)
                  loggingToFile=False  logFilePath=speed-cam.log
                  SQLITE3 DB_PATH=/home/pi/speed-camera/data/speed_cam.db  DB_TABLE=speed
Speed Trigger ... Log only if max_speed_over > 0 kph
                  and track_counter >= 5 consecutive motion events
Exclude Events .. If  x_diff_min < 1 or x_diff_max > 20 px
                  If  y_upper < 60 or y_lower > 180 px
                  or  x_left < 40 or x_right > 280 px
                  If  max_speed_over < 0 kph
                  If  event_timeout > 0.30 seconds Start New Track
                  track_timeout=0.00 sec wait after Track Ends (avoid retrack of same object)
Speed Photo ..... Size=960x720 px  image_bigger=3.0  rotation=0  VFlip=True  HFlip=True 
                  image_path=media/images  image_Prefix=speed-
                  image_font_size=12 px high  image_text_bottom=True
Motion Settings . Size=320x240 px  px_to_kph_L2R=0.211500  px_to_kph_R2L=0.199059 speed_units=kph
                  CAM_LOCATION= None
OpenCV Settings . MIN_AREA=100 sq-px  BLUR_SIZE=10  THRESHOLD_SENSITIVITY=20  CIRCLE_SIZE=5 px
                  WINDOW_BIGGER=1 gui_window_on=True (Display OpenCV Status Windows on GUI Desktop)
                  CAMERA_FRAMERATE=20 fps video stream speed
Sub-Directories . imageSubDirMaxHours=0 (0=off)  imageSubDirMaxFiles=1000 (0=off)
                  imageRecentDir=media/recent imageRecentMax=100 (0=off)
Disk Space  ..... Disabled - spaceTimerHrs=0  Manage Target Free Disk Space. Delete Oldest jpg Files
                  spaceTimerHrs=0 (0=Off) Target spaceFreeMB=500 (min=100 MB)

----------------------------------------------------------------------
2020-08-02 11:53:15 INFO     isSQLite3  Success: File is sqlite3 Format /home/pi/speed-camera/data/speed_cam.db
2020-08-02 11:53:15 INFO     db_check   Success: sqlite3 Connected to DB /home/pi/speed-camera/data/speed_cam.db
2020-08-02 11:53:15 INFO     speed_camera sqlite3 DB is Open /home/pi/speed-camera/data/speed_cam.db
2020-08-02 11:53:15 INFO     speed_notify Plugin Disabled per pluginEnable=False
2020-08-02 11:53:15 INFO     speed_notify Logging to Console per Variable verbose=True
2020-08-02 11:53:15 INFO     speed_notify Press lower case q on OpenCV GUI Window to Quit program
2020-08-02 11:53:15 INFO     speed_notify         or ctrl-c in this terminal session to Quit
2020-08-02 11:53:15 WARNING  speed_notify IMPORTANT: Camera Is In Calibration Mode ....
2020-08-02 11:53:15 INFO     speed_notify Begin Motion Tracking .....
client_loop: send disconnect: Broken pipe`
`

Regards

will-code-for-pizza, 2020-08-02:
It seems to be an issue ith motion tracking:

2020-08-02 12:14:21 INFO     speed_notify Begin Motion Tracking .....
2020-08-02 12:14:48 INFO     speed_camera Reset- event_timer 26.83>0.30 sec Exceeded
2020-08-02 12:14:48 INFO     speed_camera New  - 0/5 xy(46,0) Start New Track
--> Crash

will-code-for-pizza, 2020-08-02:
Testing my SD-card with 'agnostics' tool shows 

`
Raspberry Pi Diagnostics - version (unknown)
Sun Aug  2 12:45:56 2020

Test : SD Card Speed Test
Run 1
prepare-file;0;0;10556;20
seq-write;0;0;14160;27
rand-4k-write;0;0;855;213
rand-4k-read;3534;883;0;0
Sequential write speed 14160 KB/sec (target 10000) - PASS
Random write speed 213 IOPS (target 500) - FAIL
Random read speed 883 IOPS (target 1500) - FAIL
Run 2
prepare-file;0;0;10609;20
seq-write;0;0;12089;23
rand-4k-write;0;0;767;191
rand-4k-read;3742;935;0;0
Sequential write speed 12089 KB/sec (target 10000) - PASS
Random write speed 191 IOPS (target 500) - FAIL
Random read speed 935 IOPS (target 1500) - FAIL
Run 3
prepare-file;0;0;11487;22
seq-write;0;0;12845;25
rand-4k-write;0;0;865;216
rand-4k-read;3829;957;0;0
Sequential write speed 12845 KB/sec (target 10000) - PASS
Random write speed 216 IOPS (target 500) - FAIL
Random read speed 957 IOPS (target 1500) - FAIL
Test FAIL
`
Could probably depend on my SD Card, which is an "Intenso Micro SD HC I" Class 10 32 GB

will-code-for-pizza, 2020-08-02:
For further investigation I would like you to collect some SD card performance tests for me on your hardware:

With 'hdparm':
`
$ sudo hdparm -t /dev/mmcblk0
/dev/mmcblk0:
 Timing buffered disk reads:  62 MB in  3.07 seconds =  20.21 MB/sec

$ sudo hdparm -T /dev/mmcblk0
/dev/mmcblk0:
 Timing cached reads:   1366 MB in  2.00 seconds = 683.40 MB/sec
`

With 'dd':
`
$ dd if=/dev/zero of=/tmp/test bs=1048576 count=500
500+0 Datensätze ein
500+0 Datensätze aus
524288000 bytes (524 MB, 500 MiB) copied, 28,6949 s, 18,3 MB/s

pi@speedcam:~ $ mv /tmp/test  ~/

pi@speedcam:~ $ dd if=/home/pi/test of=/dev/null bs=1048576
500+0 Datensätze ein
500+0 Datensätze aus
524288000 bytes (524 MB, 500 MiB) copied, 0,681278 s, 770 MB/s
`
Many thanks, Claude.

pageauc, 2020-08-02:
There are a few bare exceptions in the code that will block python
trackback. The camera code runs in a thread and broken pipe
indicates communications with camera thread was lost.

Does the problem occur under python 2 and/or 3.  The default is to run
under python env
Can you manually start speed-cam.py from ssh session under python 2 and
also try with python3
eg

python2 ./speed-cam.py

also

python3 ./speed-cam.py

also send a copy of your config.py

How did you install opencv 4.4.0 ?

Will try to replicate the problem.  I think communication with camera
thread is failing after it is initialized

Claude ...

will-code-for-pizza, 2020-08-02:
Hi Claude,

> How did you install opencv 4.4.0 ?

I used https://github.com/pageauc/opencv3-setup and changed the version to 4.4.0 in cv3-install-menu.conf.

Probably it is an undervoltage issue. I am using an 5V = 1A power supply.
Now I changed to 5V = 2,5A and [my config]https://pastebin.com/VYT2NA06
runs stable for serveral minutes.

will-code-for-pizza, 2020-08-02:
Seems to be solved by using a better power supply.

closed 2020-08-02 by will-code-for-pizza