pageauc/speed-camera #59: gnuplot dependency missing

will-code-for-pizza, 2020-08-02:
Hi Claude.
Small issue... Reports only could be generated after

$ mkdir -p ~/speed-camera/media/reports
$ touch ~/speed-camera/speed-cam.csv
$ sudo apt-get install python-gnuplot

Cheers
Bjoern

pageauc, 2020-08-02:
gnuplot is installed with speed-install.sh
per

sudo apt-get install -yq gnuplot
if [ $? -ne 0 ]; then
   sudo apt-get install -yq Gnuplot
fi

sudo apt-get install -yq python-gnuplot
if [ $? -ne 0 ]; then
   sudo apt-get install -yq python-Gnuplot
fi

Also the folders should be created automatically
Not sure what is causing your issues

Did you run via the curl install method or did you clone repo.

Claude ...

will-code-for-pizza, 2020-08-02:
Hi Claude,
I used the curl install method on newest Raspberry Pi OS
https://downloads.raspberrypi.org/raspios_lite_armhf_latest

Regards
Will

pageauc, 2020-08-02:
thanks.   glad you resolved your power supply issue.  Good
troubleshooting.  I will keep this issue in mind and update wiki

Sounds like things are working for you now.
Regards Claude ...

will-code-for-pizza, 2020-08-03:
Many thanks for this great support and project.
We suffer extremely from vehicles with excessive speeds and your project will help me to document this.
Regards
Will

closed 2020-08-03 by will-code-for-pizza