pageauc/speed-camera #61: Feature Request: (Automatic Number Plate Recognition) ANPR , additional video source.

jaimzj, 2020-08-09:
Hi Claude,

This is a very cool project you have done. I would like to add a feature request if you are able to someday implement it 👍 

As the primary video feed is facing the side view of vehicles. the feature request is to add an additional video source (for front view) where the second video source can be of a camera placed facing the front of the vehicles movement.

And every-time a picture is stored from the primary video, a picture could be stored from the front-view camera feed as well.

This can then be enhanced further to have (Automatic Number Plate Recognition) ANPR capabilities as well.

pageauc, 2020-08-09:
I previously thought of adding a multi camera feature (two or more RPI's)
to my pi-timolo project to allow multiple images to be taken at the same
time then auto stitched to produce larger landscape time lapse sequences or
stereo vision.  Was thinking of using pigpio remote feature
https://gpiozero.readthedocs.io/en/stable/remote_gpio.html  so second
camera could be triggered remotely at the same time via wifi, bluetooth,
ethernet or a hard wired gpio connection.   This allows file naming and
saving to be controlled as well as post image file transfer and processing
(stitching)

Your idea for speed camera could be implemented in a similar way but there
are a few issues since it takes time for the gpio connection to trigger it
might be better to have a second camera stream.
capture the second image or sequence from video stream since it takes time
for rpi or web camera to warm up.  Steam images are not as good a quality
as stills.
Will take a look at your idea.
Claude ..

jaimzj, 2020-08-09:
Thank you so much for taking time out to respond :)

closed 2021-07-02 by pageauc