pageauc/speed-camera #62: purge database option

peterb99, 2020-10-11:
I haven't been able to find any info on this.
Add an option to the menu to purge the database to a certain date, including cleaning up the images and reports before that date

pageauc, 2020-10-11:
Will take a look at adding a data retention days feature.  This could purge
database and associated images. older than specified days.
This would be a separate script so it does not slow down operation of
speed-cam.py.  The script could  then be added as a crontab
to run on a schedule to cleanup anything older than specified days..
Claude ...
..

bushibot, 2021-06-30:
Any good way to do this? I have a bunch of test and setup garbage in the db?

pageauc, 2021-06-30:
There are several ways.  First stop speed_cam.py

1  Rename data/speed_cam.db to another name  eg *mv** data/speed_cam.db
data/speed_cam_30jun21.db*
    a fresh copy will be created on restart.  If you do not delete images
you can restore back to old database by
  changing the DB_NAME in config.py.

2 Simply delete data/speed_cam.db  eg *rm** data/dpeed_cam.db*
   The database and structure will be recreated on next restart but your db
history will be lost.

3 Change configuration sqlite3 settings in config.py

You can change any of the following or any combination in config.py. I
suggest just creating a new
db table.  eg *DB_TABLE= "speed2"*.  The speed table will still exist and
you can change back to it if you
want to run historical reports or graphs.

# Sqlite3 Settings   in config.py
> # ----------------
> DB_DIR   = "data"
> DB_NAME  = "speed_cam.db"
> DB_TABLE = "speed"

You can then decide if you want to keep media/images or start a new
media/images folder or delete everything..  Depending on what you
do above and how much space you have..

Simplest would be just to change db name   eg DB_NAME = "speed_cam_2.db"

Restart speed_cam.py

If you are having problems with media/recent then the symbolic links can
just be deleted and new ones will be created
as new data and images are collected.  eg *rm media/recent/**

Hope this answers your question

I have tried not to hardcode any relevant settings such as database and
directory and subfolder locations etc..  config.py can control most speed
camera configuration settings so you
can change or reconfigure to suit your needs
You could also create plugins for customizing any user special
configuration changes or settings.

Hope this answers your question.  Note sometimes I tell users to just
delete/rename  the speed_camera folder and reinstall.
from curl script if they are having issues and they are not conscerned with
keeping history.

Claude .... .

bushibot, 2021-06-30:
Yes it does, thank you very much!

pageauc, 2021-07-02:
Added wiki at https://github.com/pageauc/speed-camera/wiki/How-to-Manage-Sqlite3-Database

closed 2021-07-02 by pageauc