pageauc/speed-camera #63: Help with architecture

ianmaciel, 2020-11-01:
Hello,

First of all, thank you for this project. It's awesome!

I'm building a public speed camera for my street - it is residential area were we have problem with the speed of the vehicles.

My plan is to show the speed in a large monitor, live, just like these:
[image]

I'm willing to develop the new interface I'll need and PR back to you. So before I get started do you have any architecture suggestion that will most likely be great for you?

I haven't plan yet, but I might do this using a web server receiving the speed (so I could show in a a second device) or show direct on the monitor connected on raspberry pi.

Besides this decision I still need to figure out the best place to add de needed code.

Regards,
Ian Maciel

pageauc, 2020-11-02:
Will you use an LED panel to display speed?

The speed camera would need to be pointing perpendicular with the road and
would
see cars in both directions. Speed camera records direction using separate
calibration for each direction.
You would then need to communicate with the LED Matrix Panel either
directly or to another device, Arduino or RPI. via cable, wifi, RF Etc

I would use something like zmq for communication.  Here is an example with
multiple RPI's that sends jpg images to a hub rpi for creating pano images
but all you would need is to send text. See
https://github.com/pageauc/panopi.  This would need ethernet cable or be
within wifi distance.

Here is code to convert text to a 2d matrix.
See char to pixel code
https://stackoverflow.com/questions/36384353/generate-pixel-matrices-from-characters-in-string

https://www.instructables.com/Raspberry-Pi-LED-Scrolling-Text-Display/

Hope this helps

inm-certi, 2020-11-02:
Hi, I will use an LCD TV - I think it will be easier to find in "large size" (will reasonable price) and also easier to implement (just create a scree and plug on a HDMI device). 
[image]

Yesterday night I started doing a scratch... I started a web server listen to a endpoint where the speed-cam would post each measure. I started using python+flask, but I am thinking to move to a framework more UI focused.
To the endpoint would receive the "speed value". Other extra information could be sent (like the direction, the picture itself or the configured speed limit).
I could add a menu entry to start the speed-monitor on menubox.sh and add some settings on config.py to setup the address of a remote monitor

Regarding the camera position, I think it will be on the top of light pole, with the camera pointing down to the street to avoid distortions (If I just put in a pole on the car's level, it will be too close to the street and I probably get high level of distortion).

Thank you for your feedback.
Ian

pageauc, 2020-11-02:
Here is text to image code that might help
https://github.com/akapila011/Text-to-Image

pageauc, 2020-11-02:
Last example was for encoding decoding information into/out of image.  Not
displaying text as an image.

This example should work better. https://github.com/pythad/nider
You can overlay text onto an image.  Also you could use opencv to overlay
text on an image similar
to what is done already in speed camera.  You would just need to increase
font size.  In fact you
could use the speed image as the background and make the font size and font
very large to
fill the available image.  The image can be small and resized to fill the
display because I don't think
high image quality is needed to display the text.
You could easily do a test with speed camera by changing the text font size
(pixels) and location that
is not a config.py variable although I could do that.
Claude ....

pageauc, 2020-11-03:
Greetings

I did some quick coding to allow putting larger text on images.  new
variables are in new config.py.
This is just a quick proof of concept but should work.  Could add setting
to use a blank image with a specific colour.
Could then show green background for OK, Orange for Warn and Red for
Speeding.  Just an Idea.
Extra variables would need to be added to do this.
Let me know what you think
speed-sign.py is just a copy of speed-cam.py but with a bit of code added
at approx line 1307
Not much code to implement as you can see.

Claude ...

pageauc, 2020-11-04:
Just checking to see if you received the latest speed_sign sample code
Let me know what you think
Claude ...

ianmaciel, 2020-11-04:
Hi, 

I'm sorry for the late reply.

I had been busy last days and I couldn't work on it again. I will probably get back on next weekend.

But yes, I received.. Thank you!! I soon as I can take a look with a more time I'll update here.

In the meantime, I'm running as a proof-of-concept here:
[image]

ianmaciel, 2020-11-23:
I finally got a time to continue with this.. I opened a PR adding the needed settings to show the speed sign.
I also included a container to work with vscode (with the container there is no need to setup the environment with the opencv and python dependencies).

The container will work on Linux (under Windows or Mac it won't be able to connect to USB devices - still possible to run using RTSP) and does not support GUI yet.

closed 2021-05-02 by pageauc