pageauc/speed-camera #65: Calibration issues

mike-bike-kite, 2021-01-10:
Your project looks great but I'm having issues trying to calibrate. I have an image of a car which is a length of 18 marks on your red ruler on the image. I assume this is 180 pixels. The car is 4044mm long. I believe I'm using the default plugin so I should edit the config.py file. I've tried various settings but nothing is producing believable results - a guy ambled past in his car at 143 mph a moment ago so I figure I've messed up :) Could you explain what I need to change?

Many thanks
Mike

PS it would be a great application of machine learning to identify the makes of cars in the images, look up the dimensions and then auto calibrate.

pageauc, 2021-01-10:
If you are using the default camera settings 320x240 then you are probably
too close to the road (objects to be tracked).  Speed camera uses opencv to
track the largest moving object in the camera view.  In my experience the
contour that is normally tracked is around a wheel.  There are a few
filters to eliminate large changes in contour tracking distance.  Having
the camera too close means there may be contour shifts to another contour
that disrupts the opencv tracking.  Also the camera lens has distortion so
that the farther away from the center of the image will represent a greater
distance plus about 20 percent due to lens distortion.  This will mess up
the pixel to distance measurements and the resulting speed.  Ideally the
vehicle should be measured mostly in the middle of the camera view and be
small enough that opencv contour tracking will not switch to another nearby
contour during tracking.  My default pixel length is about 80 pixels
with mm length about the same as yours in the 4-5 metre range.  Most small
cars are in this range.

There are two calibration settings for each road direction since one lane
will be closer.  Camera needs to be perpendicular to the object movement.
Anything moving at an angle will mess up the speed calibration that is
based on the number of pixels moved, the calibration of mm per pixel (using
calibration data provided) and the time taken.

At any rate if you are using the default camera setting then vehicles are
too close for accurate tracking.  This gets worse if you increase the
camera resolution size since it magnifies the problem.  If the camera is
calibrated for objects that are eg 60 ft away (30 metres) then moving
objects that are closer to the camera eg 30 ft  ( 10 metres) will show a
much faster speed due to being closer even though they are physically
moving at the same rate of speed.

Let me know if this answers you question
Claude ...

mike-bike-kite, 2021-01-10:
Thanks for getting back to me so soon, it's much appreciated, though I'm not sure what to do to get your program to work.

The webcam is a 1080P cam but I've not set that in the config.py file - the image appears fine in the camera view from you program. The camera is about 10 meters/yards from the road and the far side of the road is 17 meters away. Cars do tend to be parked at the side of the road but your program does "see" the moving cars fine. I tried setting the config.py settings to 180 and 4044.

```
calibrate = False         # Create a calibration image file with calibration hash markers 10 px per mark

cal_obj_px_L2R = 180      # L2R Moving Objects, Length of a calibration object in pixels
cal_obj_mm_L2R = 4044.0  # L2R Moving Objects, Length of the calibration object in millimetres

cal_obj_px_R2L = 180      # R2L Moving Objects, Length of a calibration object in pixels
cal_obj_mm_R2L = 4044.0  # R2L Moving Objects, Length of the calibration object in millimetres

```

It's currently night time here but this was the 1st can that came past. It was unfortunately a black car but it seemed to see it fine. I think I've uploaded the image file. 
[image]

The csv entry that was generated was the following. The actual speed was approx 20mph. One small comment would be to alter the slashes in the file name so they all match and perhaps put column headers in the csv file - I have no idea what the fields after the image file are (apart from L2R).

"20210110","19","47",13.55,"mph","media/images\speed-20210110-1141/speed-20210110--947575.jpg",93,60,101,51,5151,"L2R",",None,"

Thanks for helping out.

pageauc, 2021-01-10:
Road is way to close for good tracking.  You could try lowering the camera
so the red rectangle sees just bottom half of vehicle so tires are
visible.  This may allow better tracking, since the tires usually are good
contours
Also increase the MIN_AREA in the config.py file.  Default is 100 but you
should make this larger eg 200 or more since the image is really close.
Also set image_show_motion area to True.  This will show the final image
with the last contour it was tracking.  I should be a green rectangle.That
will allow you to see what is being tracked.  Note at night headlights are
usually tracked.

image_show_motion_area = True   # True= Display motion detection rectangle
area on saved images
MIN_AREA = 200         # Default= 100 Exclude all contours less than or
equal to this sq-px Area

Best solution would be if camera was farther away but I guess moving the
road or the house is not an option.

Let me know how you are doing.
Stay Healthy Claude ...

mike-bike-kite, 2021-01-11:
In all honesty I'm not having much luck. I unfortunately can't move my house or the road so the distance is fixed at 10m. The program does seem to recognise vehicles in that I'll get a decent green square appear around a car as it appears but the green square disappears as the car goes behind the parked car then appears to match individual sections of the moving car when it reappears (at least I think that's what's happening). Is there any way the vehicle image can be broken into a grid and then do partial matches on each section? This might stop it loosing the vehicles when it passes behind parked cars. Or perhaps just do edge matches and calculate from that? I guess I'm probably trying to teach my grandmother to suck eggs though :)

I'll admit I couldn't find the variables that control the red square so I couldn't alter this. I wonder if you should have some control screen through a python form rather than using a myriad of config files. Having calibration using machine learning to match a few popular cars would be a major step. People could then point a webcam at a road and there'd be no need for calibration. 

I think your application is great but, at the moment, I'm a bit frustrated because I can't use it :)

pageauc, 2021-01-11:
I you have a second floor you could mount camera pointing down at the
road.  This would give you distance and possibly view over parked car(s)

I wrote this program as a learning project for myself and made it available
for anyone on GitHub.  The source code is available and anyone can modify
it to suit their needs assuming they have some programming abilities in
python opencv, Etc.  I run my speed camera under Jessie on a very old 256K
RAM Raspberry pi with two USB ports, so pretty old.  I do this to test
software.  speed camera runs OK even on this old hardware.but using AI
would bring everything to a standstill.  Not everyone has the latest and
greatest.  Speed camera can also be used as a motion tracking security
camera where speed calculation is not important.  There is a plugin for
this

Opencv is very sensitive to lighting, contours, Etc.  Eg finding a white
car on a snow white background or green with green background can be
challenging for opencv.  There are many other similar constraints,.so
environmental conditions are important and not everyone will have a
suitable setup for optimal software performance.   There are other speed
camera type programs on the web, but they may have the same issues.

Claude ....

mike-bike-kite, 2021-01-11:
Thank you for all your assistance Claude. I've been looking through your code (which is very well documented by the way) but my Python is fairly basic and unfortunately I have no experience with image processing. My aim was to use the data from your app to provide info to neighbourhood groups on traffic flow and, if necessary, work out when and where to place mobile speed camera units.

closed 2021-01-11 by mike-bike-kite