pageauc/speed-camera #66: webhook calls on capture

mlapaglia, 2021-01-15:
have you considered adding the ability to call webhooks with the data from the captures? i'm working on a project that overlays plate data on a lpr camera and think it would be great to also put the speed there too.

https://www.youtube.com/watch?v=0hwYuqhNGIU

pageauc, 2021-01-16:
I can add a reference link to
https://github.com/mlapaglia/OpenAlprWebhookProcessor  So users can
implement this themselves.  Not sure
how Docker would work on Raspberry Pi.  At this time I don't think I will
add this feature but others can modify my code to suit their needs.
Note the speed camera only works with road perpendicular to camera view but
ALPR need to have the camera pointing parallel in order
to see the license plates.  The best setup would be two cameras that
communicate with each other.  One for speed and one for ALPR.

My Panopi project https://github.com/pageauc/panopi uses imagezmq to get
multiple cameras communicating with a HUB process that can be on a separate
computer or one of the camera computers.

Regards
Claude ...

mlapaglia, 2021-01-17:
by using webhooks the raspberry pi would call openalpr running on another box, it wouldn't need to run on the same device. 

I have one camera pointing perpendicular to the road that sends a payload to the processor via http. speed-camera would also capture the speed then make a http call with the information. Then theprocessor would take the two pieces of information to create the overlay for the camera.

Best thing about webhooks is the payload can be sent to any number of things for integration.

Thanks for your response!

pageauc, 2021-01-18:
Do you have implementation details and/or Github code.  I looked at the
webhooks github page but have never worked with it before.
It would be helpful if you could provide some expertise or a github push.
Claude ...

mlapaglia, 2021-06-10:
a [webhook ](https://en.wikipedia.org/wiki/Webhook) is basically just HTTP POSTing a JSON payload to a given url. If you can give me some direction I can get a pr together for this.

Where in `speed-cam.py` should I put a class that will handle sending a request to a random URL, On first glance the JSON object would look like

`{
   timeStam: '6/9/2021 9:42PM',
   speed: 25
   unit: 'MPH'
}`

pageauc, 2021-06-10:
Just create a stand alone script that can be imported with a try except so
it can be optional

mlapaglia, 2021-06-10:
something like `webhook.py` 
   ```
#!/usr/bin/env python

import requests

class Webhook:
    def Post(webhookUrl, date, speed, unit, travelDirection):
        r = requests.post(webhookUrl, data = {
            'date': date,
            'speed': speed,
            'unit': unit,
            'travelDirection': travelDirection,
        })
```

then in `speed-cam.py` after logging to csv:
```
if webhook_url:
   Webhook().Post(webhook_url, log_timestamp, ave_speed, speed_units, travel_direction)
```

and in `config.py`
```
# Webhook Settings
# ----------------
webhook_url = ''
```

pageauc, 2021-07-02:
In the latest version of speed-camera, I have implemented a user_motion-code.py that gets imported on speed-cam.py startup and gets executed after a motion tracking image is taken.  The filename is passed to this empty function.  There are comments in the script to allow executing user specific code.  The user_motion_code.py is not updated during speed camera UPGRADE so user modifications will not be lost.

I suggest you try putting your webhooks code there.  If you need more data you can query the sqlite3 database using the filename as a key for the query since it should be unique.  You can also set the config.py variable image_filename_speed = True to add the speed value to the image name right after the image_prefix as part of the image file naming.

closed 2021-07-02 by pageauc