pageauc/speed-camera #67: Lines appearing over image text after a few captures.

ItGetsTheRice, 2021-02-17:
Lines appear over image text after a few captures.

[image]

pageauc, 2021-02-17:
in the speed-camera folder, Edit config.py using nano eg nano config.py

in the Camera Image Section set

image_show_motion_area = False

ctrl-x y Enter to save change. Stop and Restart speed Camera from
menubox.sh menu pick

Let me know if this solves your problem.

Stay Safe
Claude ...

jabella, 2021-02-17:
Same issue here.  Changing the image_show_motion_area setting didn't resolve it.

pageauc, 2021-02-17:
John.

Are you sure False is capitalized and that you stopped are restarted speed
camera.  Also it will not change past images so
check current speed images after making the change. You can also start
speed-cam.py manually to check log activity.
Make sure config.py setting verbose = True is set.

You can manually start using the terminal command.

./speed-cam.py

Also check that you are not running with a plug-in that might be overriding
the image_show_motion_area setting.

Let me know how you are making out.

Claude ...

pageauc, 2021-02-18:
Did a test on two RPI's using latest update and upgrade of Buster.  I misunderstood the problem and now see you have blue text with a line.  Default text is white so not sure where the problem is.  I tested  image_show_motion_area = False and it works OK.

Image text is blue and default should be white so I think there might be a problem with your setup somewhere.  Here are the default image settings in config.py.  Image text for blue is (255, 0, 0 ).

If you still have problems you might want to try doing a menubox UPGRADE or mv your speed-camera folder and try a fresh install to see it the problem still persists.
Let me know what you find.

```
# Camera Image Settings
# ---------------------
image_path = "media/images"   # folder name to store images
image_prefix = "speed-"       # image name prefix
image_format = ".jpg"         # Default = ".jpg"  image Formats .jpeg .png .gif .bmp
image_show_motion_area = True # True= Display motion detection rectangle area on saved images
image_filename_speed = False  # True= Prefix filename with speed value
image_text_on = True          # True= Show Text on speed images   False= No Text on images
image_text_bottom = True      # True= Show image text at bottom otherwise at top
image_font_size = 12          # Default= 12 Font text height in px for text on images
image_font_scale = 0.5        # Default= 0.5 Font scale factor that is multiplied by the font-specific base size.
image_font_thickness = 2      # Default= 2  Font text thickness in px for text on images
image_font_color = (255, 255, 255)  # Default= (255, 255, 255) White
image_bigger = 3.0            # Default= 3.0 min=0.1 Resize saved speed image by specified multiplier value
image_max_files = 0           # 0=off or specify MaxFiles to maintain then oldest are deleted  Default=0 (off)

```

ItGetsTheRice, 2021-02-26:
Made the changes as recommended but still have the lines appearing.
[image]

pageauc, 2021-02-26:
Must be some other issue since I cannot reproduce your problem.  Try
reseating camera ribbon cable as well as camera lens mount socket.
Does line appear when text is turned off?  Also could be version of opencv
cv2.putText or even pi-camera video driver.
Claude ...

ItGetsTheRice, 2021-02-27:
It does not appear with text turned off. Seems to be a text specific problem as the lines colour are consistent with the text colour.

pageauc, 2021-02-27:
Could be an opencv issue in cv2.putText.

If you have not already done so do a *sudo apt update* and *sudo apt
upgrade*
Also try running speed-cam.py with python2 and 3

*cd ~/speed-camera*
* python2 speed-cam.py*

exit and

*python3 speed-cam.py*

Let me know the results
Claude ..

closed 2021-04-24 by pageauc