pageauc/speed-camera #69: headlights and trees

anthonymobile, 2021-03-01:
Thanks! great tool.

I have a windows that's got 2 curbside tree trunks splitting the view of the road. during the day, things are working well, but at night, especially with dark vehicles, when the headlights go behind a tree and reappear they seem to get detected as a new object. then there is the separate problem of multiple tracks on the same vehicle. is there a good way to filter that out?

curious if anyone has faced this particular problem.

these images taken just now with tracking_counter=10 and track_timeout = 2.0 (which doesn't seem to be working)

[image]
[image]
[image]
[image]
[image]

pageauc, 2021-03-01:
You are probably using a RPI4.
To reduce multiple events you can increase the delay between tracking
events

Edit config.py using nano and increase the event_timeout.  I would try 0.5
and adjust up or down depending on results.
This should also help with the

    event_timeout = 0.3 # Default= 0.3 seconds to wait for next motion
event before starting new track

Also config.py settings are somewhat optimized for 320x240 stream size. You
may want to try reducing size of stream image or
customize the crop area by overriding the default calculated settings per
comments about removing the comments. This will allow
you to reduce the size if the tracking crop area to a smaller size. This
will reduce multiple tracks for the same object.

>
> # Allow user to customize the motion crop area (width) x values
> # If variables not found then values will be set automatically base on
> image size.
> # x_left = 50          # uncomment and change values to override auto
> calculate
> # x_right = 300        # uncomment and change values to override auto
> calculate
> # Allow user to customize the motion crop area (height) y values
> # If variables not found then values will be set automatically base on
> image size.
> # y_upper = 80         # uncomment and change values to override auto
> calculate
> # y_lower = 180        # uncomment and change values to override auto
> calculate

Note this can take a bit of time to get the tracking crop area you want.

anthonymobile, 2021-03-01:
thx, using RPI3. knocked the resolution and track_counter back down. realized i was using the plugin which explains why the timeout wasn't working. i will try these suggestions. a couple of questions tho:

1. what's the difference between event_timeout and track_timeout? i thought that track_timeout was what you describe for event_timeout above

2. is there any disadvantage to restricting the tracking crop area? assuming some very fast objects might get through undetected, if they dont get captured on at least track_count # of frames?

anthonymobile, 2021-03-01:
here's a good example just now, using a much more limited crop area, and event_timeout=0.3 
not sure how to prevent this. perhaps i should just invest my energy in a better camera location?

[image]
[image]

pageauc, 2021-03-01:
Motion tracking is made up of multiple events until the track counter is
exceeded.

event_timeout is the time between tracking events.  A motion track is made
up of multiple events that are determined by how
close each event is to the previous event.  If the distance is too far away
it is rejected since it may have been another larger motion area someplace
else or
at a different location on a large moving object eg bus or truck.
Sometimes the object will stop in the
middle of a track.  The program will wait for the next event and give up
after the timeout and look for a new start event.
This avoid waiting forever. eg car stops and parks in the middle of a track.

At the end of a track there is a timeout to try to avoid tracking the same
object several times.  The track timeout allows
time for the object to move out of the motion crop rectangle so it won't be
tracked again.

The tracking crop area can also be modified to be smaller in the width or
height.  making the width smaller can
also avoid duplicate motion tracks.

Hope this helps
Claude ...

anthonymobile, 2021-03-01:
i am now calibrated and configured well, at least for daylight. here's my settings:

and the calibration which i verified with a Bushnell Speedster III radar gun

```
# Motion Event Settings
# ---------------------
SPEED_MPH = True      # Set Speed Units   kph=False  mph=True
track_counter = 5      # one cpu core=5 four cores=10 Number of Consecutive Motion Events to trigger speed photo

track_timeout = 2.0    # Default= 0.0 Optional seconds to wait after track End (Avoids dual tracking)
event_timeout = 0.3    # Default= 0.3 seconds to wait for next motion event before starting new track
max_speed_over = 15     # Exclude track if Speed less than or equal to value specified 0=All

y_upper = 130         # uncomment and change values to override auto calculate
y_lower = 220        # uncomment and change values to override auto calculate

```

```
# calibration completed verified by radar 2021-03-01 1230pm
cal_obj_px_L2R = 68      # L2R Moving Objects, Length of a calibration object in pixels
cal_obj_mm_L2R = 4700.0  # L2R Moving Objects, Length of the calibration object in millimetres

# calibration completed verified by radar 2021-03-01 1230pm
cal_obj_px_R2L = 65      # R2L Moving Objects, Length of a calibration object in pixels
cal_obj_mm_R2L = 4700.0  # R2L Moving Objects, Length of the calibration object in millimetres
```

getting a fairly nice set of accurate, non-duplicate data now!

[image]

closed 2021-03-01 by anthonymobile