pageauc/speed-camera #70: makehtml.py IndexError: list index out of range

anthonymobile, 2021-03-02:
it seems that that `make_web_page(cur_link, next_row, next_link)` on line 191 is always feeding an empty `next_row` to the `make_web_page` function but i cant figure out why

```
/home/pi/speed-camera/makehtml.py ver 6.70 - written by Claude Pageau
Processed 5 web pages in 0 seconds into Folder /home/pi/speed-camera/media/html
Done ...
Traceback (most recent call last):
  File "./makehtml.py", line 201, in <module>
    read_from_csv(source_csv)
  File "./makehtml.py", line 190, in read_from_csv
    make_web_page(cur_link, next_row, next_link)
  File "./makehtml.py", line 40, in make_web_page
    YYYYMMDD=row_data[0]
IndexError: list index out of range

######################################
#          Review Output             #
######################################
  Press Enter to Return to Main Menu

```

anthonymobile, 2021-03-02:
looks like the position of fields changed. i am working on a PR

anthonymobile, 2021-03-02:
fixed [https://github.com/pageauc/speed-camera/pull/71](https://github.com/pageauc/speed-camera/pull/71)

anthonymobile, 2021-03-02:
> Thanks Anthony. Greatly Appreciated. This was due to changing csv and
> sqlite schema to make one datetime field and eliminate hour and minute
>
> Claude ..

n.b. i think there is still one error in the area computation. i can fix it
later.

pageauc, 2021-03-03:
Must be your data.   I ran it with no problems on a fresh install.  You
should do an upgrade
Stop speed camera.
Clear *speed-cam.csv* file (rename or delete)  Clear all items in
*media/recent,* (these are symlinks) and backup *media/images*  to another
location so images folder is empty.
and make a fresh start.  Should also clear (rename or delete)
*data/speed_cam.db*

Start speed camera and Collect new images and data and retry running
makehtml.py  I ran under python 2 and 3 without problem.  You should also
run *sql_speed_gt.py* to test graphing data. View graphs in browser under
reports folder on web page  You can also pass it a speed parameter to avoid
being prompted.  I usually run at 20 km to  avoid bikes and other slow
objects.  I plan on adding days to query eg past days from now.  eg past 7
days, Etc.  This would also be a parameter.  This could be run from a cron
job to automate.
I am thinking of changing from Gnuplot to MatLab for plotting graphs.

Let me know how you make out

Let me know if clearing your data works.  I have had no problems in my
testing.

Regards Claude ....

anthonymobile, 2021-03-03:
not sure what we are talking about—my errors happened before i submitted the pull requests updating how `makehtml.py` parses  the new row format in `speed-cam.csv`, which you accepted. my last comment was about one update i overlooked with was how the area computation was done (computed now, vs fetched from the row). i think you also accepted that PR. so this is all good. re-closing.

closed 2021-03-03 by anthonymobile

pageauc, 2021-03-03:
Ok Sound like everything is good with you as well.  Thanks for your update.
Thanks for your contribution to the project.
Your help is appreciated
Claude ...