pageauc/speed-camera #73: Issues with plugins

ItGetsTheRice, 2021-03-02:
Changed to picam480 but now motion tracking seems buggy. Only capturing 1 out of 10 vehicles and captures are registering speeds way higher than can be accurate.

pageauc, 2021-03-02:
Lower  x_diff_max.  I think this is a typo on my part.  Adjust setting slightly larger if missing too many fast objects.

x_diff_max = 20

ItGetsTheRice, 2021-03-09:
In motion event settings? Sorry if I misunderstood, that is already in the code as x_diff_max = 20. Do you mean lower it for the issue I am experiencing?

pageauc, 2021-03-09:
Here are my config.py motion settings image resolution 320x240 No plugin

If you are using a plugin you can copy any required config.py settings to
the plugin.  The plugin settings
will over ride the current config.py settings.  You can also create your
own plugin.  plugins allow
customizing settings for certain situations.  Eg some users have multiple
cameras and/or camera locations

# Motion Event Settings
> # ---------------------
> SPEED_MPH = False      # Set Speed Units   kph=False  mph=True
> track_counter = 5      # one cpu core=5 four cores=10 Number of
> Consecutive Motion Events to trigger speed photo
>                                  # Testing with RPI2 B one core using
> opencv ver 3.4.2
> MIN_AREA = 100         # Default= 100 Exclude all contours less than or
> equal to this sq-px Area
> show_out_range = True  # Default= True Show Out of Range Events per x_diff
> settings below False= Off
> x_diff_max = 20        # Default= 20 Exclude if max px away >= last motion
> event x position
> x_diff_min = 1         # Default= 1 Exclude if min px away <= last event x
> position
> x_buf_adjust = 10      # Default= 10 Divides motion Rect x for L&R Buffer
> Space to Ensure contours are in
> track_timeout = 0.0    # Default= 0.0 Optional seconds to wait after track
> End (Avoids dual tracking)
> event_timeout = 0.3    # Default= 0.3 seconds to wait for next motion
> event before starting new track
> max_speed_over = 0     # Exclude track if Speed less than or equal to
> value specified 0=All
>                                      # Can be useful to exclude
> pedestrians and/or bikes, Etc or track only fast objects

Take a look at the data logs when you run speed-cam.py to view logs (Not as
background task).  Look at data of the problem tracks
A good track should look like

> ----------------------------------------------------------------------
> 2021-03-09 06:47:58 INFO     speed_camera New  - 0/5 xy(61,73) Start New
> Track
> 2021-03-09 06:48:52 INFO     speed_camera Reset- event_timer 53.84>0.30
> sec Exceeded
> 2021-03-09 06:48:52 INFO     speed_camera New  - 0/5 xy(163,69) Start New
> Track
> 2021-03-09 06:48:52 INFO     speed_camera  Add - 1/5 xy(152,70) 83.33 kph
> D=11/20 C=4 49x35=661 sqpx R2L
> 2021-03-09 06:48:52 INFO     speed_camera  Add - 2/5 xy(142,70) 28.14 kph
> D=10/20 C=5 49x35=638 sqpx R2L
> 2021-03-09 06:48:52 INFO     speed_camera  Add - 3/5 xy(132,71) 52.88 kph
> D=10/20 C=5 48x35=562 sqpx R2L
> 2021-03-09 06:48:52 INFO     speed_camera  Add - 4/5 xy(122,70) 49.85 kph
> D=10/20 C=4 49x36=566 sqpx R2L
> 2021-03-09 06:48:52 INFO     speed_camera  Add - 5/5 xy(112,70) 26.94 kph
> D=10/20 C=4 51x37=652 sqpx R2L
> 2021-03-09 06:48:52 INFO     speed_camera  Saved
> media/images/speed-20210308-1215/speed-20210309-0648529.jpg
> 2021-03-09 06:48:53 INFO     isSQLite3  Success: File is sqlite3 Format
> /home/pi/speed-camera/data/speed_cam.db
> 2021-03-09 06:48:53 INFO     db_check   Success: sqlite3 Connected to DB
> /home/pi/speed-camera/data/speed_cam.db
> 2021-03-09 06:48:54 INFO     speed_camera  SQL - Inserted sqlite3 Data Row
> into /home/pi/speed-camera/data/speed_cam.db
> 2021-03-09 06:48:54 INFO     saveRecent    symlink
> /home/pi/speed-camera/media/recent/speed-20210309-0648529.jpg
> 2021-03-09 06:48:54 INFO     speed_camera End  - R2L Ave Speed 48.2 kph
> Tracked 51 px in 0.248 sec Calib 85px 4700mm
> ----------------------------------------------------------------------

[... 38 lines cut ...]
> ----------------------------------------------------------------------

Example track above show one OUT log entry in bold.  You can see D=
exceeded the max so was rejected but track was still
successful since the timeout was not exceeded.  If settings are not right
or vehicle speed is too great for your situation you will
get a lot of OUT entries and track may timeout.  You will then need to
adjust the config.py motion settings.  Larger objects (trucks, busses,
limos, Etc) and event some larger objects
can be detected as multiple contours like wheel wells, trunk, graphics,
shadows, reflected lighting, Etc.  Check the C= value
to see how many contours were detected..  Too many can be a problem and you
may need to change camera position, resolution, Etc.
Note you may also see OUT due to x-diff being less than 1 (contour detected
behind previous point)

If objects are moving faster and you get a lot of track timeouts and
incomplete tracks eg could not get

If objects are too close or too large eg camera is too close or objects too
large in camera view then speed camera will have problems.
Take a look at your log tracking info and try to diagnose what your problem
could be base on info above.

Claude ...

closed 2021-04-24 by pageauc