pageauc/speed-camera #75: Reduce file size of saved jpeg

RandallKent, 2021-03-06:
Allow users to reduce the file size of saved jpeg, by exposing [ImwriteFlags](https://docs.opencv.org/3.4/d8/d6a/group__imgcodecs__flags.html#ga292d81be8d76901bff7988d18d2b42ac) in the config to enable togging jpeg optimization and adjusting the quality of the jpeg.

closed 2021-03-07 by pageauc

RandallKent, 2021-03-07:
@pageauc I see you merged #76 and but also updated the [speed-cam.py#1355](https://github.com/pageauc/speed-camera/blob/master/speed-cam.py#L1355) to have logic that requires `image_jpeg_optimize` to be set to `True` in order for the `image_jpeg_quality` variable to take effect.

Your logic is incorrect.

Each flag is independent: optimization is different from compression (quality). Each can and should be able to be to configured separately.

The real file size savings comes from `image_jpeg_quality` therefore, a user should be able to set the quality of an image without setting `image_jpeg_optimize` to `True`.

pageauc, 2021-03-07:
Yes after looking more closely at the docs I see that optionize parameter
can be toggled on or off.  I just want to be able to toggle the feature on
or off so maybe another boolean
to enable or disable the quality/optimization ver of imwrite.

How about
image_jpg_custom_on = False  # Allow custom jpg settings for quality and
image optimization.

This would allow customization of quality or optimization.

RandallKent, 2021-03-07:
By setting the defaults within the speed-camera to match the [defaults of OpenCV](https://docs.opencv.org/3.4/d8/d6a/group__imgcodecs__flags.html#ga292d81be8d76901bff7988d18d2b42ac), we are effectively providing your desired functionality.

In other words, the feature is not toggled on unless it is changed from the default value. If the user leaves the default value, it's as if the feature is "disabled".

pageauc, 2021-03-07:
Ok.  Thanks for your help.  I have reverted change to your previous logic
and added comments
certainly your solution is simpler and fewer config.py variables.

I learned something today.  I have enjoyed working with you and your
patience while I got up to speed.

Regards Claude ...

pageauc, 2021-03-07:
Testing image optimize feature and image quality to my eye appears to be
sharper.  Today was sunny. Images were quite clear with no noticeable blur.
Will do more testing especially during overcast periods which is quite
often at this time of year in Ontario Canada
Regards
Claude ...

PR #76 (2021-03-07): Add ability to configure jpeg optimize and quality
closes #75
commit ff8b7328 (2021-03-07): Add ability to configure jpeg optimize and quality

closes #75