pageauc/speed-camera #81: Divide by Zero

gregbonnette, 2021-04-22:
I can't seem to keep the program stable

e. After one or two passes of a vehicle it exits with a divide by zero error:

> Traceback (most recent call last):
>   File "C:\speed-camera\speed-cam.py", line 1669, in <module>
>     speed_camera() # run main speed camera processing loop
>   File "C:\speed-camera\speed-cam.py", line 1245, in speed_camera
>     cur_ave_speed = float((abs(cur_track_dist /
> ZeroDivisionError: float division by zero

pageauc, 2021-04-22:
This is the code at 1245 

                       if travel_direction == 'L2R':
                            cur_ave_speed = float((abs(cur_track_dist /
                                                       float(abs(cur_track_time -
                                                                 prev_start_time)))) *
                                                  speed_conv_L2R)

division by zero is something below the / 
I don't think the cur_track_time - prev_start_time would be zero so I am thinking that the **speed_conv_L2R**
might be causing the problem.  Please check you calibration settings in config.py.  It might be that one 

cal_obj_px_L2R = 80      # L2R Moving Objects, Length of a calibration object in pixels
cal_obj_mm_L2R = 4700.0  # L2R Moving Objects, Length of the calibration object in millimetres

Also check 

track_counter = 5      # one cpu core=5 four cores=10 Number of Consecutive Motion Events to trigger speed photo
                                 # Testing with RPI2 B one core using opencv ver 3.4.2

These are the default settings.  Please let me know what your config.py settings are since some value in the formula below the division / is resulting in a zero value being calculated and thus causing the division by zero error.

Let me know
Claude ....

pageauc, 2021-04-24:
Let me know if you resolved your issue.

gregbonnette, 2021-04-24:
Here are my config settings after calibrating:

cal_obj_px_L2R = 70      # L2R Moving Objects, Length of a calibration object in pixels
cal_obj_mm_L2R = 4700.0  # L2R Moving Objects, Length of the calibration object in millimetres

track_counter = 5      # one cpu core=5 four cores=10 Number of Consecutive Motion Events to trigger speed photo
                       # Testing with RPI2 B one core using opencv ver 3.4.2

Thanks

pageauc, 2021-05-02:
What are the R2L calibration values?  Defaults are below

cal_obj_px_R2L = 85      
cal_obj_mm_R2L = 4700.0

This is the other lane of traffic.  If you are still having problems I recommend you run menubox.sh UPGRADE option to see if that resolves your issue.  I have this running on a RPI4 with no issues and have not seen the divide by zero message.

I have one speed camera running on a single core 512 kb RAM memory RPI with not issues if I run as default 320 x240 stream size.
Let me know if you still have divide by zero problem
Claude ...

closed 2021-05-27 by pageauc