pageauc/speed-camera #82: SQLite Database not created

schnuerle, 2021-05-02:
Ok so this is a tech support request, so I hope this is the best place to put this. Let me know otherwise.

First, thanks for this work! I'm getting a version of this running in Louisville, KY, which I hope to replicate with a group of interested people here.

I've got a brand new RPi 4 and have everything up and running. You can see my [step by step documentation here](https://docs.google.com/document/d/17REUa8e6ZNDZWHUYEdvYXbzbNEvhjly00XKJlYgC6mE/edit#). I'm at the configuration part. 

I've got it running and the webserver running locally at http://192.168.1.2:8080/ - but no data is coming through on this page yet.

When I run it I get an error with the DB. Basically the DB does not exist.

`2021-05-02 15:31:15 ERROR    db_check   Failed: sqlite3 Not DB Format /home/pi/speed-camera/data/speed_cam.db`

I feel like I may have missed some step where the DB and tables in it are created. 

Also of note is there is no `speed-cam.csv` file showing up in the speed-cam folder. Maybe all of these get created on the first image detection?

Related to that thought, I have a USB camera that is working fine with VLC streaming. I have these settings setup for it.

```
CAM_LOCATION = 'None'  # Specify an address, physical location Etc for camera
WEBCAM = True         # Default= False False=PiCamera True= USB Webcam or RTSP,IP Camera

# Web Camera Settings
WEBCAM_SRC = 0         # Default= 0   USB camera device connection number
```

If I run `lsusb`, I get this which shows the camera as device $004 (Microdia). I can confirm this by removing the camera and re running:

```
pi@raspberrypi:~/speed-camera $ lsusb
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 004: ID 0c45:63ab Microdia 
Bus 001 Device 003: ID 1997:2433  
Bus 001 Device 002: ID 2109:3431 VIA Labs, Inc. Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
```

However if I change the config to this: `WEBCAM_SRC = 4` then I get this error, which is not there when it's equal to 0:

`VIDEOIO ERROR: V4L: index 4 is not correct!`

Any help would be appreciated and happy to provide more details!

pageauc, 2021-05-02:
I tried one of my RPI4's with a web cam.  lsusb is below

``
Bus 002 Device 002: ID 2109:0715 VIA Labs, Inc.
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 016: ID 1415:2000 Nam Tai E&E Products Ltd. or OmniVision Technologies, Inc. Sony Playstation Eye
Bus 001 Device 002: ID 2109:3431 VIA Labs, Inc. Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

When a video device eg webcam is plugged in there are one or more video device files created in /dev  The first webcam should be **/dev/video0**  If you only have one webcam it should always be 0. video1 will be second camera.  So opencv will look for the video number you specify.  0, 1, 2, Etc depending on number of webcams are plugged in.  The lsusb is just listing the unix bus location of the device for hardware purposes so don't confuse the video device file with the hardware bus location. 

So if you only have one usb camera plugged in then it should always be video0 and the default config.py should work just fine assuming camera works OK.   

This is just a brief explanation but should be good enough to get you started.  Devices can be more complicated than my basic info above.

The sqlite database will be created in /home/data/speed-camera/data/speed_cam.db and is created when speed camera is first launched.

If you are having problems make sure all files are owned by the pi user.  You can check this in a directory with the example command. the -l is the lower case letter L not a number 1

    cd ~/speed-camera
    ls -l 
    pi@rpi-box3:~/speed-camera $ ls -l
    total 280
    -rwxr-xr-x 1 pi pi  7487 May  2 17:34 alpr-speed.py
    -rw-r--r-- 1 pi pi 11626 May  2 17:34 config.py
    -rw-r--r-- 1 pi pi  3134 May  2 17:35 config.pyc
    -rw-r--r-- 1 pi pi 11625 May  2 17:34 config.py.new
    drwxr-xr-x 2 pi pi  4096 May  2 18:10 data
    -rwxr-xr-x 1 pi pi  6898 May  2 17:34 makehtml.py
    drwxr-xr-x 6 pi pi  4096 May  2 17:35 media
    -rwxr-xr-x 1 pi pi 24860 May  2 17:34 menubox.sh
    drwxr-xr-x 2 pi pi  4096 May  2 17:30 plugins
    -rw-r--r-- 1 pi pi     0 May  2 17:30 -q
    drwxr-xr-x 2 pi pi  4096 May  2 17:30 rclone-samples
    -rwxr-xr-x 1 pi pi  4353 May  2 17:34 rclone-security-sync-recent.sh
    -rw-r--r-- 1 pi pi 10848 May  2 17:34 Readme.md
    -rwxr-xr-x 1 pi pi   641 May  2 17:34 remote-run.sh
    -rwxr-xr-x 1 pi pi  2049 May  2 17:34 search_config.py
    -rw-r--r-- 1 pi pi   592 May  2 17:35 search_config.pyc
    -rwxr-xr-x 1 pi pi 14162 May  2 17:34 search-speed.py
    -rwxr-xr-x 1 pi pi 79468 May  2 17:34 speed-cam.py
    -rwxr-xr-x 1 pi pi  1245 May  2 17:34 speed-cam.sh
    -rwxr-xr-x 1 pi pi  7463 May  2 17:34 sql_speed_gt.py
    -rwxr-xr-x 1 pi pi 13445 May  2 17:34 watch-app.sh
    -rwxr-xr-x 1 pi pi 13285 May  2 17:34 webserver3.py
    -rwxr-xr-x 1 pi pi 12611 May  2 17:34 webserver.py
    -rwxr-xr-x 1 pi pi  1318 May  2 17:34 webserver.sh

I recommend you install using the curl command method and upgrade using menubox.sh UPGRADE menu pick.

Let me know your progress.
Claude ...

pageauc, 2021-05-02:
Just FYI the WEBCAM_SRC can also point to an RTSP (real time streaming protocol)  ip camera per wiki reference below.  This is also mentioned in the config.py entry comments.

https://github.com/pageauc/speed-camera/wiki/How-to-Setup-RTSP--IP-Camera-video-stream

schnuerle, 2021-05-03:
Thanks for the quick reply and useful info.

Confirmed that my USB camera (not IP camera) is on video0.  That is what I have the config file: WEBCAM_SRC = 0 

For the permissions, good suggestion, and I checked and everything is owned by the pi user, and that is the user I am logged in as.

```
pi@raspberrypi:~/speed-camera $ ls -l
total 284
-rwxr-xr-x 1 pi pi  7487 Apr 30 17:19 alpr-speed.py
-rw-r--r-- 1 pi pi 11618 May  2 15:31 config.py
-rw-r--r-- 1 pi pi  3139 May  2 15:31 config.pyc
-rw-r--r-- 1 pi pi 11625 Apr 30 17:19 config.py.new
drwxr-xr-x 2 pi pi  4096 Apr 30 18:04 data
-rwxr-xr-x 1 pi pi  6898 Apr 30 17:19 makehtml.py
drwxr-xr-x 7 pi pi  4096 Apr 30 20:22 media
-rwxr-xr-x 1 pi pi 24860 Apr 30 17:19 menubox.sh
drwxr-xr-x 2 pi pi  4096 Apr 30 17:17 plugins
-rw-r--r-- 1 pi pi     0 Apr 30 17:17 -q
drwxr-xr-x 2 pi pi  4096 Apr 30 17:17 rclone-samples
-rwxr-xr-x 1 pi pi  4353 Apr 30 17:19 rclone-security-sync-recent.sh
-rw-r--r-- 1 pi pi 10848 Apr 30 17:19 Readme.md
-rwxr-xr-x 1 pi pi   641 Apr 30 17:19 remote-run.sh
-rwxr-xr-x 1 pi pi  2049 Apr 30 17:19 search_config.py
-rw-r--r-- 1 pi pi   592 Apr 30 17:20 search_config.pyc
-rwxr-xr-x 1 pi pi 14162 Apr 30 17:19 search-speed.py
-rw-r--r-- 1 pi pi   292 May  2 15:31 speed-cam.log
-rwxr-xr-x 1 pi pi 79468 Apr 30 17:19 speed-cam.py
-rwxr-xr-x 1 pi pi  1245 Apr 30 17:19 speed-cam.sh
-rwxr-xr-x 1 pi pi  7463 Apr 30 17:19 sql_speed_gt.py
-rwxr-xr-x 1 pi pi 13445 Apr 30 17:19 watch-app.sh
-rwxr-xr-x 1 pi pi 13285 Apr 30 17:19 webserver3.py
-rwxr-xr-x 1 pi pi 12611 Apr 30 17:19 webserver.py
-rwxr-xr-x 1 pi pi  1318 Apr 30 17:19 webserver.sh
```

I did originally install with curl on Apr 30, and have not needed to upgrade yet, and have used menubox.sh a bunch.

So I think the issue is that the DB and CSV are not being created at speed camera runtime, even with `log_data_to_CSV = True` and `loggingToFile = True`.

Any thoughts on that? I'm not sure how else to troubeshoot it.

pageauc, 2021-05-03:
I am thinking you may have a corrupt or damaged sd card.  Take another sd
card and created a new Buster Lite image and install speed-camera on this
fresh image.
Then run new install with calibrate = False and update any calibration
settings or use stock to do the initial testing.
Damaged SD did happen to me. Not sure what the problem with the SD card was
like a firmware issue or some other issue media issue..

Also,  I would not set loggingToFile = True since it writes all log
messages to the log file and this will slow the motion tracking so you will
see a lots of track timeouts.  You can monitor logging messages by sending
to console / terminal.
Using command below.  Make sure verbose = True in config.py and
loggingToFile = False.

    cd ~/speed-camera
    ./speed-cam.;py

Monitor the messages on the console to confirm things are working OK
This advice has helped other users with weird problems.

Let me know your progress
Claude ...

schnuerle, 2021-05-05:
Ok @pageauc I got it running!

I tried a few things before going down the new SD card route (didn't have one) or starting over with Buster Lite (wanting to use the default Rasp Pi OS instead).

Tried a few config file changes and permissions changes. Then ran a full software upgrade on the Pi again. But none of that worked.

Then moved speed-camera folder to trash, emptied, rebooted. Did a curl install of speed-camera, and left all the settings default.  Ran `./speed-cam.py`. That didn't work right away, with an error that the camera was not detected. But changing one setting, `WEBCAM = True`, that detected the USB camera.  Then it started pulling motion events and logging pictures!

Thanks for your help, and I'll let you know here when it's up and calibrated and running well!

pageauc, 2021-05-05:
Thanks for the update.  Glad it is now working.  May still be an SD card
issue since trashing previous install would force new install onto
different track sectors of card.
or it may have been some file corruption during original install due to a
network problem.  This has happened to me in the past. Most of my long term
setups use hard drives to store media files.  Sometimes I install

We are still in lockdown so it is always nice hearing from others
Stay Safe
Claude ....

schnuerle, 2021-05-05:
I'm just going to keep asking you questions then! This is a great lockdown project, wish I would have found and started it sooner. 

I'm working on changing some settings and calibrating the speeds now. I have a 1080p webcam, and would like to bump up the resolution from 240 to 480 or even 720, hoping it may be more accurate.

If I set pluginName = "webcam480", I can see higher res images, and have adjusted x_left, x_right, y_upper, y_lower for the right crop area and it looks good.  

What other settings do I need to configure? Do I have to change WEBCAM_WIDTH and WEBCAM_HEIGHT? 

Is there a WEBCAM_FRAMERATE option since my cam supports 30? 

I also doubled my cal_obj_pxL2R and cal_obj_pxR2L numbers since the resolution is doubled. Haven't driven my vehicle yet to do real calibration.

With all that said, at 240 the speed numbers were pretty consistent for the same vehicle as it went past, eg, 20-30mph. But now at 480 the same vehicle will have anything from 10 to 120mph! So something is off with my config at that resolution.

pageauc, 2021-05-05:
That is a problem a lot of users get into.  Trying to push the resolution
too high for the RPI model or situation.  Speed camera uses motion tracking
of the largest contour with larger images there seems to be more chance
that the largest contour will jump around inconsistently on the moving
object, especially if the camera is too close to the moving objects.  Works
best if objects are small in relation to the camera frame.  This is what is
driving your errors and inconsistencies.  The plugins were setup as a way
to save different configurations.

The settings for motion tracking are grouped in the config.py file per
below.  See comments for what each does.
Monitor the logging to see what is happening.  You will see track starts,
timeouts, exclude, Etc logging messages.
Speed will be calculated when there are 5 consecutive successful track
events that do not violate the excludes.
The logging messages will tell you briefly what tracking events were
excluded and why.
On some low end RPI's I was missing quite a few cars going by and had to
reduce track counter to 4 and set x_diff_max higher.
You may also have change the MIN_AREA to exclude smaller contours..  This
will take a bit of trial and error to tune camera settings.
You could then save them as a plugin for a particular camera, location, etc.
Speed Camera works best if the moving objects are small.  Eg Some users
have camera where vehicle takes up most of the camera view.  This can be a
problem and hard to get working properly.  I still have problems with speed
for large trucks going by.

# Motion Event Settings
> # ---------------------
> SPEED_MPH = False      # Set Speed Units   kph=False  mph=True
> track_counter = 5      # one cpu core=5 four cores=10 Number of
> Consecutive Motion Events to trigger speed photo
>                        # Testing with RPI2 B one core using opencv ver
> 3.4.2
> MIN_AREA = 100         # Default= 100 Exclude all contours less than or
> equal to this sq-px Area
> show_out_range = True  # Default= True Show Out of Range Events per x_diff
> settings below False= Off
> x_diff_max = 20        # Default= 20 Exclude if max px away >= last motion
> event x position
> x_diff_min = 1         # Default= 1 Exclude if min px away <= last event x
> position
> x_buf_adjust = 10      # Default= 10 Divides motion Rect x for L&R Buffer
> Space to Ensure contours are in
> track_timeout = 0.0    # Default= 0.0 Optional seconds to wait after track
> End (Avoids dual tracking)
> event_timeout = 0.3    # Default= 0.3 seconds to wait for next motion
> event before starting new track
> max_speed_over = 0     # Exclude track if Speed less than or equal to
> value specified 0=All
>                        # Can be useful to exclude pedestrians and/or
> bikes, Etc or track only fast objects

Try to keep things simple and change one thing at a time.  Also calibrate
for more accurate results.

Also see wiki section
https://github.com/pageauc/speed-camera/wiki/How-to-Tune-Motion-Tracking-Settings

and
https://github.com/pageauc/speed-camera/wiki/Program-Description#accuracy-issues

Hope this helps
Claude ...

schnuerle, 2021-05-05:
Ok I'll take a look at all of that thank you. Especially I'll look at the Motion Tracking page. I'll let you know.

FYI I've got a [R Pi 4v2 with 4GB RAM](https://www.adafruit.com/product/4296) and a [64GB speed class 30 SD card](https://www.amazon.com/gp/product/B07G3GMRYF/ref=ppx_yo_dt_b_asin_title_o03_s00?ie=UTF8&psc=1), so it should be able to handle 480p.

And for your pleasure, here's an [animated gif](https://i.imgur.com/H8jmMEz.gif) I made from some stills at the 240p resolution, with consistent speeds. I think the road distance and framing is good (some of the image is cropped FYI for this GIF). Those trees may end up being in the way though.

Here's a [full frame still image](https://i.imgur.com/4Z5sqOB.jpg) at 480p. Oh, what are those 2 white lines through the text in each image?

schnuerle, 2021-05-07:
I think I've got it all working and calibrated now! I tried at 480p but the results were still all over the place. So I kept it at 240p and things are much more consistent! 

You can see what I did in my [document](https://docs.google.com/document/d/17REUa8e6ZNDZWHUYEdvYXbzbNEvhjly00XKJlYgC6mE/edit#).

But the trees you see in my sample image are really affecting it I think. I may have to figure out how to run an extension cord to an outdoor-proof IP camera mounted on the side of my house near the driveway if I want to get a better results (but I don't think I'm going to do that!). 

Thanks for your help! Next I'll mess around with some data and reports.

schnuerle, 2021-05-10:
I've narrowed my crop area to be just the space between trees. It's small but the speed results are remarkable consistent ([see it here](https://i.imgur.com/xoru9OA.png)). I'm happy with it now as is and it's working well so thank you for your help!

In the interest of tinkering, I'm thinking of trying 2 things and would like your advice to see if it's worth it.

1. Bumping up the resolution to 480p or 720p. My pi4 4GB can handle it after some testing. I want to see if the results are more accurate since the number of pixels in the small crop area would be 4 times or 9 times greater. Do you think more pixels leads to better speed accuracy in this case?

2. Is there a way to increase the USB camera framerate, from 20fps to 30fps (which my webcam supports)?

pageauc, 2021-05-10:
Bump up the resolution but keep crop size the same as 320x240 area you are
using.  OpenCV processes the crop area for contours, motion tracking and
speed calculation.
This should give you larger resolution but keep the accuracy of the 320x240
crop area.  If this works let me know and I will change the plugin

Also it is possible to set a webcam property for framerate but not all
cameras support this and you will get an error if you try to set beyond
camera ability

For reference, the first argument in the cap.set() command refers to the
enumeration of the camera properties, listed below:

0. CV_CAP_PROP_POS_MSEC Current position of the video file in milliseconds.
> 1. CV_CAP_PROP_POS_FRAMES 0-based index of the frame to be
> decoded/captured next.
> 2. CV_CAP_PROP_POS_AVI_RATIO Relative position of the video file
> 3. CV_CAP_PROP_FRAME_WIDTH Width of the frames in the video stream.
> 4. CV_CAP_PROP_FRAME_HEIGHT Height of the frames in the video stream.
> 5. CV_CAP_PROP_FPS Frame rate.
> 6. CV_CAP_PROP_FOURCC 4-character code of codec.
> 7. CV_CAP_PROP_FRAME_COUNT Number of frames in the video file.
> 8. CV_CAP_PROP_FORMAT Format of the Mat objects returned by retrieve() .
> 9. CV_CAP_PROP_MODE Backend-specific value indicating the current capture
> mode.
> 10. CV_CAP_PROP_BRIGHTNESS Brightness of the image (only for cameras).
> 11. CV_CAP_PROP_CONTRAST Contrast of the image (only for cameras).
> 12. CV_CAP_PROP_SATURATION Saturation of the image (only for cameras).
> 13. CV_CAP_PROP_HUE Hue of the image (only for cameras).
> 14. CV_CAP_PROP_GAIN Gain of the image (only for cameras).
> 15. CV_CAP_PROP_EXPOSURE Exposure (only for cameras).
> 16. CV_CAP_PROP_CONVERT_RGB Boolean flags indicating whether images should
> be converted to RGB.
> 17. CV_CAP_PROP_WHITE_BALANCE Currently unsupported
> 18. CV_CAP_PROP_RECTIFICATION Rectification flag for stereo cameras (note:
> only supported by

The framerate enumeration number is 5.  I use the enumeration number in
speed-cam.py rather than the opencv constant name.
Did some testing with my playstation camera that can go to 240 fps but it
would only accept 30 fps when initializing the webcam in speed-cam.py

eg

in
class WebcamVideoStream:

I added line below setting the  height to test  framerate without passing
it as a variable where 30 is a framerate value.  You will get errors if
framerate value is not compatible with your camera.  Note 5 is the opencv
enumeration value for setting frame rate.

        self.stream.set(5, 30)

You could try this to see if your camera can accept other frame rate values.

My experience is that webcams I have tried logitech, playstation only
support specific frame rate values, otherwise you will get an error.  To
make life simple I just left off setting  framerate for web cameras and the
webcam will use its own internal default

Claude .....

schnuerle, 2021-05-10:
Working on the **resolution upgrade first.** 

I have the same issue I had before, which is now it seems to count the same vehicle multiple times (instead of just once) and each time the speeds are wildly different. 

Here my settings before:

```
cal_obj_px_L2R = 70 
cal_obj_px_R2L = 65
pluginName = "webcam240" 
track_counter = 5 
track_timeout = 0.0
MIN_AREA = 100 
x_left = 100   
x_right = 220  
y_upper = 100  
y_lower = 175
WEBCAM_WIDTH = 320    
WEBCAM_HEIGHT = 240  
```

And here's what I changed from just those variables (trying some things [per this issue](https://github.com/pageauc/speed-camera/issues/55) to accommodate the 4x resolution:

```
cal_obj_px_L2R = 140 // doubled since cam with is doubled
cal_obj_px_R2L = 130
pluginName = "webcam480" 
track_counter = 10 // increased
track_timeout = 0.0 // no change
MIN_AREA = 200 // increased to prevent small motion detection, maybe should be quadrupled from 100 to 400 though
x_left = 200   // doubled
x_right = 440  
y_upper = 200  
y_lower = 350 
WEBCAM_WIDTH = 640    // doubled
WEBCAM_HEIGHT = 480  // doubled
```

But the speeds come in super fast, so walkers are 18 or 27 MPH, cars going about 25mph (I can see them out my window) are 60 to 75 MPH. So maybe 2-5 times faster than they should be.  And it misses some fast cars (me eyeballing them at around 35-40 MPH).

I started these variable edits small but kept adjusting and adding more variables so I'd get some results as I tested them. Maybe I'm doubling up somewhere?

closed 2021-05-27 by pageauc