pageauc/speed-camera #83: Reconcile Dockerfiles

RandallKent, 2021-05-07:
We have two Dockerfiles:

[/.devcontainer/Dockerfile](https://github.com/pageauc/speed-camera/blob/master/.devcontainer/Dockerfile) for Development

[/Dockerfile](https://github.com/pageauc/speed-camera/blob/master/Dockerfile) for "production" 

To reduce confusion we should consider combining the two.

ianmaciel, 2021-05-07:
Usually it make sense to have two different images: for development and production. So my comment [here](https://github.com/pageauc/speed-camera/pull/79#issuecomment-834314815) might be mistaken.

Anyway, it is easy to setup on devcontainer to use an external Dockerfile.
I am just not sure if the difference between both and if how can this cause any issue...

RandallKent, 2022-01-04:
I'm good with keeping the two separate. My only concern is around issues in production if the two images deviate materially. I'll close this out for now.

closed 2022-01-04 by RandallKent