pageauc/speed-camera #86: Calibration Issues

aximuseng, 2021-06-09:
I was trying to get this up and running - my calibration images will only trigger when something slow goes by.  For example these people walking by.  I will be able to go out an actual measure the distances etc but for now I think I need a proper calibration image.

Is there an initial default setting I need to change to trigger this for calibration and testing?

[image]

pageauc, 2021-06-09:
That is the calibration image.  It would appear the camera is too close.
Faster objects will likely go through the speed rectangle (red)  without
triggering.

You might try another location.  Also it would appear you are not on the
default stream size so you might want to revert to the default config.py to
start.

aximuseng, 2021-06-09:
I have the default config file and I can't really move the camera.  I have set the camera to the smaller frame size.  Here is what I get when a car drives by (not triggered a calibration image):

```
2021-06-09 17:14:42 INFO     speed_camera New  - 0/6 xy(108,35) Start New Track
2021-06-09 17:14:43 INFO     speed_camera  Add - 1/6 xy(91,58) 36.72 kph D=17/20 C=6 44x23=640 sqpx R2L
2021-06-09 17:14:43 INFO     speed_camera  Add - 2/6 xy(80,40) 33.37 kph D=11/20 C=4 56x42=1094 sqpx R2L
2021-06-09 17:14:43 INFO     speed_camera  Add - 3/6 xy(62,36) 68.08 kph D=18/20 C=6 64x47=1669 sqpx R2L
2021-06-09 17:14:43 INFO     speed_camera  Add - 4/6 xy(53,50) 46.50 kph D=9/20 C=7 40x33=702 sqpx R2L
2021-06-09 17:14:43 INFO     speed_camera  Add - 5/6 xy(34,38) 110.74 kph D=19/20 C=6 64x46=1880 sqpx R2L
2021-06-09 17:14:43 INFO     speed_camera  Out - 5/6 xy(59,38) Max D=25>=20px C=5 19x31=293 sqpx L2R
2021-06-09 17:14:43 INFO     speed_camera  Out - 5/6 xy(30,44) Max D=29>=20px C=3 19x25=262 sqpx R2L

```

aximuseng, 2021-06-09:
I lowered the track counter to 4 and got a car to trigger:

[image]

aximuseng, 2021-06-09:
Also to clarify this is from an IP camera.

pageauc, 2021-06-10:
I would decrease track_counter = 5

and increase
x_diff_max = 30 increase if you get a lot of OUT log entries. X-diff is the
maximum distance away in pixels from the previous track point. Since you
are close you can increase.
Adjust as required but not too high to avoid problems. Contours can move to
a different part of the vehicle due to lighting, shapes on object Etc.

Give that a try and see how that works.

Claude ...

aximuseng, 2021-06-10:
We were on the same page - settings I tweaked:

```
track_counter = 4
event_timeout = 1.0
x_diff_max = 60
cal_obj_px_L2R = 240      # L2R Moving Objects, Length of a calibration object in pixels
cal_obj_mm_L2R = 7920  # L2R Moving Objects, Length of the calibration object in millimetres

cal_obj_px_R2L = 240      # R2L Moving Objects, Length of a calibration object in pixels
cal_obj_mm_R2L = 7920  # R2L Moving Objects, Length of the calibration object in millimetres
```

Walked out an laid down some chalkiness.  The red box is roughly 26'.  Enough accuracy for now.

So here is me driving by @ 40 km/hr

[image]
[image]

and the corresponding logs:

```
2021-06-09 20:04:50 INFO     speed_camera New  - 0/4 xy(30,29) Start New Track
2021-06-09 20:05:09 INFO     speed_camera Reset- event_timer 18.67>0.50 sec Exceeded
2021-06-09 20:05:09 INFO     speed_camera New  - 0/4 xy(142,62) Start New Track
2021-06-09 20:05:09 INFO     speed_camera  Add - 1/4 xy(89,31) 135.73 kph D=53/100 C=4 80x50=2021 sqpx R2L
2021-06-09 20:05:09 INFO     speed_camera  Add - 2/4 xy(81,30) 19.89 kph D=8/100 C=1 131x52=3870 sqpx R2L
2021-06-09 20:05:09 INFO     speed_camera  Add - 3/4 xy(61,30) 38.09 kph D=20/100 C=1 143x53=4563 sqpx R2L
2021-06-09 20:05:09 INFO     speed_camera  Add - 4/4 xy(53,29) 18.97 kph D=8/100 C=2 134x52=4028 sqpx R2L
----------------------------- Create Calibration Image -----------------------------

  Instructions for using media/images/calib-20210609-2005094.jpg image for camera calibration

  Calibration Image Saved To /home/parallels/speed-camera/media/images/calib-20210609-2005094.jpg  
  View Calibration Image in Web Browser (Ensure webserver.py is started)

---------------------- Press cntl-c to Quit Calibration Mode -----------------------
2021-06-09 20:05:09 INFO     speed_camera  Saved media/images/calib-20210609-2005094.jpg
2021-06-09 20:05:09 INFO     isSQLite3  Success: File is sqlite3 Format /home/parallels/speed-camera/data/speed_cam.db
2021-06-09 20:05:09 INFO     db_check   Success: sqlite3 Connected to DB /home/parallels/speed-camera/data/speed_cam.db
2021-06-09 20:05:09 INFO     speed_camera  SQL - Inserted Data Row into /home/parallels/speed-camera/data/speed_cam.db
2021-06-09 20:05:09 INFO     speed_camera End  - R2L Ave Speed 53.2 kph Tracked 89 px in 0.207 sec Calib 240px 7920mm
2021-06-09 20:05:09 INFO     speed_camera track_timeout 0.10 sec Sleep to Avoid Tracking Same Object Multiple Times.
----------------------------------------------------------------------
2021-06-09 20:05:09 INFO     speed_camera New  - 0/4 xy(69,61) Start New Track
2021-06-09 20:05:09 INFO     speed_camera  Add - 1/4 xy(42,60) 78.59 kph D=27/100 C=5 19x20=238 sqpx R2L
2021-06-09 20:05:09 INFO     speed_camera  Add - 2/4 xy(74,30) 120.46 kph D=32/100 C=4 53x51=1628 sqpx L2R
2021-06-09 20:05:09 INFO     speed_camera  Add - 3/4 xy(63,29) 18.70 kph D=11/100 C=2 54x51=1945 sqpx R2L
2021-06-09 20:05:09 INFO     speed_camera  Add - 4/4 xy(28,30) 32.49 kph D=35/100 C=2 53x50=1768 sqpx R2L
----------------------------- Create Calibration Image -----------------------------

  Instructions for using media/images/calib-20210609-2005098.jpg image for camera calibration

  Calibration Image Saved To /home/parallels/speed-camera/media/images/calib-20210609-2005098.jpg  
  View Calibration Image in Web Browser (Ensure webserver.py is started)

---------------------- Press cntl-c to Quit Calibration Mode -----------------------
2021-06-09 20:05:09 INFO     speed_camera  Saved media/images/calib-20210609-2005098.jpg
2021-06-09 20:05:09 INFO     isSQLite3  Success: File is sqlite3 Format /home/parallels/speed-camera/data/speed_cam.db
2021-06-09 20:05:09 INFO     db_check   Success: sqlite3 Connected to DB /home/parallels/speed-camera/data/speed_cam.db
2021-06-09 20:05:09 INFO     speed_camera  SQL - Inserted Data Row into /home/parallels/speed-camera/data/speed_cam.db
2021-06-09 20:05:09 INFO     speed_camera End  - R2L Ave Speed 62.6 kph Tracked 41 px in 0.270 sec Calib 240px 7920mm
2021-06-09 20:05:09 INFO     speed_camera track_timeout 0.10 sec Sleep to Avoid Tracking Same Object Multiple Times.
----------------------------------------------------------------------
2021-06-09 20:05:36 INFO     speed_camera Reset- event_timer 26.62>0.50 sec Exceeded
2021-06-09 20:05:36 INFO     speed_camera New  - 0/4 xy(30,35) Start New Track
2021-06-09 20:05:36 INFO     speed_camera  Out - 0/4 xy(30,35) Min D=0<=1px C=3 182x71=7573 sqpx R2L
2021-06-09 20:05:36 INFO     speed_camera New  - 0/4 xy(148,80) Start New Track
```

I also drove by the other way right after and that didn't even trigger.  The one above is a double.

pageauc, 2021-06-10:
Set track timeout to 0.5 sec. This will avoid tracking object twice per
comment in config.py

closed 2021-06-26 by pageauc