pageauc/speed-camera #87: Ignoring People?

aximuseng, 2021-06-10:
Here is a really fast lady walking her dog 🤣

[image]

```
2021-06-10 13:36:00 INFO     speed_camera New  - 0/4 xy(48,65) Start New Track
2021-06-10 13:36:00 INFO     speed_camera  Add - 1/4 xy(84,99) 178.55 kph D=36/60 C=3 31x21=414 sqpx L2R
2021-06-10 13:36:01 INFO     speed_camera  Add - 2/4 xy(99,63) 7.56 kph D=15/60 C=3 32x57=1239 sqpx L2R
2021-06-10 13:36:01 INFO     speed_camera  Add - 3/4 xy(109,62) 26.84 kph D=10/60 C=3 28x58=1050 sqpx L2R
2021-06-10 13:36:01 INFO     speed_camera  Add - 4/4 xy(111,64) 4.84 kph D=2/60 C=3 29x56=684 sqpx L2R
2021-06-10 13:36:01 INFO     subDirChecks No sub folders Found in media/images
2021-06-10 13:36:01 INFO     subDirCreate Created media/images/speed-20210610-1336
2021-06-10 13:36:01 INFO     speed_camera  Saved media/images/speed-20210610-1336/speed-20210610-1336013.jpg
2021-06-10 13:36:01 INFO     isSQLite3  Success: File is sqlite3 Format /home/parallels/speed-camera/data/speed_cam.db
2021-06-10 13:36:01 INFO     db_check   Success: sqlite3 Connected to DB /home/parallels/speed-camera/data/speed_cam.db
2021-06-10 13:36:01 INFO     speed_camera  SQL - Inserted Data Row into /home/parallels/speed-camera/data/speed_cam.db
2021-06-10 13:36:01 INFO     log_to_csv    CSV - Appended Data into /home/parallels/speed-camera/speed-cam.csv
2021-06-10 13:36:01 INFO     saveRecent    symlink /home/parallels/speed-camera/media/recent/speed-20210610-1336013.jpg
2021-06-10 13:36:01 INFO     speed_camera End  - L2R Ave Speed 54.5 kph Tracked 63 px in 0.353 sec Calib 240px 7920mm
2021-06-10 13:36:01 INFO     speed_camera track_timeout 0.50 sec Sleep to Avoid Tracking Same Object Multiple Times.
```

Any tweaks that can be made to prevent these ones from getting logged?

pageauc, 2021-06-10:
This can be caused by contours moving from one object to another. also lhey
are closer than the calibration distance.

aximuseng, 2021-06-10:
So can this work still with cameras at an angle?  I have my normal house security cameras that are higher resolution and set back farther - they are at an angle to this same street.

closed 2021-06-26 by pageauc