pageauc/speed-camera #88: Reports Inop on my new install 6-26-21

mdtaylorlrim, 2021-06-21:
Installed speed-camera and all works well except for the reports. The menubox.sh displays the reports menu, and the reports menu calls the sql_speed_gt.py program but it does not exist. Further, the sql statement references the column log_hour which does not exist in the db.

I thought I had followed the install instructions correctly using the speed_install script, but it just did not install the sql_speed_gt.py program. And I wonder what else it might not have installed.

Should I do in upgrade from the menubox menu? Will it alter my config settings if i did that? Or use some other method to retrieve the missing files?

Thanks
Mark

pageauc, 2021-06-21:
Recently added  "graphssql-make-graph-count-totals.py" and "sql-make-graph-speed-ave.py".  These use matplotlib instead of gnuplot.  They can be run using parameters.  type command with -h to see help.  They can also be used to run multiple graphs using the config.py parameters. The gnuplot report is still on github but is not installed.  I have not yet update menubox.sh.
You can install per

    cd ~/speed-camera
    wget -O sql_speed_gt.py https://raw.github.com/pageauc/speed-camera/master/sql_speed_gt.py
    chmod +x sql_speed_gt.py

Note I am considering removing this gnuplot report but will look at adding query to generate html listings with date range.  The current aql_speed_gt.py does not include date range and this can be a problem with a lot of data.  Still have to update menubox.sh.

Hope this answers your question.
Claude ...

mdtaylorlrim, 2021-06-21:
Thanks! It does, but gnuplot is not installed on my pi. I'll have to look up how to install it unless you know off the top of your head...

Thanks again,
Mark

pageauc, 2021-06-21:
sudo apt install python-gnuplot

mdtaylorlrim, 2021-06-21:
Got it. Thanks!

closed 2021-06-21 by mdtaylorlrim