pageauc/speed-camera #89: [idea/question] How could I implement a flash? (speed camera as party attraction

LeoLTM, 2021-06-22:
First of all, thanks for your awesome project and the ton of work you put into it!

Together with my friends, I wanted to build a speed camera clone for events like kid birthday partys where 
it would act as a very special photobox and people could drive through it with shopping carts, bobbycars and bicycles.

I'm not very good at programming python scripts but very familiar with Linux and RPIs, so I thought maybe the creator itself
can think of a way to include a flash (could just be the activation of a relais). I've read of the possibility to create own plugins for the project, so maybe this is possible somehow. The part I'm stuck at the moment is the "how to get the correct "flash-time" from the running python program. When taking a picture, the program should simply f.e. activate a GPIO pin or trigger another routine to send multiple GPIO signals to a DMX compatible LED washer that then turns on for a quarter of a second.

pageauc, 2021-06-22:
What model and brand is the dmx led wall washer.
How Is it controlled, USB, Serial or can it be controlled by GPIO signals?

Here is a video on an adventure to setup DMX control for a wall washer via
RPI.  I don't think you need something this complicated.
https://www.youtube.com/watch?v=3rJIqyxE3aY

Something like this might work and is a lot cheaper and can be controlled
via Gpio.
https://www.amazon.ca/SunFounder-Color-Module-Arduino-Raspberry/dp/B013QQJZYQ/ref=sr_1_5?dchild=1&keywords=raspberry+pi+led+flash&qid=1624377581&sr=8-5

You would need to activate the LED before speed camera saves video stream.
This could be triggered by the counter.
You might want to try something like below.  I am going to buy this for one
of my Pi-Timolo RPI cameras

https://www.amazon.ca/Bright-White-Camera-Raspberry-Pi/dp/B01N24G7J7

Let me know details for your LED solution and I can help integrate into
speed camera as an option. I will look at buying the kit above and get it
to trigger from speed camera.
I will also use it with my PI-TIMOLO project since it has motion tracking
(No speed).  Will also need to add option and code for the i2c LED control.

Claude ....

LeoLTM, 2021-06-22:
Sadly I don't know the exact model and I'll need a few days to look it up because I store the LED Washers in another place.

It looks something like this: 
[image]
The basic two DMX connectors (in and out for control with a DMX interface), the 4 digit display and the 4 buttons for on-device setup and the 230V socket. 

While browsing the web I found this: 
[https://www.hackster.io/max_loser/control-a-led-spotlight-set-up-with-an-arduino-via-dmx-904b66](url)
Maybe it's possible to hook an arduino UNO up to the GPIO pins and communicate with it using the TX/RX pins.
(Because there is a DMX libary for arduino)

It's possible to set the LED color to red with the integrated buttons but there has to be a signal input for the dimmer (default 0%, when the speed camera takes a photo set it to 100%, wait 0,25 seconds, set it to 0% again)

The flash has to be very bright so that it'll be visible in the sunlight because it'll be hidden behind a piece of red plastic. The clone imitates a very widespread speed camera model:
The Vitronic PoliScan Speed in its stationary form. 
Here's a picture for reference:
[https://external-content.duckduckgo.com/iu/?u=https%3A%2F%2Fi.ytimg.com%2Fvi%2FwM2zzcZCxnw%2Fmaxresdefault.jpg&f=1&nofb=1](url)
(don't be confused because of the 45° measuring angle, ours will be 90° for best results with the webcam 👍🏻 )

Regarding the use of a RPi: 
I've tested the speed-camera on my RPi 3b with a 720p USB webcam but openCV ran very choppy (different resolution options were tested) so I set everything up on a old i3 notebook with Python 3 and openCV. While using an rtsp ip webcam, the performance on 720p is really great so I'd like to implement this solution into the final build because of the flexibility (no RPi in the speed camera itself and better problem solving due to direct access to the machine that is processing the images).

I know that this will make the connection between the machine where the calculations are done and the LED flash way more complicated but maybe there's a way.

pageauc, 2021-06-22:
You might try just using a PIR sensor switch.  See item below.  Can be set
to turn off after 5 seconds.  Just connect to lights you want to control.
I don't think this has a light sensor so will work OK during daylight since
switch activates by infra red body heat motion.

https://www.amazon.ca/Motion-Detector-Infrared-Inductor-Outdoor/dp/B08GXBGK3G/ref=sr_1_15?crid=WZ71TQQM3INR&dchild=1&keywords=pir+motion+sensor+switch&qid=1624402205&sprefix=pir+motion%2Caps%2C167&sr=8-15

Just a thought.

LeoLTM, 2021-06-22:
Yep, this'd be possible. Another way would be a light barrier triggering the arduino but this could be activated without being above the speed limit and would so false-indicate a saved picture for the participants,

pageauc, 2021-06-22:
You could just wire an extension plug to one side of PIR switch and a plug
into the other side.  Could be wired with one extension cable using plug
side for PIR power and receptacle side for PIR outlet side.
Then just plug your lights into PIR sensor plug side and other end into 120
v outlet for power.  Then just set PIR sensor for shortest time on.  May
even be shorter than 5 seconds.
No changes would need to be done to speed camera.

Let me know your progress.
Claude ...

LeoLTM, 2021-06-30:
I ordered and now recieved a 50W white LED floodlight which I want to turn on/off with an relais hooked up to a ESP8266.
The code for triggering the relais over the local network should be no problem for me, but I need your help on implementing a line in the python code (in the correct place) to send the network command to a specific IP address (e.g. 192.168.1.102).

pageauc, 2021-06-30:
In my pi-timolo project I added  user_motion_code.py with an empty function
that users can insert their code into.This is not overwritten during an
UPGRADE.
pi-timolo.py will import user_motion_code.py if it exists and run the
user_motion_code.userMotionCode(filename) function.  The filename is passed
to allow
image to be processed if required by the user. See
https://github.com/pageauc/pi-timolo/blob/master/source/pi-timolo.py
and do ctrl-f to search the code for user_motion_code text.

I can implement a similar setup for speed-cam.py if you like.  pi-timolo
executed the code after the image was taken but not sure if this would
be suitable for you.  The only problem with executing user code before the
image stream frame is taken is that it would slow or delay capturing the
image
and possibly miss the object in motion.

It is also possible to trigger the led with an infrared detector so not
changes to speed-cam.py would be required and it would work in low light
conditions.  Eg

   1. Dark Out
   2. infrared motion detector sees motion and turns on LED
   3. Area now lit up
   4. speed-camera now has lighting and tracks object and takes speed image.
   5. infrared motion detector delay timer turns off led and waits for next
   infrared motion.
   6. Dark Out

Here is an example PIR motion sensor as an example.
https://www.amazon.ca/Outdoor-90-250V-Security-Infrared-Detector/dp/B07FFWGCXD/ref=sr_1_26?dchild=1&keywords=infrared+motion+detector&qid=1625054026&sr=8-26

Claude ....

pageauc, 2021-06-30:
here is another PIR switch
https://www.amazon.ca/Sensky-Function-Control-Controller-Lighting/dp/B00L58A8EE/ref=sr_1_7?dchild=1&keywords=pir+sensor+switch&qid=1625054502&sr=8-7

LeoLTM, 2021-06-30:
An implementation of the user_motion_code.py -functionality would be great!! Then I'd be able to write a python code snippet that'll be executed after a picture was taken (because the flash is just for fun, it's not that bad to first take a picture and then activate the flash with a minimal delay).

Keep me updated :)

LeoLTM

pageauc, 2021-06-30:
I have added user_motion_code feature per discussion.  This tested OK for
me.  Read the the comments in user_motion_code.py.

Note when this code is imported a user_motion_code.pyc will be created by
python.  This can be ignored and is part of python

Let me know how you do.

Claude ...

pageauc, 2021-06-30:
You can do a normal menubox.sh UPGRADE menu pick to get the new version.

LeoLTM, 2021-06-30:
Wow! Thanks for the fast implementation of the idea! I'll give it a try as soon as possible.

pageauc, 2021-06-30:
The code was already in pi-timolo so just a matter of inserting, editing
and testing in speed-cam.py
Claude ...

pageauc, 2021-07-02:
Added user_motion_code feature and also provided using PIR switch to control lighting

closed 2021-07-02 by pageauc

LeoLTM, 2021-07-02:
I've tried to insert my own python code in the user_motion_code.py but it only runs once (when starting the speed-cam.py)
What's wrong with it?

def userMotionCode(filenamePath):
  
    Users can put code here that needs to be run
    after speed camera motion tracking and image taken
    Eg Notify or activate something.
    Note all functions and variables will be imported.
    speed-cam.py will execute this function userMotionCode(filename)
    in speed-cam.py per example below
        user_motion_code.userMotionCode(filename)
    
    # Insert User python code Below
    # print("User Code Executing from userMotionCode function")
    # print("file path is %s" % filenamePath)

import urllib.request
import http

base = "http://192.168.0.102/"

def transfer(my_url):   #use to send and receive data
    try:
        n = urllib.request.urlopen(base + my_url).read()
        n = n.decode("utf-8")
        return n

    except http.client.HTTPException as e:
        return e

two = transfer("two")
one = transfer("one")
print(one,two)
_ = transfer("255")   #Send this data

pageauc, 2021-07-02:
User motion code should run after each speed image is taken.  Will take
another look at your code.
Also request is only python3 so you would need to run

python3 speed-cam.py

Not sure what your indent above gut should look something like this

def userMotionCode(filePath):
> import urllib.request
>     import http
>
>     base = "http://192.168.0.102/"
>
>     def transfer(my_url): #use to send and receive data
>         try:
>             n = urllib.request.urlopen(base + [...]
 speed-cam.py imports user_motion_code
then executes

user_motion_code.userMotionCode(filename)

I could not run the code because not sure what is at 192.168.0.102 tor
receive transfer
Claude ...

LeoLTM, 2021-07-03:
On the other end there is an Wemos D1 Mini (ESP8266 board) connected to a relay switching the flash on and off.

Arduino CODE:

#include "ESP_Wahaj.h" 

#define relais D4
int pwm = 0;  //255
String path = "nothing";
void setup(){
  Serial.begin(9600);
  start("speedcamera","speedcamera");  
  

}

void loop(){
  

  pinMode(relais, OUTPUT);

  if(CheckNewReq() == 1)
  {
   
    if (getPath()=="/two"){
    returnThisStr("two replied");
    }
    else if (getPath()=="/one"){
    returnThisStr("one replied");
    }
    else if(getPath()=="/favicon.ico"){   
      returnThisStr("garbage");
    }

    else        
    {
      path = getPath();
      Serial.println(path);   
     
      path.remove(0,1);   
      Serial.println(path);
      pwm = path.toInt();   
      Serial.println(pwm);
    }
    
  }
  

  if(pwm == 255) {

  digitalWrite(relais, HIGH);
  delay(2000);
  digitalWrite(relais, LOW);
  }

}

pageauc, 2021-10-11:
> Hey there! After a few months, I finally found the time to finish my
> little project. I filled a pyhton request code into the user_motion_code.py
> file which informs the ESP8266 that a picture has been taken. The ESP then
> sets a pin to HIGH (opens relay) for 100ms and the LED flash connected to
> the relay flashes once.
> Now I'm busy dialing in all the settings for the motion tracking to get
> perfect results.
> One of the problems I'm facing right now is the delay between my camera
> and openCV (the user_motion_code.py gets executed a bit late).

The way the logic goes the user_motion_code is well after the speed stream
image is read (line 1208) and processed.  You may want to take another
image (note without text and motion area lines)
This could overwrite the original image but would be close to your flash
logic but there still will be a slight delay that could allow the moving
image to be out of frame.
The logic is based on a stream image so not sure what affect the flash
would have since the camera would need time to adjust white balance in the
video stream.

Here is my suggestion (untested)
Set config.py track_counter = 4  or maybe even 3

in user_motion_code  Turn on Led then read another image stream

example

    # code to turn on LED lighting and leave on for short period to allow
camera to adjust or turn on then turn off after image taken below
    image2 = vs.read()
    # image2 = speed_image_add_lines(image2, cvRed)
    image_led = cv2.resize(image2, (image_width, image_height))
    cv2.imwrite(filename, image_led)

See if this works

pageauc, 2021-10-13:
> Tried to add your code to my user_motion_code.py, then this happened:
>
> Traceback (most recent call last): File
> "E:\Dokumente2\speed-camera-master\speed-cam.py", line 1698, in <module>
> speed_camera() # run main speed camera processing loop File
> "E:\Dokumente2\speed-camera-master\speed-cam.py", line 1390, in
> speed_camera user_motion_code.userMotionCode(filename) File
> "E:\Dokumente2\speed-camera-master\user_motion_code.py", line 8, in
> userMotionCode image2 = vs.read() NameError: name 'vs' is not defined
>
> *This is my user_motion_code.py:*
>
> `def userMotionCode(filePath):
>
> import requests
> r =requests.get('http://192.168.2.105/5/on')
>
> code to turn on LED lighting and leave on for short period to allow camera
> to adjust or turn on then turn off after image taken below
>
> image2 = vs.read()
> # image2 = speed_image_add_lines(image2, cvRed)
> image_led = cv2.resize(image2, (image_width, image_height))
> cv2.imwrite(filename, image_led)`

OK realized the problem.  I have modified speed_cam.py to send required
variables to user_motion_code and updated user_motion_code
to accept the variables.  This is not elegant but it works.

Move your existing user_motion_code.py to a backup since the UPGRADE will
not update this file if it already exists.

    cd ~/speed_camera
    mv user_motion_code.py user_mottion_code_bak.py

Do a menubox.sh UPGRADE

Edit user_motion_code.py to add your code.  See sample code below

   import requests
r = requests.get('http://192.168.2.105/5/on')
    import cv2
    cvRed = (0, 0, 255)

    image2 = vs.read()
    image_led = cv2.resize(image2, (image_width, image_height))
    cv2.imwrite(filenamePath, image_led)
    print("user_motion_code - Saved new LED Flash image to %s"
 %filenamePath)

Let me know if this works.