pageauc/speed-camera #91: gnu plot missing after initial install

peterb99, 2021-06-23:
Im using a raspberry pi with minimal kernel image ( no desktop ) as headless node and using ssh to remote manage it.
When trying to create a report via the menubox.sh following error message pops up:
Loading ...
Traceback (most recent call last):
  File "./sql_speed_gt.py", line 12, in <module>
    import Gnuplot
ImportError: No module named Gnuplot

View this report in html format
From Speed-Camera Web Page in media/reports folder
Web browser url is http://192.168.0.141:8080

######################################
#          Review Output             #
######################################
  Press Enter to Return to Main Menu

pageauc, 2021-06-23:
The latest version of speed-camera on github fixes the problem.  Note I am
trying to replace gnuplot with the more popular matplotlib. Also there are
new matplotlib scripts that generate various graphs from command line or
multiple reports from config.py settings.
Claude ....

peterb99, 2021-06-24:
gnu plot does not seem to work when no desktop ( x11 ) is installed
This Report will Display all Records with Speeds Over
Specified Value. SPEED_OVER Must be Integer. 0=all
Enter SPEED_OVER: 35
2021-06-24 11:35:11 INFO     make_html  Working.  Please Wait ...
2021-06-24 11:35:11 INFO     make_html  Connect to Database data/speed_cam.db
2021-06-24 11:35:11 INFO     make_html  Saved html File to media/reports/hour_count_gt35_list.html
2021-06-24 11:35:11 INFO     make_graph_data Working.  Please Wait ...
2021-06-24 11:35:11 INFO     make_graph_data Connect to Database data/speed_cam.db
2021-06-24 11:35:11 INFO     make_graph_data Saved html File to media/reports/hour_count_gt35_totals.html
2021-06-24 11:35:11 INFO     make_graph_image Saved Graph File To media/reports/hour_count_gt35_graph.jpg

gnuplot: unable to open display ''
gnuplot: X11 aborted.
2021-06-24 11:35:11 INFO     <module>   Processing Took 0.26 s.
2021-06-24 11:35:11 INFO     <module>   View Reports from Speed Camera Web Page under media/reports

pageauc, 2021-06-24:
> 2021-06-24 11:35:11 INFO Processing Took 0.26 s.
> 2021-06-24 11:35:11 INFO View Reports from Speed Camera Web Page under

try running the matplotlib graphs.  They are not setup on menubox but can
be run directly from console

Not sure why gnuplot is failing but must be newer version.  matplotlib had
similar issue but can be turned off.
possibly the newer version of gnuplot needs command to avoid needing an
open display.  Will look into problem
but I am really trying to not support gnuplot and prefer matplotlib.

closed 2021-06-26 by pageauc

pageauc, 2021-06-26:
Please Note I have update speed camera menubox.sh REPORTS menu to run various graphing and html report listings.''

You can update using menubox.sh UPGRADE menu pick.  When done Exit menubox.sh and restart to see the new REPORTS selections.

The reports can also be run from the command line per program instructions.
Claude ...